sui77/rc-switch #147: Need help with a sample for 2262 chip clone

uckoo, 2017-05-30:
Firstly, a big thanks to all the contributors of this library. It looks so simple to implement.

I am here for some help to clone the functionality of PT2262 as is. By that I mean I would like to know about an example where I can control all the A0-A5 and D0-D5, i.e. I am not specific about any equipment to be controlled, but just implement what a 2262 can do.

I have gone through the WiKi link https://github.com/sui77/rc-switch/wiki/KnowHow_LineCoding
and understand the format of codeword being sent 

However my questions:
(a) The data sheet shows the order of data being
0 Data: A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 A10 A11 Sync Bit 
but the wiki link shows the "Sync Bit" first
so which is the correct way/order to implement on the arduino?

(b) I have seen few examples and I couldnt find a direct example as such which would emulate PT2262.
Can someone please help me with a relevant one?
I am supposing it would be sendTriState(?) but the examples show only 12trits data, so what should be the equivalent including the Sync Bit?

Thank you! If you need more info please do let me know.(I plan to implement this on TINY85 eventually)

Martin-Laclaustra, 2017-05-30:
Quick answers:
(a) Several repeats of the train of pulses are sent, and the sync bit is in-between. That means it is equivalent upfront or behind. Different datasheets describe them in one or the other position.
(b) The sync bit is sent besides the 12 trits = 24 bits of data. It is already included in the function. I am not sure what you mean with an example or what you do pretend to emulate. Explaining what you intend would make it easier to receive help.
Do you have the chip at hand? can you use it to encode information? You could compare it with what the function provides. You might find helpful this [branch](https://github.com/sui77/rc-switch/issues/146) in which I coded a way to access to pulses generated by the functions. You could compare those with the ones from the chip.
(c) I only tested the HS2260A-R4. I believe that the signal is generated most significant pins (MSB) first, but should re-check. If the chip pin is connected to Vcc, it places a tri-1 (equivalent to bits 11), if it is connected to ground, it places a tri-0 (equivalent to bits 00), if it is connected to neither, that is, it is floating, it places a tri-F (equivalent to bits 01). I would need to check whether these are the exact equivalences (could be Vcc=0;G=1;floating=10). The datasheet will be clear with this. One more detail. There are variants of this chip R2; R4; R6... The difference is that for the least significant 2 (or 4, or 6) pins the floating state generates tri-0, there is no tri-F for these pins. This characteristic might be programmable in other chips.

uckoo, 2017-06-02:
Thank you for your reply. Appreciate your time.

By emulate I mean what should be the function to execute so that I send the same data a PT2262 would send. I.e. say when A0-A7 are grounded a d D0-D1 are pulled up to Vcc.

Thank you also for clarifying about the synch bit, that helped me understand.

I ll actually need to go and check what would be the decoder PT2272 version whether M2/3/4. The 2262 I see is just one variant.

I shall get a encoder/decoder pair and try do a comparison.

Another quick question, is there a certain delay I need to maintain between two send functions, so that messages don't get mixed up?This way I ll just send all possible combinations at a go and not worry about my particular bits.

Martin-Laclaustra, 2017-06-06:
Sorry. I do not know. Please experiment and share the knowledge.
Most implementations repeat the same code several times. I do not know if separation is needed at all. I believe that receivers will have varying requirements.

uckoo, 2017-06-07:
Thank you, appreciate your help. I shall update here what I find out.