sui77/rc-switch #151: I can't get this switch to work (HN3031)

donnib, 2017-07-03:
Hi,
I have this switch http://www.harald-nyborg.dk/p3031/stikk-fjernbetjent-hvid-1609x which is extremely similar to Elro Home Control Funk Set / Brennenstuhl Comfort RCS 1000N and Arendo on the form factor.

I used the remote and i register the code, bits and pulse length to following :

Protocol : 2
Bits: 32
Length: 714

I have tried to send the code from remote but the switch does not react (i have not quite tested the sender but i think it works). The weird this is that the remote shows multiple codes on the same button all with same details as above so i tried all the codes but nothings happens on the switch. 

Any ideas ?

NicolaiJAndersen, 2017-10-10:
I have the same device, but no remote, and I want to control it with arduino.
Does anyone have an idea of which codes it accepts?
HN3031 - Model 55006DKX10

Regards
Nico

NicolaiJAndersen, 2017-10-11:
Hi Donnib.

Do you have a raw output sample from your remote control (HN 3024) with
code, bit, pulse length,
just for key A On?

Regards
Nico
Phone: +45 29878698

sgh, 2017-11-10:
Did you get it to work? The problem as I see is that the switch initialy has to learm codes that controls it. This probably means that that there are very specific constraints as to what we can initialy send to it. It also means that just grabing someone elses code will not work.

NicolaiJAndersen, 2017-11-10:
Håber det er OK, på dansk.
Nej, jeg har ikke fået det til at fungerer med HN3031.
Så jeg bruger de gamle HN6137 med DIP-switch, de kan også modificeres fra 1000W til 2300W.

sgh, 2017-11-10:
øv - jeg købte 1 stk. HN3031 til 10 kr og dens storebror med lysdæmper til 20 kr i Harald Skrald. Hvilke koder sender du til HN6137?

NicolaiJAndersen, 2017-11-10:
Hehe jeg købte også 1 stk. til 10kr.
Koden til HN6137, kan ses på dip-switch'en bagpå.
ON-stilling = 1 og OFF-stilling = 0
Fx. 10101 - 10000 = 1,3,5 - A

NicolaiJAndersen, 2017-11-10:
Ang. HN6137 så se evt. Arduino demo koden.

https://github.com/sui77/rc-switch/blob/master/examples/TypeA_WithDIPSwitches/TypeA_WithDIPSwitches.ino

Caraculo, 2017-11-23:
Hej drenge

Jeg vil også købe sådanne nogle [HN 3024'er](https://www.harald-nyborg.dk/p3031/stikkontakt-til-fjernbetj-hvid) til en 10'er hos HN, men vil ikke gøre det, hvis jeg ikke kan styre dem. Jeg vil derfor høre om I er kommet videre.

Jeg bruger RFLink i Arduino sammen med Domoticz. En kode fra en fjernbetjening vil i sådan et setup se sådan ud (fra min dørklokke): **_20;22;Aster;ID=0790;SWITCH=01;CMD=ON;_**
Dette kan nemt fanges af RFLink, når man trykker på fjernbetjeningen, hvorfra man kan bruge det til at programmere Domoticz. **RFLinks syntaks for datapakker (a la ovenstående eksempel) er beskrevet [her](http://www.rflink.nl/blog2/protref).**

Det fjernbetjenings output i denne format jeg har brug for og vil derfor høre om I er kommet videre med at aflæse/afkode senderens koder. Havde jeg en af fjernbetjening kunne jeg nemt aflæse koden, men det har jeg desværre ikke.

sgh, 2017-11-24:
Det hjælpe så vidt jeg ved ikke at aflæse koderne da kontakterne skal lære koderne.

Caraculo, 2017-11-24:
Hmmm... det kan jeg godt følge. Tænker dog, at hver knap sender sin egen kode. Jeg tror jeg vil købe et af de nyere sæt og aflure, hvad de gør.

Har i øvrigt fundet et sted, hvor der uploades koder [letscontrolit.com (HN 2794)](https://www.letscontrolit.com/forum/viewtopic.php?t=2734).