sui77/rc-switch #154: Etekcity Outlets, sending/learning new custom code

jlarocheva, 2017-07-17:
Hi,

I bought a bunch of the Etekcity 433mhz outlets. I sniffed the codes, and have successfully controlled these outlets via RCSwitch/Arduino/433mhz transmitter.

I am trying to make the outlet learn a new code. I can use the remote that came with it to change the code but it will not learn a code sent out every 20-50ms (I tried 21...22...23...24...25 all the way to 50). The outlet is not learning the new code. I sent out a six digit code at 180 and 190 for the pulse width.

Has anyone managed to set a custom code on one of these outlets with RCSwitch?

Thanks

timleland, 2017-12-06:
Check out this article https://hackaday.io/project/4298-autoflower/log/14215-reverse-reverse

williams6946, 2018-09-28:
This is an old post but if you are still trying to make arduino work with Etekcity using rcswitch I may have
a solution.  I had similar problem.  Unplug the Etekcity outlet from the wall.  Hold the button while plugging it back in.  Then release the button.  This will erase the old code so you can program a new one.
Hope this helps you