sui77/rc-switch #160: Superheterodyne receiver ( WL101-341) problem

dajak, 2017-08-29:
I recently bought 433 MHz Superheterodyne receiver ( WL101-341) and I'm having problem receving signal from my 433 MHz remote. When I test my script everything is fine if the Serial monitor in Arduino enviroment is on. If i close the serial monitor, the 433 MHz Superheterodyne receiver stops receving signal.
So the problem is related to this specific reciver model which requires Serial monitor to be opened for the rc-switch library to function correctly.
If i replace 433 MHz Superheterodyne receiver ( WL101-341) with cheap XY-MK-5v receiver the script and rc-switch library runs correctly regarding the status of Serial monitor.
Has anyone had similar issue with this 433 MHz Superheterodyne receiver ( WL101-341) ?

Jason2866, 2018-10-04:
@dajak 
I use it without any issue with a Wemos mini (ESP8266).
https://qiachip.com/product/superheterodyne-433mhz-rf-transmitter-and-receiver-module

oh2xx, 2019-01-20:
Hi I have the same problem. I am using QIACHIP 433mhz RF Transmitter and Receiver and ESP8266 (Wemos). I do not receive anything if serial monitor is closed. I tested also so that I do not even open the serial connection in my script but even then I need to open the serial monitor (which is blank) to receive anything. Mysterious.

oh2xx, 2019-01-21:
Solved this one today. It seems that for some reason interrupt 0 (Wemos pin D3) does not work without serial monitor being open. Changed the receiver to interrupt 5 - mySwitch.enableReceive(5) - that being Wemos pin D1 and all is OK now.

gmonitoring, 2023-03-25:
Serial.println(mySwitch.getReceivedValue()) and mySwitch.available() always = 0. Tried pin D4 and D0 on mySwitch.enableReceive(0). esp8266 board, module (WL101-341). When you press the button on the remote, the light on the esp flashes, but the port always outputs false. What can I try please help