sui77/rc-switch #162: Protocol 4 doesn't send

deW1, 2017-09-05:
I have set up a XY-FST transmitter. Now when I use protocol 1, 2, 3, 5 I get the information back without any problem. The second I switch to protocol 4 nothing is happening. (same code just setProtocol(4);)

When receiving the data from the remote I'm trying to read it's:
`Decimal: 2 (4Bit) Binary: 0010 Tri-State: not applicable PulseLength: 812 microseconds Protocol: 4
Raw data: 4712,3840,924,2876,924,2868,2828,972,924,`

How would I send this received data properly? The pulse length seems a bit off to me, though.

Martin-Laclaustra, 2017-09-08:
That seems a extremely short signal (4 bits). You are probably missing something and that is not the signal from your remote.
Try listening with my branch: https://github.com/sui77/rc-switch/issues/103
and/or recording a longer set of timings with https://github.com/sui77/SimpleRcScanner
(There is information on how to interpret results across other issues)

deW1, 2017-09-10:
I got the same result with your branch. either received 1, 2, 4, 8 decimal, which corresponds to each button pressed on the remote. I'll investigate, thanks for the tools.

closed 2017-09-10 by deW1