sui77/rc-switch #163: AC144 Projector Screen Remote

pree, 2017-09-08:
Hey everyone o/

I've tried out receiving the signals from a AC144 Remote which is used for my Projector Screen.
I'm Using an ESP8266 (GND<>GND, VCC<>VIN, DATA<>D3) and a couple of different 433MHz RF Receivers, just to be sure.

When pressing a button on the remote nothing happens in the serial console of the esp8266 (Example Receive Sketch, Simple & Advanced tested).  So i cracked the remote open to see what Chip is inside and its not a listed one.
I can read `EM78P153KSO14J` `1617P` `BTS164M4` on the Chip and found a [Datasheet](http://www.emc.com.tw/twn/database/Data_Sheet/8BIT/EM78P153K.pdf) which could be the right one. 

Can someone help me receiving the signals ?

--- Image of the Chip: ---

[image]

Martin-Laclaustra, 2017-09-08:
Try listening with my branch: #103
and/or recording the timings with https://github.com/sui77/SimpleRcScanner
(There is information on how to interpret results across other issues)
(sorry, by now, this is just generic advice, as the datasheet is really lengthy)

pree, 2017-09-08:
Thanks @Martin-Laclaustra, you brought me bit further.

I received the button down from the remote with the SimpleRcScanner and plotted it on the website:

[image]

Now its time for me to learn how to decode it :p

pree, 2017-09-08:
I came up with `{ 290, { 18, 18 }, { 1, 2 }, { 2, 1 }, false }`.
Added it into the protocol array in RCSwitch.cpp but cannot see anything when pressing the buttons.
Have I done the calculations right ?

Martin-Laclaustra, 2017-09-09:
Good job! You almost got there.
Unfortunately, you may get into trouble. You will not be able to use the library without modifying.
Your code is 64 bits ({ 2, 1 }, { 1, 2 }) + 2 sync bits ({2,18},{18,2}). You may get away with:
{ 290, { 18, 2 }, { 2, 1 }, { 1, 2 }, false } and sending 65 bits, with the last bit always 1 ({2,1})
but it is uncertain. It might work only for receiving or sending or neither.
For sure you will need to modify the code variable to host such lengthy bit-stream.
May be you can have a look at branches from some collaborators for the new nexa, or kaku. I think those are 64 bits and with 2 sync bits.

Martin-Laclaustra, 2017-09-09:
If you just want to control your screen, you could hardcode the timings.
Check #146 for ideas. I did not develop that branch much more because I did not perceive much interest, but the idea was to allow rcswitch to send arbitrary hardcoded timings sequences when protocols were not supported.

pree, 2017-09-09:
Thanks Again!

Just tried it out to send & receive with the protocol and 65 bits, but sadly nothing happened.
Also tried sending the raw data with your branch like this:

[image]

I don't know if I did this right, but it's sadly not working either.
I've received the signal twice yesterday to make sure, but could it be possible that the signal is not right or some sort of rolling code ?

pree, 2017-09-11:
I've just found the [PDF](https://fccid.io/YC5AC114-01/Test-Report/Test-Report-1286584.pdf) for the Test-Report done on the remote. There are some diagrams on site 19-22. Could those be usefull ?

Martin-Laclaustra, 2017-09-12:
Try plugging in these values:
`5238,5172,611,568,299,263,599,580,290,262,610,262,609,261,612,569,301,569,294,270,600,580,289,263,610,570,300,262,598,582,290,271,600,579,297,259,610,262,609,261,600,581,289,270,603,579,290,580,292,567,305,261,602,578,291,577,296,258,612,568,303,566,305,258,600,579,296,269,602,259,613,258,612,258,614,259,602,269,602,259,611,579,296,258,612,261,610,261,601,271,600,259,612,262,608,262,611,259,616,260,602,271,598,580,292,260,610,261,610,261,609,572,289,582,294,577,292,580,290,570,303,568,290,580,290,580,292,259,612,579,288,568
`
or these ones, cleaned to fit deduced timings:
`5220,5220,580,580,290,290,580,580,290,290,580,290,580,290,580,580,290,580,290,290,580,580,290,290,580,580,290,290,580,580,290,290,580,580,290,290,580,290,580,290,580,580,290,290,580,580,290,580,290,580,290,290,580,580,290,580,290,290,580,580,290,580,290,290,580,580,290,290,580,290,580,290,580,290,580,290,580,290,580,290,580,580,290,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,580,290,290,580,290,580,290,580,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,290,580,580,290,580
`
This are "first timing low". Make sure that they are emitted that way.

If this does not work, record the "artificial" signal and compare it to your previously recorded signal. There is no mystery. Tweak the timings and when they are equal to the ones that you recorded at the beginning your receptor will react.

Best luck (do not forget to report back if you succeed)!

pree, 2017-09-13:
I just tried to send it and record it with a different board. 
After 4 tries I didn't get any similar looking signal so i switched the transmitter to a different one and voila I get a signal that is looking similar but way shorter:

`244,340,536,336,544,332,556,580,288,304,568,596,272,604,276,588,272,312,572,596,272,600,268,328,536,640,236,636,240,336,544,616,5492,5236,560,604,264,320,560,608,264,324,556,304,568,308,572,588,272,600,276,312,568,592,272,308,572,596,272,300,584,584,280,304,580,580,284,304,576,296,576,296,580,588,276,308,576,580,288,588,280,592,280,304,580,580,288,588,284,300,580,580,284,580,300,292,584,576,5520,5208,612,552,316,272,604,560,312,268,604,268,608,268,612,544,324,556,312,268,604,564,308,276,604,556,312,272,600,564,308,276,600,560,316,268,604,268,604,272,600,564,312,268,604,568,300,572,304,568,304,284,596,556,320,560,304,280,596,564,312,560,312,272,600,564,5540,5188,628,536,332,252,624,540,328,256,620,252,620,252,612,560,312,560,316,268,604,556,320,264,608,556,316,268,608,556,316,268,616,544,328,252,620,252,620,252,624,544,328,248,624,544,324,548,328,544,332,260,600,564,312,564,308,276,600,560,312,564,308,272,600,564,5548,5184,624,540,332,252,620,540,324,272,600,272,600,268,608,564,308,564,308,268,608,556,316,268,604,560,316,264,608,556,324,260,608,556,320,264,608,264,612,260,608,556,320,260,612,556,316,556,316,556,320,260,612,556,316,556,320,264,612,552,320,556,316,264,612,548,5544,5200,612,548,328,256,612,552,324,260,612,260,612,260,636,516,356,520,352,232,636,528,344,240,632,532,344,240,628,536,340,244,628,536,336,248,616,264,608,260,620,548,320,260,620,540,332,540,340,532,336,248,632,532,336,536,340,236,636,532,360,508,364,220,644,528,5552,5176,664,500,368,216,656,512,356,224,644,232,640,232,640,524,348,528,344,236,640,532,328,256,616,548,324,256,620,544,332,252,620,544,328,252,624,248,624,252,628,536,332,248,620,540,344,532,336,540,328,256,620,540,336,540,332,248,628,536,336,540,332,248,624,540,5560,5176,648,520,340,244,628,532,336,248,628,248,624,248,624,540,332,540,332,252,624,540,332,252,624,532,360,220,664,500,368,216,652,512,356,228,648,224,648,228,640,520,352,236,636,524,352,524,336,544,328,252,628,536,332,544,328,252,624,540,332,544,332,248,628,532,5560,5184,628,536,336,248,`
Or as graph:
[image]
vs the remote:
[image]

Is this a possible limitation of the library ?

Martin-Laclaustra, 2017-09-18:
Change RCSWITCH_MAX_CHANGES (in RCSwitch.h, line 61) to a value greater than your number of bits (sync + data) * 2. That is 66 * 2 = 132. Set it to 150.
Good luck. Please report back. If this proves to be useful to you I will set a pull request to add my code to the main library.

pree, 2017-09-18:
Hey Martin!

When I try to send arbitrary timings with the ESP-8266 its just crashing on me.
Using this code:
[image]
produces this error:
[image]

Uploading the same code to an Arduino Uno is working fine. Its sending and gets received by the Screen! Little funny thing, its sending a different button than I recorded, gonna try rerecording the buttons and sending them.

pree, 2017-09-18:
Looking around delayMicroseconds seems to be the Problem: https://github.com/esp8266/Arduino/issues/2866

adhambadr, 2017-09-19:
@pree great work man. I also have an AC114 projector screen and would love to get it working on the arduino. I tried Martin's branch with your suggested settings (protocol, width, repeats) but I cant listen to anything, and tried sending your suggested timings as well as martins nothing, would you mind sharing your "successfully" relaying code so maybe I can contribute in a final solution.

pree, 2017-09-19:
Sure @adhambadr

I recorded the Signals of the Remote with https://github.com/sui77/SimpleRcScanner 
and plotted them on the website to check if the signal was recorded correctly.

After that I take the 132 timings corrosponding to that signal and normalize the data (the first two are 5220 (sync-bits) the rest is either 580 or 290).

Then I put those 132 normalized timings in an unsigned int array with a trailing 0 like this:

`unsigned int data[] = {5220, 5220, 580, 290, ...., 580, 0};`

Importing Martin's branch an editing RCSwitch.h, replacing RCSWITCH_MAX_CHANGES to 150.

And with `mySwitch.send(data)` I can send the right data and my screen receives the signal.

Full Code (Data is for the UP-Key):

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

unsigned int data[]   = {5220,5220,580,580,290,290,580,580,290,290,580,290,580,290,580,580,290,580,290,290,580,580,290,290,580,580,290,290,580,580,290,290,580,580,290,290,580,290,580,290,580,580,290,290,580,580,290,580,290,580,290,290,580,580,290,580,290,290,580,580,290,580,290,290,580,580,290,290,580,290,580,290,580,290,580,290,580,290,580,290,580,580,290,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,290,580,580,290,290,580,580,290,580,290,580,290,580,290,580,290,290,580,290,580,580,290,290,580,580,290,580, 0};

void setup() {
  mySwitch.enableTransmit(10);  
}

void loop() {
  mySwitch.send(data);
  delay(5000);
}
```

closed 2017-10-02 by pree

giotms, 2018-01-12:
Hi. I've got a a AC114 Remote. 
I tried https://github.com/Martin-Laclaustra clone, using protocollessreceiver branch with 
 
  { 150, { 34,  3 }, {  1,  3 }, {  3,  1 }, false },    // protocol 7 (AC114)
  { 360, { 13,  4 }, {  1,  2 }, {  2,  1 }, false },    // protocol 8 (DC250)
  { 650, {  1, 10 }, {  1,  2 }, {  2,  1 }, true }      // protocol 9 (Mandolyn/Lidl TR-5

in RCSwitch.cpp 

But it doesn't work. Any suggestion? 
Now I'm interested in the decoding of RF signal - not cloning protocol trasmitting.
I want just receive the pressed button.

Martin-Laclaustra, 2018-01-12:
Please use this:
https://github.com/sui77/SimpleRcScanner
... and if that does not clarify, open a new issue with your particular problem and full information (paste the timings, not the image).
As you can read above, your remote probably emits 64 bits of data and can not be recognized by my branch without previously editing the code.

paperadio, 2018-06-22:
@pree I don't know how to normalize the data to get '580' and '290', I'm confused in lots of google results. which tool did u use? excel or other methods? thx

pree, 2018-06-26:
It's the around the average of the high and low timings.

kuchod, 2018-06-26:
in fact, there is no need to look for an exact value. 

take a look at the signal, the sampling process will induce errors, so you can try to normalize high values close to 600 to 600, and low values close to 300, to 300. then you send the normalized signal, capture it and compare to the original. adjust the values and repeat.

paperadio, 2018-06-27:
@pree @kuchod thanks for all. I didn't normalize,  just copy the sequence captured by scanner, and everything goes well, maybe it has some risk,  I'll try to normalize later. I commented a few details in #237 .

neodevit, 2019-01-20:
is possible have a 8266 code for command up and down my projector monitor plate
in this post i think you try to make my same project my projector monitor plate use remote AC114 
thanks

pree, 2019-01-20:
Hello @neodevit yes it is possible to run this on an esp8266, thats exactly what I did and it is working like a charm.

ivnvsd, 2020-11-07:
> 
> 
> Change RCSWITCH_MAX_CHANGES (in RCSwitch.h, line 61) to a value greater than your number of bits (sync + data) * 2. That is 66 * 2 = 132. Set it to 150.
> Good luck. Please report back. If this proves to be useful to you I will set a pull request to add my code to the main library.

I just described the same experience I had here https://github.com/sui77/rc-switch/pull/170#issuecomment-723478011 RCSWITCH_MAX_CHANGES to 167 helped in my case.

MaxESP, 2022-06-30:
Thanks to Martin for 64 bit changes

I have a similar remote for sceen up and down with pree it's a EM78P153SNJ  , I have follow Martin work 
I have done this 

1 simple Scanner
204,609,205,26,610,206,610,609,206,210,612,207,612,610,206,612,206,613,206,613,206,209,612,610,206,210,614,207,611,610,206,612,206,210,613,206,612,610,206,210,613,207,612,207,611,207,613,207,612,206,612,207,612,609,206,210,614,207,612,207,612,207,612,207,612,207,613,207,611,207,611,208,614,206,614,207,613,206,612,609,206,210,612,610,206,612,206,210,613,609,206,210,613,609,206,209,612,206,612,609,206,612,206,613,5066,5050,600,621,205,212,604,618,198,216,617,202,604,203,609,608,231,611,205,209,611,206,613,206,612,610,206,209,612,206,610,207,614,206,612,610,206,210,613,206,611,610,206,613,206,612,206,613,206,209,612,610,206,210,614,206,612,609,206,612,206,210,612,206,611,611,206,209,616,205,611,207,611,207,613,207,611,207,611,207,611,610,206,210,614,206,612,207,612,206,612,207,613,207,612,206,611,207,611,208,613,207,613,207,611,207,612,609,206,210,614,608,206,613,206,209,613,609,206,210,612,609,206,210,613,207,612,609,205,613,206,612,5086,5043,593,619,198,224,601,606,229,182,635,206,610,206,610,609,206,613,206,209,614,207,610,207,611,609,206,209,614,206,611,206,611,207,613,610,206,210,613,207,611,610,206,613,205,613,206,612,206,209,613,611,206,209,613,206,612,609,206,612,206,209,612,207,611,609,206,210,614,207,611,207,611,207,613,207,612,206,612,206,611,609,206,211,614,206,613,207,613,206,611,207,611,207,612,207,611,207,612,207,612,207,612,207,613,207,611,610,206,209,612,609,206,613,206,209,614,612,204,209,612,609,206,209,614,206,611,609,206,613,206,611,5068,5049,601,621,205,213,602,616,197,217,617,201,604,204,609,608,230,611,205,210,612,206,611,206,611,609,206,209,613,206,612,207,611,207,611,610,207,210,614,207,612,609,206,612,206,612,206,613,205,210,612,610,206,209,612,207,611,610,206,612,206,210,613,206,611,609,206,210,613,207,612,207,613,206,611,207,611,207,613,206,613,609,206,210,613,207,611,207,613,207,610,207,612,207,612,206,613,207,611,208,611,207,612,207,613,207,612,609,206,209,612,610,206,612,206,210,615,609,206,210,611,609,206,210,613,206,612,609,206,612,206,613,

2 decode 
{ 270, { 18,  1 }, {  1,  2 }, {  2,  1 }, true } ,  protocol 14 i have made

3 advance receive .ino with STX 882

Decimal: -964714063595500 (65Bit) Binary: 01010001100010000100111101001100100000001000000000010010001101100 Tri-State: not applicable PulseLength: 279 microseconds Protocol: 14
Raw data: 5040,661,565,244,159,657,576,236,170,649,174,639,183,632,596,220,596,226,183,639,183,632,189,625,603,213,192,631,190,624,197,620,199,619,610,207,199,625,196,620,606,209,607,213,601,219,598,220,186,640,593,219,187,634,188,628,599,215,601,219,188,634,186,629,600,217,192,629,193,623,198,618,201,615,206,610,209,608,212,605,617,206,201,622,199,618,202,613,207,609,212,604,217,601,215,602,218,602,220,598,220,598,623,195,211,613,206,611,611,208,197,626,194,625,199,616,609,210,605,214,192,630,599,214,602,217,189,633,188,625,601,

When I test with CC1101 ebyte advance receive ino 

Decimal: 16786540 (65Bit) Binary: 00000000000000000000000000000000000000001000000000000000100000000 Tri-State: not applicable PulseLength: 281 microseconds Protocol: 14
Raw data: 5067,618,596,199,223,602,606,229,182,636,206,610,206,610,608,206,613,207,209,613,207,611,206,613,609,206,209,614,206,612,207,611,207,611,610,207,210,614,207,612,609,206,612,207,612,206,613,206,209,613,610,206,210,613,207,612,609,206,613,206,210,612,207,612,609,206,210,613,207,613,208,611,207,613,207,612,206,613,206,613,609,206,210,613,207,612,206,614,207,611,207,613,206,612,207,611,207,614,207,612,207,612,610,206,210,613,207,612,610,206,209,613,206,612,207,613,610,206,614,206,209,613,609,206,612,206,210,613,206,613,608,

stop button
Decimal: 16786283 (65Bit) Binary: 00000000000000000000000000000000000000001000000000000000100000000 Tri-State: not applicable PulseLength: 280 microseconds Protocol: 14
Raw data: 5085,597,622,201,205,609,610,230,207,587,230,585,231,610,612,208,586,229,207,615,206,612,207,609,606,207,210,610,208,613,207,611,207,612,586,231,208,610,207,609,612,207,617,207,591,230,607,205,210,613,608,203,209,619,210,591,634,184,622,206,211,613,206,613,614,206,209,614,154,655,207,607,206,616,210,609,182,642,209,617,590,233,210,590,235,585,194,635,148,660,198,617,224,602,199,607,177,653,205,618,127,680,612,259,146,612,210,644,129,682,586,235,588,218,190,662,575,198,612,199,223,605,629,206,182,611,607,230,610,206,611,

i am stuck at this place i known i am close but need help