sui77/rc-switch #165: 20 bits protocol

Flavien06, 2017-09-10:
Hi

My remote use 20 bits protocol #103 [https://github.com/sui77/rc-switch/issues/103#issuecomment-328281528](https://github.com/sui77/rc-switch/issues/103#issuecomment-328281528l)

`GENERAL
=======
Received 210
Bitlength 20
Protocol 0
Delay(pulse length in us) 641
(delay provided from rc-switch)

STATISTICS
==========
Duration of data bits in pulse train :  38447
Data bit duration = 1922 and standard deviation = 14.18
Coefficient of variation of data-bit duration = 0.74 % (should be less than 10%)
Do not use the rest of the information if big coefficient of variation
Long-to-short duration ratio for data bits (rounded) = 2
Sync bit (in multiples of the pulseLength) = 115 1

Proposed protocol for RCswitch
{ 641, { 115, 1 }, { 1, 2 }, { 2, 1 }, true }

STATISTICS OF VARIATION BY LEVELS
=================================
(Differences here are probably artifacts from signal creation or detection)
(This might be completely ignored, but pay attention to them if big differences are present and emission does not work)
number of bits set (in 210): 4
longup, longdown, shortup, shortdown
1286, 1308, 610, 636
this might be useful for tweaking emmitting algorithms

=============================================

RAW SIGNAL (us)
===============
first level down
73811 606
645 1298 621 1301 625 1296 632 1293 642 1303 623 1292 634 1291 634 1292
646 1289 637 1289 647 1275 635 1289 1311 614 1298 616 636 1245 1285 615
649 1274 641 1272 1340 598 636 1288

=============================================`

But default is 24 bits.
How to change the bit-length?
Thank for help

Martin-Laclaustra, 2017-09-24:
You provide the number of bits in the send function:
```
mySwitch.send(code, 24);
mySwitch.send(code, 20);
```
If you are using [433Utils](https://github.com/ninjablocks/433Utils), edit line [54](https://github.com/ninjablocks/433Utils/blob/b858d9bc6b555439f721a2092051ec531db25135/RPi_utils/codesend.cpp#L54)

Flavien06, 2017-09-27:
Thank you
It works.

closed 2017-09-27 by Flavien06

ckiw, 2017-10-28:
Hey @Flavien06 I'm trying to work with the same Mandolyn/Lidl TR-502MSV with RC-710

But my remote isn't working so I don't know what codes to use. I've added the protocol in rc-switch.

Can you help me with that by giving me the codes you use ?
Also can you explain me how to use the teach-in button ? (Have I to just push the button once and then send the codes. Or push the button during the code sends)

Thank you

Flavien06, 2017-10-28:
For me code is : 
(Not sure if it works with yours)
`code=255  # 1 OFF
code=10   # 1 ON

code=130  # 2 OFF
code=147  # 2 ON

code=65   # 3 OFF
code=80   # 3 ON

code=195  # 4 OFF
code=210  # 4 OFF

code=225  # ALL OFF
code=240  # ALL ON

code=201  # +
code=216  # -

protocol=10
sudo ./codesend $code $protocol `

If the remote control does not work anymore you can try it one by one!

`i=0
while true ; do 
  echo "Test code n° : $i"  
  ./codesend $i 10  
  i=$(($i + 1))
  sleep 1s
done`

The teach-in button is only used to pair remote control with electrical socket.

ckiw, 2017-10-28:
I get some result (on and off) with` { 641, { 115, 1 }, { 1, 2 }, { 2, 1 }, true }` in rc-switch
And 210 and 195 as codes for on and off.
What other codes can I use ?
Do you know how to control the dimmer ?