sui77/rc-switch #167: Not receiving anything with Hunter fan remote IN2TX41.

avias82, 2017-09-12:
I have a Arduino Uno connected to a 434MHz receiver as indicated in the Wiki Receiver example.  I tried using both the ReceiveDemo_Simple.ino and ReceiveDemo_Advanced.ino sketches.  I get absolutely nothing out of the serial monitor when sending signals with my Hunter IN2TX41 remote.  I do not have an o-scope, but I did put a DMM on the data line out of the receiver and I can see voltage changes on the pin.  

I put some Serial.print statements in various places and can get those print statements on the serial monitor.  I put a print statement in the mySwitch.available() if statement and never get that print statement.  My code never gets into the mySwitch.available() statement.  I assumed that even if the protocol was not supported I would get the "Unknown encoding" print statement, but I get nothing.

16 void loop() { 
17   if (mySwitch.available()) { 
18      Serial.print("Switch available");
19     int value = mySwitch.getReceivedValue(); 
20      
21     if (value == 0) { 
22       Serial.print("Unknown encoding"); 
23     } else { 
24       Serial.print("Received "); 
25       Serial.print( mySwitch.getReceivedValue() ); 
26       Serial.print(" / "); 
27       Serial.print( mySwitch.getReceivedBitlength() ); 
28       Serial.print("bit "); 
29       Serial.print("Protocol: "); 
30       Serial.println( mySwitch.getReceivedProtocol() ); 
31     } 
32 
 
33     mySwitch.resetAvailable(); 
34   } 
35 } 
 
I have read through the various "Issues" and have not found anything that seemed to help.  I looked at issues 103 and 112.  I tried Martin-Laclaustra's  protocollessreceiver branch.  I see the "SimpleRcScanner" project, but that just looks like a tool for plotting data.  I can not figure out how to capture the data that tool wants.  It looks to me that you have to get at least raw data out of ReceiveDemo_Simple.ino or ReceiveDemo_Advanced.ino and I do not get anything.

finity69x2, 2018-01-27:
I have exactly the same issue with exactly the same remote. I also have a hunter fan remote IN2TX36 remote that acts the same way.

I know the hardware works as I've loaded the send & receive demos on 2 different NodeMCU boards and I see activity in the serial monitor.

The remotes are 433.92mHz and they are learned by the broadlink universal remote devices.

Is there any way to add the functionality to the library?

kuchod, 2018-01-27:
both ReceiveDemo_Simple.ino or ReceiveDemo_Advanced.ino decode the data received as defined in the supported protocols. 

what you need is the output of SimpleRcScanner. this is the info you can plot and analyze to create the new protocol or replay it raw.

finity69x2, 2018-01-29:
I used the SimpleRcScanner and all I seem to get is noise.

I'm using a NodeMCU and using pin D4 as the receiver pin. Is that correct? 

I never get any zero's in the serial monitor so I don't know where my remote signal starts or stops.

nwesoccer, 2018-02-03:
Any progress on this?  I'm actually trying to do the same thing with the NodeMCU and Hunter IN2TX36 remote, but getting the same results as everyone else.

finity69x2, 2018-02-04:
@nwesoccer 

Not really. I'm beating my head against a wall trying to get these remotes decoded.

I've posted in a few other threads here and have gotten a smidgen of help but nothing to point me in a definite direction.

I'm shocked honestly that no one has gotten these remotes supported yet. they are ubiquitous in the US. the fan manufacturer's products are sold in almost every big box/hardware store that I know of here.

If I make any significant progress I'll post back here to give the information.

nwesoccer, 2018-04-17:
@finity69x2 Sounds good.  I haven't had much time to spend on it, but have done some research and am also fairly surprised no one has these remotes decoded yet.  I purchased 4 of these fans to replace old ones in the house, thinking that it would be easy to get automation working with it.

nwesoccer, 2018-04-22:
@finity69x2 @avias82 Spent some time this weekend trying to get things to work.  I was not able to get rc-switch to work, but I was able to get recording and playback of the codes to work using https://github.com/sillyfrog/RFreplayESP.  I did struggle to get clean signals, but finally found the key.  I had to modify recording code slightly so that it recorded after I push a button.  The steps I took were to click and hold the button on the hunter remote, push the button to record, then immediately let go of the hunter button.  This seemed to get the cleanest signal, works every time after figuring this out.  If I don't do it this way, I record interference/junk.  I'm now able to control all the functionality (light toggle, fan off, fan low, fan medium, and fan high) for 3 different ceiling fans.  I just saved a signal file per action per fan and use those to playback the signals.

I've include a gist of my test code, depending on what arduino you use the pins may need to change.  I'm using D1 to receive, D2 to transmit, D4 to start my recording, D5 to rotate through the array of functions, and D6 to replay the signal.  This code is used for one room, change the names and you can record for a new room.  The signal files stay on the device, but can easily be read off the device to Serial output.  I used the arduino ide to include the Automaton library for the buttons releases, and I took the RF433.h and RF433.cpp from the RFreplayESP repository and placed them in my arduino libraries folder.

This is a very rough application.  My plan is to get this all running on my ESP8266 along with wifi and mqtt so that everything can be controlled from Home Assistant, and therefor Alexa.  Let me know if this does/does not work for you.

https://gist.github.com/nwesoccer/e42dba77300d639272d300f48127c1c6

greiginsydney, 2020-01-03:
@finity69x2 @avias82 @nwesoccer 

Oh look, I've fallen into the Hunter RF codes trap... Hey everyone, glad to see I'm not alone here.

Have any of you made any more progress on this since the last post here?

I'm about to order an ESP8266 as that seems to be the only way I'm going to be able to sniff and reproduce these codes, but I thought I'd check in case anyone's made progress anywhere else.

Thanks.

MacMan02, 2022-02-16:
after 2 years Have any of you made any more progress on this since the last post here??

greiginsydney, 2022-02-16:
Hi @MacMan02,

There's been no progress here, as we've since sold that home and moved.

Coincidentally however, we've just installed 5 new Matthews/Atlas (depending on where you live) "Irene" DC ceiling fans, all of which only have basic RF control, so I'll be resuming my own quest for some integration, at least initially with Home Assistant.

\-G.