sui77/rc-switch #168: Protocol for Kambrook power sockets

mpember, 2017-10-01:
I have a few sockets that appear to be a different protocol to most of the code I've seen.  Currently, the only way I can send the commands is using a C-based program I found here:
http://www.openmediacentre.com.au/tutorials/raspberry-pi/home-automation-with-the-raspberry-pi/

Thanks to the code posted by @Martin-Laclaustra over on another thread, I was able to get a scan of the signal (I think).

The output of the On / Off codes is seen here:
https://cloud.pembo.it/index.php/s/GkhlQdOUyCgVAjb
https://cloud.pembo.it/index.php/s/0aaIartRNRapq8g

Does this make sense to anyone?  Is it possible to create a protocol for this type of code?

Martin-Laclaustra, 2017-10-02:
1. Paste the numbers instead of pictures.
2. Record "silence". You might be in a RF "polluted" environment. Make sure that you see silence with a low level most of the time. Your reciever could also be failing.
3. Record longer signals (make the buffer bigger in the code). Press your remote a shorter period so you can see silence before and after the signals.

Best luck.

mpember, 2017-10-02:
> Paste the numbers instead of pictures.

The output from the two codes are:
`97,208,88,194,88,214,86,213,85,188,84,206,52,213,97,184,62,193,53,196,52,203,52,208,85,198,85,207,100,196,99,180,88,192,74,183,94,177,90,209,87,208,53,220,55,214,98,202,86,175,71,196,52,196,99,190,86,187,85,216,85,208,85,174,100,199,86,200,89,189,92,250,47,204,108,187,85,200,92,183,97,177,107,162,70,186,87,177,98,180,86,185,70,186,85,185,109,175,104,174,109,181,74,178,73,184,97,205,87,190,71,193,95,179,72,209,72,178,70,169,60,187,70,200,59,202,70,181,100,180,74,175,103,192,72,184,90,189,90,185,52,185,85,189,85,236,87,198,85,195,71,203,52,196,54,220,86,185,101,294,61,252,72,206,92,217,52,242,87,222,708,30,323,28,592,29,812,28,625,28,623,35,684,275,30,346,29,297,28,331,35,328,36,642,31,346,28,270,36,302,35,273,41,291,29,268,35,309,35,303,35,386,294,35,336,277,35,356,35,518,29,275,38,317,28,281,35,280,41,366,28,269,36,311,35,730,29,276,28,324,30,322,35,297,34,634,35,304,28,377,341,262,29,380,28,320,35,305,35,324,35,329,35,481,30,638,36,292,35,324,36,340,29,489,29,299,38,377,43,254,43,330,42,257,51,257,52,219,158,197,97,172,95,172,130,52,201,56,182,106,49,168,35638,246,88,261,58,245,53,234,87,262,53,222,60,264,43,214,86,232,43,260,54,190,52,207,70,229,53,194,52,219,42,240,94,192,71,204,56,212,53,223,53,220,74,233,86,217,70,215,52,231,51,207,53,205,52,271,52,228,71,209,51,198,71,235,51,237,71,215,53,204,51,198,52,212,71,201,54,234,69,200,60,229,55,227,43,226,84,226,53,221,87,251,41,218,52,216,60,223,43,218,53,245,42,223,53,230,61,208,43,230,53,217,70,275,42,213,60,211,86,221,87,208,52,243,52,214,70,226,76,201,89,197,72,221,53,213,61,240,53,213,53,236,44,224,52,225,70,209,51,192,52,214,52,207,85,199,52,252,72,241,53,214,74,198,86,226,85,210,99,268,54,226,88,214,60,191,58,206,85,206,87,230,53,208,88,204,85,201,55,201,71,193,57,213,53,249,85,208,42,219,52,210,52,208,53,214,85,214,53,176,85,207,52,214,53,211,85,208,85,198,53,230,70,218,52,246,75,216,54,201,70,222,70,220,86,203,90,250,53,233,44,238,42,207,52,211,52,201,51,254,52,236,51,218,86,246,56,210,52,223,43,222,52,211,42,210,72,213,92,233,51,209,72,216,53,211,85,188,52,214,52,219,52,261,54,230,52,233,53,219,53,217,54,220,53,223,55,210,51,204,86,235,86,205,54,201,70,190,89,233,61,211,54,189,70,205,91,186,52,222,71,226,91,212,53,196,86,214,85,202,52,205,87,197,85,187,98,199,85,251,43,249,72,215,86,213,85,220,71,209,89,232,71,210,49,300,42,225,52,236,52,234,51,209,51,253,42,11392,29,637,1659,29,497,343,35,691,30,823,29,581,28,578,1196,28,1064,28,963,29,1349,3519,576,1045,317,52,300,88,251,62,222,104,217,53,220,71,194,54,213,73,227,73,201,86,227,43,187,85,200,85,193,51,237,88,200,71,227,44,212,86,179,52,177,71,199,88,181,71,198,96,179,87,189,54,186,86,211,89,229,86,192,90,189,90,218,70,250,43,209,85,184,52,199,85,203,91,193,94,194,89,214,74,210,90,188,98,180,91,215,73,208,109,219,52,183,60,196,71,215,88,188,52,193,51,201,87,213,85,193,97,224,87,219,71,225,71,199,52,222,52,207,54,239,53,247,100,198,54,234,52,213,52,211,85,201,86,200,52,223,52,191,52,201,97,205,70,181,84,202,72,222,70,199,54,194,53,186,52,207,71,217,97,201,88,203,86,193,52,194,85,229,89,196,53,211,99,205,91,198,89,296,54,207,52,194,93,184,52,188,86,193,85,181,86,185,91,189,73,189,54,189,96,190,58,202,89,204,88,188,90,185,73,205,52,200,70,178,71,189,86,218,72,217,86,208,86,216,95,195,95,174,52,225,84,229,55,209,86,243,53,213,94,192,71,204,52,193,85,201,90,202,139,257,85,218,84,211,89,183,53,207,53,230,72,208,52,215,43,181,73,183,85,214,52,188,74,197,85,229,45,210,87,231,52,186,85,190,86,201,52,222,58,231,61,191,72,248,76,183,108,214,85,212,71,201,73,204,70,198,87`

AND

`180,79,186,75,187,79,177,71,179,68,183,83,176,70,194,74,196,78,174,91,164,75,190,80,178,84,172,78,223,87,179,71,201,72,182,69,184,74,187,83,186,73,188,74,190,83,185,89,197,87,198,83,193,83,185,73,197,80,171,71,188,57,206,63,201,69,209,78,214,74,171,90,190,80,181,76,193,70,191,90,162,89,162,81,163,67,171,86,195,71,176,85,170,85,166,89,168,79,193,76,192,79,184,63,188,75,171,77,202,75,179,73,181,90,169,79,168,72,160,55,168,80,177,73,175,87,206,74,180,90,181,71,165,88,176,89,183,91,154,90,171,91,165,96,211,91,185,32326,250,60,238,68,229,77,225,70,216,62,230,58,256,48,254,77,223,47,247,69,234,58,217,63,260,53,254,59,229,71,230,68,213,64,216,56,237,71,218,77,214,71,204,70,206,67,222,55,238,66,229,82,217,74,273,61,228,63,214,71,208,75,230,71,233,64,229,71,223,78,217,63,223,66,201,72,204,61,225,64,238,43,207,63,210,61,235,57,211,70,195,65,217,68,194,60,208,62,190,71,206,62,192,72,191,66,211,65,234,72,253,51,217,71,229,53,213,72,209,49,228,69,204,104,224,73,229,56,225,57,220,71,206,65,207,70,206,70,187,59,226,63,211,68,192,67,214,68,223,62,245,70,284,58,249,70,225,60,232,62,217,73,205,72,209,62,232,74,224,71,193,72,202,69,210,80,201,64,206,64,198,66,219,73,207,77,194,65,199,71,200,65,223,73,204,83,188,77,202,69,198,80,199,62,212,76,192,83,206,84,204,87,202,72,207,67,188,60,222,66,192,69,258,247,74,189,70,208,74,218,78,209,72,217,73,204,68,205,70,214,64,200,68,206,63,207,77,223,76,221,62,193,67,192,81,195,65,229,67,225,61,213,69,203,93,189,54,213,65,219,65,224,79,205,73,212,65,216,76,215,55,242,66,216,73,204,83,213,71,203,73,212,64,225,56,232,77,200,59,208,72,206,79,182,67,185,71,200,72,219,78,207,56,216,83,199,74,194,67,219,66,219,83,206,77,199,67,183,83,180,73,209,73,193,66,203,65,201,61,220,79,217,77,203,96,174,74,186,71,199,73,202,82,213,77,188,73,195,70,198,72,170,59,190,74,190,102,208,82,204,75,199,83,180,75,209,70,211,85,200,71,192,72,213,67,195,72,19 [... cut ...]
Is there any reason I shouldn't temporarily jumper the GPIO pin being used to send the code to the GPIO pin being used to receive the code?

Does the reference in [the above link](http://www.openmediacentre.com.au/tutorials/raspberry-pi/home-automation-with-the-raspberry-pi/) offer any guides on how best to replicate the protocol?

>A digital zero: is an on pulse of 280ms followed with an off pulse for 300ms
A digital one: is an on pulse of 650ms followed with an off pulse for 300ms
The gap in-between the 5 messages is 9850ms

I got confused when the protocol definitions started talking about 'sync_high' and 'sync_low' parameters.

Martin-Laclaustra, 2017-10-05:
The protocol seems simple:
{ 290, { 1, 34 }, { 1, 1 }, { 2, 1 }, false }
The problem is that rcswitch currently has a limit of 32 bits if you do not change the length of the variables in the library (and in 433utils, if you use them) that hold the code to send.
According to your link, every signal is pre-pended:
01010101000000000011011001101000
(32bits)
then you have to insert the codes of your sockets (what is the length?)
and the append:
11111111
(8bits)
Only the fixed bit amount for 40bits.
Is there a particular reason why you want to use rcswitch?

mpember, 2017-10-09:
> Only the fixed bits amount for 40bits.
The fixed part is that all codes start with "11111111" and end with "01010101".  The full code is 48 bits long. The rest defines the 'group', 'socket' and 'state' being sent.

e.g.
The full signal transmitted for a ON command would look like this:
01010101000000000011011001101000001000**01**11111111 **(93459401155071)**
01010101000000000011011001101000001000**10**11111111 **(93459401155327)**

> Is there a particular reason why you want to use rcswitch?
I would like to reduce the need to have my code use two different ways of transmitting the RF codes.  Since there is a python-based system that appears to accept the same 'protocols' as rcswitch, I was hoping to remove the need for an external binary.

The python port of rc-switch I am using can be found here:
https://github.com/milaq/rpi-rf

It works fine with the sockets I own that support 'protocol 1'.

Martin-Laclaustra, 2017-10-10:
Suggested changes to RCSwitch to allow for very long codes (>32 bits)

**RCSwitch.h**
line 80
`void send(unsigned long code, unsigned int length);`

`unsigned long` to `unsigned long long`

**RCSwitch.cpp**
line 486
`void RCSwitch::send(unsigned long code, unsigned int length) {`

`unsigned long` to `unsigned long long`

line 500
`if (code & (1L << i))`

`1L` to `1LL`

I did not test these changes. Please report back if you take on this path.

mpember, 2017-10-15:
@Martin-Laclaustra 

I had an attempt.  The scanner never displayed any output.other than a screen full of 0s.

Since the reason I started down this path was to try and replace an executable with a python equivalent, I have decided to simply replace the few Kambrook sockets that use the unsupported protocol.

I appreciate all the assistance.  In the end, it will be easier to give the Kambrook sockets to a mate who already has a number of them.  Since he is not using a python-based setup, the advantage of an all-Python codebase isn't applicable.

closed 2017-10-15 by mpember

Martin-Laclaustra, 2017-10-18:
Please reopen the issue to be able to dicuss here.

Updated protocol according to your recording that you posted in the other thread.

{ 360, { 1, 1000 }, { 1, 1 }, { 2, 1 }, false }

Beware of 1000, try smaller values (your recorded signal showed variable gaps > 3000 * 360). You will not be able to DETECT this protocol but it should be easy to send it.
Modify the size of the variables as explained above.
Use this updated protocol.
If this does not trigger your plug, record the signal that you generate with rcswitch and compare it with the signal that actually does trigger the plug.

The 3 signals you recorded were not exactly equal. The first two signals were 47 bits and the last one 48 bits (your OFF signal: 010101010000000000110110011010000010001011111111; the first signals missed the first 0).

mpember, 2017-10-19:
I appreciate the help.  I have already given away my Kambrook sockets. As such, my only chance to test your code will be when I assist my mate with setting them up on his Pi.  I'll see how I go.  It will depend on when we are both free.  I'll keep you posted.

Thanks.