sui77/rc-switch #169: My remote sends double signals (24 and 28 bit)

ronnyandre, 2017-10-03:
I have a Luxorparts RF remote that sends two signals to turn on/off my remote device. First it sends a 28 bit decimal (tri-state not applicable) on ~505 ms, and then it sends 24 bit decimal with tri-state on 323 ms.

However, none of the examples provided works with my device. RFXCOM manages to control it, so I know it should be able to control.

Anyone with this kind of experience?

`Decimal: 38372374 (28 Bit)
Binary: 0010010010011000010000010110
Tri-State: not applicable
PulseLength: 505 microseconds
Protocol: 2
Raw data: 5064,260,1056,260,1056,924,404,256,1064,256,1052,924,388,276,1040,276,1044,928,400,264,1048,268,1056,196,320,196,1348,196,1344,204,324,324,192,1356,192,320,196,1348,196,1352,188,196,1352,196,1352,192,20,312,24,300,220,224,20,64,328,188,1364,

Decimal: 344341 (24 Bit)
Binary: 000001010100000100010101
Tri-State: 00FFF00F0FFF
PulseLength: 323 microseconds
Protocol: 1
Raw data: 10028,260,1040,276,1052,260,1056,264,1052,268,1048,924,400,264,1056,924,396,264,1056,924,392,268,1048,268,1048,268,312,1228,320,1228,316,1240,312,2376,1092,456,1092,448,328,1216,460,1088,460,320,1220,1096,448,324,1220,`

Martin-Laclaustra, 2017-10-04:
If your aim is controlling the device, you could follow:
https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310

rctgamer3, 2019-06-01:
My remote also sends 28bit and 24 bit codes. I don't understand that comment. Could you please understand how to convert decimal/binary/tri-state/raw-state data captured with rc-switch and convert it to something that works? Does my remote only send 28bit with 24bit as a fallback, or is both the 28bit and 24bit sequence required?