sui77/rc-switch #177: Repeat transmissions missed

cimba007, 2017-11-13:
Setup:

Node A: rc-switch sender
https://pastebin.com/dajLZSe3

Node B: rc-switch receiver
https://pastebin.com/RG29GGrs

My Problem is that the sending Node A sends the message **22 times** (checked with oscilloscope).
My Node B only receives ~half of the messages, only **10 received**

```
87656;10;2552
```

Am I doing something terrible wrong or can some1 point me in the right direction?

Both nodes are Arduino ProMini @ 8MHz Int RC-OSC.
Node A uses the defaultish sender from this bundle (https://www.ebay.de/itm/433MHz-RF-Sender-Empfanger-Kit-Funkmodul-Transmitter-fur-Arduino-Raspberry-Pi/112277710415?hash=item1a2445e64f:g:naAAAOSw5cNYg-CN)

Node B usess Receiver from this bundle (https://www.ebay.de/itm/315MHz-433MHz-Wireless-transmitter-receiver-Wireless-module-SYN115-SYN480R/272387133128?hash=item3f6b89cec8:m:m27-PiH0fK7Oi9HdN9Kr-0w)

cimba007, 2017-11-14:
I noticed this problem is not only limited to my setup but is also presenth with off the shelf 433MHz reed contact alarm sensors.

They repeat the signal 22 Times (oscilloscope) but only 10-11 times are received with rc-switch simple sketch.

cimba007, 2017-11-14:
Is there any reason why to drop every second repetition? I use the number of received frames as a quality indicator. Counting as accurate as possible seems to be a good idea but I feel like missing every 2nd frame.

```
      repeatCount++;
      if (repeatCount == 1) {
        for(unsigned int i = 1; i <= numProto; i++) {
          if (receiveProtocol(i, changeCount)) {
            // receive succeeded for protocol i
            break;
          }
        }
        repeatCount = 0;
      }
```

casiobearing, 2023-01-23:
Thought I would bump this issue as it's still present. The (repeatCount == 2) line in the receiver breaks the transmitter setRepeatTransmit() function. You end up receiving only half the frames you sent. In fact if you set an odd number of transmit repeats you get inconsistent frame repeats at the other end.