sui77/rc-switch #180: Timing Issues on ESP32?

zahlex, 2017-12-11:
First of all: Thank you for this amazing library. This code is structured and written so well, it almost makes sense to me, even if I have literally zero experience with C++.

But nevertheless, I'm sadly not able to get the sending part working on my ESP32 board. (Receiving works just fine.) After discovering the codes my remote is using via the ReceiveDemo_Advanced Sketch, I modified the SendDemo Sketch to the following:

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);
  
  // Optional set protocol (default is 1, will work for most outlets)
  mySwitch.setProtocol(1);

  // Optional set pulse length.
  mySwitch.setPulseLength(185);
  
  // Optional set number of transmission repetitions.
  mySwitch.setRepeatTransmit(5);
  
}

void loop() {
  
  mySwitch.send(284099, 24); // ZAP 0311, CH2, ON
  delay(5000);  
  
  mySwitch.send(284108, 24); // ZAP 0311, CH2, OFF
  delay(5000);
    
}
```

And it works like a charm on my Arduino Uno.
After getting that working, I tried to take it one step further and compiled the Sketch for my ESP32.
Result: The Transmitter is working as a radio jammer. I cannot even use my remote, while the ESP32 is running this Sketch. So I assume some timing related things are totally out of control.

I've tried multiple pins and my board is an NodeMCU-32S with an ESP-WROOM-32 Module.
I really hope, somebody has an idea how to solve this issue.
If more information is needed, just let me know. Thanks in advance!

Martin-Laclaustra, 2017-12-13:
Just a hint on where to start looking.
I do not have an ESP32, so this is an informed guess, according to your description.
Check that the function delayMicroseconds actually counts microseconds and no other power-unit. I think early versions of the library for the ESP8266 had this problem.

zahlex, 2017-12-16:
Thank you for your response, @Martin-Laclaustra 
I looked into that and delayMicroseconds seems to be working perfectly fine in the ESP32 version of the library.
Nevertheless, thank you very much for pointing out that this issue existed earlier!

luciocorrea, 2018-07-15:
@zahlex Hi. Did you solve that? Because I'm having similar issues with BLE UART and rc-switch in a Wemos32 board.

bart, 2020-05-05:
@luciocorrea Did you solve it?