sui77/rc-switch #182: Detect release remote button

openscoreboard, 2017-12-14:
I have a remote controler 433mhz with 4 buttons
I can receiber codes in my arduino perfect.
How can detect in my arduino if release the button in the remote controler?
Now if push the button 2 seg the arduino read many codes....

Locke, 2017-12-18:
Hi,
it sounds like you're not having an issue with this library but with using it / your own code.

You need to manually debounce the received values. This is a very small and untested idea, but might point you in the right direction:

```c
unsigned long lastTime = 0;
unsigned long lastValue = 0;

void loop() {
  const unsigned long time = millis();

  if (lastValue != 0 && lastTime != 0 && time >= lastTime + 250) {
    Serial.print("Received ");
    Serial.print( lastValue );
    Serial.println(" 250 milliseconds ago.");

    lastTime = 0;
    lastValue = 0;
  }

  if (mySwitch.available()) {
    Serial.print("Received ");
    Serial.print( mySwitch.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( mySwitch.getReceivedProtocol() );

    lastTime = time;
    lastValue = mySwitch.getReceivedValue();

    mySwitch.resetAvailable();
  }
}
```

IMFROST, 2021-02-06:
I was having the same issue so I used your code as a reference and it worked. Thanks @Locke