sui77/rc-switch #186: Slow detection of the signal

erkobg, 2017-12-25:
Hello,
this is a follow up of : https://github.com/sui77/rc-switch/issues/80
The previous issue was closed before fixed.

So the setup I have:
ESP8266 and 433 receiver (Product Model: WL101-541)

Example codes works, but reaction time - detection of the signal is very slow. 
If I press a button of cheap remote only once - nothing is detected.
I have to press twice or more times or hold the button down in order to get code.

Any ideas how I can increase the detection?

anil8anil, 2019-04-04:
Hi, did you find any solution?

1technophile, 2020-01-05:
Hello,

I have never tested this receiver. Could you try with another one like [these ones](https://docs.openmqttgateway.com/setitup/rf.html#compatible-parts)

silas2222, 2020-03-26:
hello i have same problem still no fix why

1technophile, 2020-03-26:
> hello i have same problem still no fix why

Because nobody is working on it. You are using an open source library, not a paid one with support and company behind ;-)

I suggested to in january to use another receiver, did you tried that?