sui77/rc-switch #189: Silvercrest switch with remote control (LIDL)

zyronman, 2018-01-09:
Dear All,

I'm trying for several days to control a SilverCrest switch using an ESP8266 a MX-FS-03V module and a MX-05V module to record the signals.

[image]

Each button gives me 3 differents signals (exemple button A ON) :
Decimal: 16211228 (24Bit) Binary: 111101110101110100011100 Tri-State: 11F1FF1F0F10 PulseLength: 101 microseconds Protocol: 3
Raw data: 7216,960,576,956,580,952,580,952,1076,444,320,1208,320,1200,1088,444,324,1208,1080,448,316,1204,452,1080,444,320,1212,316,1208,1080,456,308,1212,316,1208,448,316,1212,1076,456,1072,456,1076,444,320,1216,312,444,1092,

Decimal: 15866204 (24Bit) Binary: 111100100001100101011100 Tri-State: not applicable PulseLength: 101 microseconds Protocol: 3
Raw data: 308,1220,1068,448,1080,452,316,1208,320,1216,1072,456,308,316,1216,316,2344,1080,452,1076,452,1076,448,1080,452,1076,456,464,304,1220,1068,456,1072,456,312,1220,1068,456,312,440,1088,952,584,948,592,432,1092,440,1088,

Decimal: 16737708 (24Bit) Binary: 111111110110010110101100 Tri-State: not applicable PulseLength: 101 microseconds Protocol: 3
Raw data: 312,1208,1080,464,304,1212,1212,1080,448,1080,452,312,1208,320,1212,320,2344,1080,452,1076,316,1204,324,1208,1080,444,324,1208,1080,448,316,1212,1080,452,1080,960,572,960,592,428,1092,948,588,948,576,444,1084,448,1096,

As the pulseLength is 101, I changed the RCSwitch.cpp file :
  { 350, {  1, 31 }, {  1,  3 }, {  3,  1 }, false },    // protocol 1
  { 650, {  1, 10 }, {  1,  2 }, {  2,  1 }, false },    // protocol 2
  **{ 101, { 30, 71 }, {  4, 11 }, {  9,  6 }, false },    // protocol 3**
  { 380, {  1,  6 }, {  1,  3 }, {  3,  1 }, false },    // protocol 4
  { 500, {  6, 14 }, {  1,  2 }, {  2,  1 }, false },    // protocol 5
  { 450, { 23,  1 }, {  1,  2 }, {  2,  1 }, true }      // protocol 6 (HT6P20B)

I could reproduce the same signal (Binary, PulseLength, Protocol) checking an comparing it with the receiver module but it doesn't switch on my device.

Do you have any idea of how I can solve it or progress in my goal? I attach a record done with the arduino serial plotter (as I don't have an oscilloscope) just in case it could gives you an idea of the kind of signal.

Many thanks in advance for your help!

[image]

Martin-Laclaustra, 2018-01-11:
Use this to record better information from your remote:
https://github.com/sui77/SimpleRcScanner
You can use the online visualizer or the ReceivedSignalVisualizer.ods file to get better plots.
Also, press your remote buttons longer. At least 3 seconds when recording. You must aim to get a repetitive pattern.
I coded other tools based on RPi to characterize remotes but as you are using ESP8266 I do not include a link here.

zyronman, 2018-01-20:
@Martin-Laclaustra thank you so much for your help!
After getting a good knowledge of the signal thanks to your tools, I finally decided to code the highs and the lows of the signal without using the rc-switch. It works very well!

Thank you again !

Martin-Laclaustra, 2018-01-21:
@zyronman 
Please share your findings here, so other people can benefit from your investigations.
How many bits? What kind of protocol?
You could share the numeric timings here, so we could inspect them and improve rc-switch if necessary.
Also, you can use rc-switch, a version modified by me, to send your custom timings. See here:
https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310

zyronman, 2018-01-23:
@Martin-Laclaustra you're 100% right. I will post everything I have this weekend.
The way I did it will probably not win a beauty contest but it works.

etimou, 2018-01-23:
I am playing with a Chacon switch that looks pretty identical to yours.[https://i2.cdscdn.com/pdt2/6/6/1/1/700x700/cha54661/rw/chacon-kit-de-2-prises-telecommandees-on-off-avec.jpg](url)
I am pretty sure that the protocole is the same.
By looking at the signal from the original remote with an rtl-sdr dongle, it is very confusing. It seems that the code is sent 4 times with a short sync bit and then 4 times with a longer sync bit.
The codes sent with the longer sync bit seem to follow the protocol 5 but also seem to have no effect on the switch.
The codes sent with a shorter sync bit can be interpreted like this:`{ 500, {  1, 4 }, {  1,  2 }, {  2,  1 }, false },    // protocol 5`
(slight modification of protocol 5) and they work for me with an arduino and a transmitter.

witnessmenow, 2018-02-24:
I have a really similar looking remote that I got from Aldi that I am having issues with.

Is there steps on how to use simpleRC or read the output of it? Do I press the button while the countdown is happening or after its finished?

Here is my timings for A on (held down the button while the countdown was finishing)

Would appreciate any pointers

```
584,928,592,924,588,420,1092,424,1088,932,580,932,580,428,1088,928,592,420,1088,928,584,932,580,428,1084,432,1096,2936,7160,924,588,924,588,928,584,932,588,420,1088,932,580,932,580,428,1092,420,1088,932,584,928,584,932,584,928,584,428,1088,420,1100,920,580,936,580,432,1080,936,580,424,1088,928,588,928,584,424,1092,424,1192,300,2336,1048,460,1052,460,1052,448,1060,452,1060,452,1056,452,304,1204,304,1212,1048,452,1060,452,300,1212,296,1220,1044,460,1052,456,300,1208,300,1204,300,1216,1052,456,1052,460,296,1212,1048,456,1056,460,296,1208,300,1216,300,2340,1048,452,1052,460,1052,460,1048,468,1044,456,1052,464,292,1208,300,1212,1048,476,1032,464,296,1208,300,1212,1052,452,1056,460,296,1208,300,1212,296,1212,1052,460,1048,460,300,1208,1052,456,1052,464,292,1216,292,1216,300,2336,1048,464,1048,456,1056,456,1048,468,1044,460,1052,456,300,1208,300,1208,1052,464,1044,468,288,1212,300,1204,1056,464,1048,456,300,1208,300,1208,300,1212,1052,456,1052,452,304,1220,1044,456,1052,460,296,1208,300,1216,300,2336,1052,456,1052,456,1052,468,1040,468,1044,464,1044,464,292,1212,300,1208,1052,456,1052,460,296,1216,296,1208,1052,456,1056,456,300,1208,296,1224,288,1212,1048,472,1040,456,300,1208,1056,452,1052,460,300,1216,292,1216,2940,7156,928,584,932,592,924,584,932,580,932,584,932,580,424,1088,428,1088,928,588,928,584,424,1092,420,1096,924,584,928,588,424,1084,428,1084,432,1088,928,592,920,596,412,1096,920,588,928,588,424,1084,432,1088,2944,7152,932,584,928,600,912,592,928,584,928,588,924,588,424,1100,412,1096,920,592,920,588,424,1084,428,1092,928,592,924,580,428,1088,424,1084,428,1096,924,580,932,588,424,1088,928,584,932,580,428,1088,424,1096,2940,7168,916,592,924,580,936,584,928,580,932,588,924,588,424,1092,424,1088,928,592,924,588,416,1092,420,1092,928,584,928,588,424,1088,424,1096,420,1084,932,592,920,592,420,1096,924,588,928,584,424,1096,416,1088,2948,7148,932,588,928,584,932,580,932,588,928,584,928,580,432,1096,416,1084,936,588,924,588,420,1096,420,1088,928,592,920,596,416,1088,424,1092,424,1092,924,588,924,588,424,1084,932,584,928,588,424,1092,424,1196,296,2336,1048,468,1040,464,1048,456,1056,456,300,1212,1048,456,304,1216,288,1208,1056,460,300,1204,304,

```

zyronman, 2018-02-27:
Dear guys,

I'm really sorry for the delay, I didn't have time to answer before.
I will give you all the results I have. I still don't use the rc-switch library but with my results, I'm sure that someone with more experience will be able to recognize the signal and tell us how to reproduce it with the lib.

As I explained on my first post, I tried to recognize the signal using the ReceiveDemo_Simple sketch. It gave me a 24 bits signal result.
When Martin-Laclaustra gave me the SimpleRcScanner sketch, I stayed focused on the 24 bits but in fact, the signal is composed by several sequences. Some of them are 24 bits and others are 124 bits. I isolated the 124 and it works perfectly.

Complete signal :
592,940,584,952,596,936,600,424,1092,952,584,948,584,952,592,432,1092,952,588,436,1088,952,588,948,588,944,588,436,1092,952,580,444,1092,440,1096,436,1092,952,584,952,584,948,580,444,1096,436,1108,2980,7232,956,580,952,576,960,592,940,580,444,1092,952,580,952,588,948,592,432,1096,948,580,444,1096,944,580,956,588,944,584,440,1092,952,596,428,1088,444,1092,440,1096,948,600,932,592,944,584,440,1092,440,1200,312,2352,1076,452,1076,456,1076,456,1072,464,304,1220,308,1216,1076,460,304,1220,308,1220,308,1216,312,1220,1072,452,1080,464,300,1216,312,1224,1068,456,312,1216,1076,460,304,1216,1076,468,1060,452,1080,472,292,1212,316,1220,316,2352,1076,456,1076,452,1076,456,1076,456,308,1216,312,1212,1080,468,300,1220,308,1212,316,1216,312,1212,1076,456,1076,456,312,1216,312,1216,1072,456,312,1220,1072,452,316,1212,1076,456,1076,452,1076,460,308,1216,312,1220,316,2364,1064,448,1080,456,1076,456,1072,460,308,1212,316,1212,1080,468,296,1216,312,1228,300,1224,304,1212,1080,456,1076,452,312,1212,316,1216,1076,452,316,1220,1068,468,300,1212,1080,456,1072,452,1080,460,308,1216,312,1224,308,2364,1064,460,1072,456,1072,460,1068,464,304,1216,312,1212,1080,448,320,1212,316,1216,312,1232,296,1216,1072,456,1076,448,316,1216,316,1212,1076,456,312,1212,1080,452,312,1220,1072,452,1080,472,1056,452,316,1216,312,1224,2988,7228,956,584,948,580,956,580,952,580,444,1088,448,1088,952,592,432,1092,444,1092,440,1088,444,1084,960,580,952,592,432,1088,444,1092,952,584,440,1088,956,580,440,1092,952,584,952,580,952,588,436,1092,440,1096,2992,7232,952,584,952,580,952,584,952,576,448,1088,444,1108,936,580,444,1096,436,1092,440,1092,440,1092,952,588,944,584,440,1092,444,1088,952,584,440,1092,952,596,428,1092,952,584,948,580,956,580,444,1088,444,1096,2992,7236,948,588,948,580,952,584,952,580,440,1092,444,1092,952,588,432,1096,440,1092,440,1092,440,1088,956,588,944,580,444,1100,432,1088,956,588,436,1088,956,580,444,1088,952,584,952,580,956,584,436,1096,440,1096,2992,7232,952,588,948,580,952,588,944,588,436,1088,444,1104,944,576,448,1088,440,1096,440,1092,440,1088,956,576,956,584,440,1092,440,1088,956,584,440,1092,952,580,440,1088,956,576,960,576,956,576,448,1088,444,1196,316,2352

Important part of the previous signal :
7232,956,580,952,576,960,592,940,580,444,1092,952,580,952,588,948,592,432,1096,948,580,444,1096,944,580,956,588,944,584,440,1092,952,596,428,1088,444,1092,440,1096,948,600,932,592,944,584,440,1092,440,1200,312,2352,1076,452,1076,456,1076,456,1072,464,304,1220,308,1216,1076,460,304,1220,308,1220,308,1216,312,1220,1072,452,1080,464,300,1216,312,1224,1068,456,312,1216,1076,460,304,1216,1076,468,1060,452,1080,472,292,1212,316,1220,316,2352,1076,456,1076,452,1076,456,1076,456,308,1216,312,1212,1080,468,300,1220,308,1212,316,1216,312,1212,1076,456,1076,456,312,1216,312,1216,1072,456,312,1220,1072,452,316,1212,1076,456,1076,452,1076,460,308,1216,312,1220,316,2364,1064,448,1080,456,1076,456,1072,460,308,1212,316,1212,1080,468,296,1216,312,1228,300,1224,304,1212,1080,456,1076,452,312,1212,316,1216,1076,452,316,1220,1068,468,300,1212,1080,456,1072,452,1080,460,308,1216,312,1224,308,2364,1064,460,1072,456,1072,460,1068,464,304,1216,312,1212,1080,448,320,1212,316,1216,312,1232,296,1216,1072,456,1076,448,316,1216,316,1212,1076,456,312,1212,1080,452,312,1220,1072,452,1080,472,1056,452,316,1216,312,1224,2988,7228

If someone wants to investigate and integrate it to the lib, I will be available to provide all the tests and details.

@witnessmenow : I recommand to start pressing the button before the fifth countdown and release it when you're sure the program finished.

Mr-Codesaur, 2019-02-17:
Hi guys! I have same remote sockets from LIDL.
@zyronman Is right. The remote for these sockets is sending 24 bit codes which are ignored. Then there is a sequence of 124 bits.  I have discovered this 124 bit sequence also consists of 24 bit codes + some sync bits. I have written simple [library ](https://github.com/Mr-Codesaur/SilverCrest_433Mhz_Sockets-Arduino) if someone wants to check.

Julien83, 2022-02-25:
Hi @zyronman

Did you find a solution to control a SilverCrest switch.

simonepernice, 2022-03-16:
Hello,
I can control the switch with Arduino + RCSwitch library (updating its protocol list).

The SilverCrest code is made by 2 different protocols. The firs is used for the beginning 24 bits and the second for the ending 32 bits.

The first is difficult to set in RCSwitch, the closer I can compute is: { 186, {  2, 12 }, {  2,  5 }, {  6,  2 }, false }.

The second is exactly: { 500, {  1, 14 }, {  1,  3 }, {  3,  1 }, false }.

To turn it on/off you need to write the following code (after updating RCSwitch with protocols 13 and 14 written above):

[...]
  mySwitch.setProtocol(13);
  mySwitch.send(data[0], 24);
  mySwitch.setProtocol(14);  
  mySwitch.send(data[1], 32); 
[...]

To detect the codes (data[0] and [1]), I used a logical analyzer linked to my remote base band signal. It can be also used a 433MHz receiver (like RXB6) if you do not want to disassemble the remote.

The last step is understand the code logic. I noticed different remotes have different codes. They are store in the not volatile memory because are retained for days without battery.

All remote codes can be computed by one because the switch learns just the on button and then it recognize also the all-on buttons. 

Good luck,
Simone

rasic, 2022-03-29:
Hi @simonepernice,
could you please post here your 4 codes for each action (on/off)? I am trying to figure out how to make a full set of codes when you only know one. Thank you for your response.

rasic, 2023-07-31:
I got it working with my remote but still don't understand the code. From what I remember the code for the button rotates between 4 different codes. My remote and remotes of other people in this thread have 24Bit codes but you seem to have 32Bit. Maybe you have different version? Could you please post output of pushing each button 5 times in this order 5xA_on, 5xA_off, 5xB_on, 5xB_off, 5xC_on, 5xC_off, 5xD_on, 5xD_off, 5xMASTER_on, 5xMASTER_off?

darkbox, 2023-08-01:
### About the remote:
- Model: 8 50 50 00075
- Manufactured: 2022-06

### BUTTON A-ON (x5)
```
Decimal: 618491283 (32Bit) Binary: 00100100110111010110110110010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6932,556,1432,548,1432,1512,484,512,1464,524,1456,1500,492,500,1472,516,1468,1496,488,1480,504,496,1484,1480,512,1464,516,1460,528,472,1496,1472,516,480,1496,1476,508,1464,520,480,1496,1472,512,1464,524,476,1496,1472,516,1460,524,472,1500,492,1488,1476,508,492,1488,500,1480,1480,500,1480,520,

Decimal: 3234986707 (32Bit) Binary: 11000000110100011111101011010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6936,1520,472,1496,496,496,1480,508,1468,520,1464,520,1456,528,1456,528,1448,1512,480,1484,504,492,1484,1484,500,492,1488,504,1476,504,1480,1488,500,1472,504,1468,516,1464,528,1448,528,1452,536,460,1512,1464,520,476,1496,1472,516,1460,524,476,1500,1472,512,484,1488,500,1480,1488,500,1472,516,

Decimal: 618491283 (32Bit) Binary: 00100100110111010110110110010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6932,560,1428,548,1432,1516,476,516,1464,520,1464,1496,488,504,1472,512,1464,1496,492,1480,512,484,1488,1480,504,1472,512,1460,528,468,1504,1468,516,484,1496,1472,508,1468,524,468,1508,1464,516,1460,520,480,1496,1472,516,1460,524,472,1500,492,1484,1480,512,488,1488,504,1472,1488,500,1476,512,

Decimal: 3234986707 (32Bit) Binary: 11000000110100011111101011010011 Tri-State: not applicable PulseLength: 693 microseconds Protocol: 2
Raw data: 6944,1516,476,1496,488,504,1480,508,1468,520,1460,524,1460,524,1456,528,1452,1508,484,1488,496,496,1484,1484,500,492,1484,504,1476,512,1468,1496,496,1476,512,1464,520,1460,528,1448,532,1448,540,456,1512,1468,520,468,1504,1472,516,1460,524,476,1500,1468,516,484,1492,496,1480,1488,504,1468,516,

Decimal: 618491283 (32Bit) Binary: 00100100110111010110110110010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6936,556,1428,552,1432,1520,476,512,1464,524,1456,1504,488,500,1480,512,1464,1504,488,1480,504,492,1488,1476,512,1468,516,1460,528,464,1504,1476,512,480,1492,1480,508,1464,528,468,1500,1476,512,1460,524,472,1504,1472,512,1460,532,468,1504,484,1492,1476,508,492,1484,504,1476,1492,496,1476,516,
```

### BUTTON A-OFF (x5)
```
Decimal: 3359271123 (32Bit) Binary: 11001000001110100110100011010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6944,1520,476,1492,496,496,1480,512,1468,1492,500,496,1480,516,1464,516,1460,528,1456,528,1452,1508,488,1480,504,1468,520,480,1500,1472,516,476,1496,496,1480,1484,508,1468,520,476,1496,1476,516,480,1496,496,1480,508,1472,1496,496,1468,520,480,1492,1480,512,484,1492,500,1476,1488,504,1472,524,

Decimal: 3250624659 (32Bit) Binary: 11000001110000001001100010010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6932,1528,464,1500,492,500,1476,512,1468,520,1460,524,1460,524,1456,1508,480,1488,500,1476,512,480,1496,496,1484,504,1472,516,1464,524,1452,532,1448,1508,488,504,1472,516,1464,1496,492,1480,516,480,1488,500,1476,512,1476,1492,492,496,1480,512,1464,1496,496,496,1480,508,1468,1500,492,1480,512,

Decimal: 3359271123 (32Bit) Binary: 11001000001110100110100011010011 Tri-State: not applicable PulseLength: 693 microseconds Protocol: 2
Raw data: 6952,1516,472,1496,500,496,1480,504,1472,1492,504,492,1480,508,1476,512,1468,520,1456,528,1456,1504,484,1488,508,1464,524,468,1504,1472,516,480,1496,492,1484,1484,504,1468,528,472,1496,1472,520,476,1496,496,1480,512,1472,1488,500,1476,512,480,1492,1476,516,484,1492,496,1484,1484,500,1472,524,

Decimal: 3250624659 (32Bit) Binary: 11000001110000001001100010010011 Tri-State: not applicable PulseLength: 693 microseconds Protocol: 2
Raw data: 6940,1524,472,1492,496,496,1484,508,1468,520,1464,524,1456,528,1452,1504,488,1484,504,1472,520,472,1504,488,1484,508,1472,512,1468,520,1456,532,1456,1500,488,504,1476,516,1464,1492,500,1480,512,480,1492,500,1476,512,1468,1496,496,496,1484,508,1468,1492,500,496,1480,508,1472,1492,504,1472,520,

Decimal: 3359271123 (32Bit) Binary: 11001000001110100110100011010011 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6948,1520,472,1496,500,492,1484,508,1468,1492,500,496,1484,504,1476,512,1464,528,1456,528,1452,1508,488,1484,500,1472,524,468,1504,1472,516,480,1496,492,1488,1484,504,1468,520,476,1496,1476,520,472,1496,500,1480,508,1472,1488,504,1476,516,480,1496,1476,512,480,1496,496,1480,1484,508,1468,520,
```

### BUTTON B-ON (x5)
```
Decimal: 273844117 (32Bit) Binary: 00010000010100101000011110010101 Tri-State: not applicable PulseLength: 694 microseconds Protocol: 2
Raw data: 6936,556,1428,552,1432,548,1432,1520,476,516,1464,524,1456,528,1452,528,1448,540,1448,1508,484,508,1468,1492,500,496,1484,508,1472,1492,492,500,1480,1488,500,492,1484,508,1472,516,1464,520,1460,1504,484,1484,504,1472,516,1456,528,472,1508,484,1492,1480,504,488,1488,1480,512,484,1488,1484,516,

Decimal: 3744201941 (32Bit) Binary: 11011111001010111111110011010101 Tri-State: not applicable PulseLength: 694 microseconds Protocol: 2
Raw data: 6936,1524,472,1496,496,496,1480,1488,504,1468,516,1460,528,1452,528,1448,536,460,1516,476,1496,1476,512,484,1492,1480,508,484,1492,1476,512,1468,516,1460,524,1452,532,1448,536,1444,540,1448,536,1436,548,448,1520,480,1496,1468,520,1460,524,472,1504,1468,520,480,1492,1476,512,484,1492,1476,516,

[... cut ...] 492,1484,1480,504,1468,520,476,1496,496,1480,1484,516,

Decimal: 2189688281 (32Bit) Binary: 10000010100000111111110111011001 Tri-State: not applicable PulseLength: 692 microseconds Protocol: 2
Raw data: 6932,1520,472,520,1456,528,1452,528,1452,532,1452,528,1448,1504,484,512,1464,1492,496,500,1476,512,1468,516,1464,524,1456,528,1452,1500,492,1480,504,1468,516,1460,524,1452,528,1448,536,1444,536,1444,540,460,1512,1460,520,1452,528,1452,532,468,1504,1464,520,1456,528,472,1500,496,1484,1476,512,
```

rasic, 2023-08-01:
@darkbox Your code rotates between 2 different codes. Last four bits will tell you the button and action is cherrypicked from convenient bits based on the button.
```
A                              vvvv--- button code
00100100 11011101 01101101 10010011 3
11000000 11010001 11111010 11010011 

11001000 00111010 01101000 11010011
11000001 11000000 10011000 10010011
                x
B
00010000 01010010 10000111 10010101 5
11011111 00101011 11111100 11010101

00000010 01001100 11110110 11010101
10100110 11011111 01101111 10010101
   x
C
10111010 11011110 11111001 10011011 11
11001011 10101100 10101001 11011011

01001000 10000000 10000001 10011011
11101101 00111110 00100000 11011011
      x               x
D
00000000 01000110 10100010 01110000 0
01100010 01111101 10011010 01000000

01101010 01111100 11001000 01000000
10000100 10111101 00001101 01110000
                        x
ALL
01111111 01101011 01110101 10011001 9
01001010 10000001 10100101 11011001

10000010 10000011 11111101 11011001
11001110 11010010 10011110 10011001
x                     x             <-- x is used to derive actionMap[]
                                        bit above x is different for ON and OFF
                                        (actionMap[] contains positions of x from right side)
                                        it needs to be reversed for button ALL (different order of 0 and 1)
```
Still don't know what the rest of the message means but at least we can decode it like this:
``` c
#include <stdint.h>
#include <stdio.h>

//                      00  01  02  03  04  05  06  07  08  09  10  11
int buttonMap[] =     {  3, 99, 99,  0, 99,  1, 99, 99, 99,  4, 99,  2 };  // 99 is invalid value
int actionMap[] =     { 16, 28, 11,  9, 11 };
int actionReverse[] = {  0,  0,  0,  0,  1 };

char* buttonName[] = {
    "A", "B", "C", "D", "ALL"
};

// we will test it here
int main(void) {
    uint32_t testData[] = {
        618491283,  3234986707, 3359271123, 3359271123,  // Button A
        273844117,  3744201941,   38598357, 2799660949,  // Button B
        3135175067, 3417090523, 1216381339, 3980271835,  // Button C
        4629104,    1652398656, 1786562624, 2226982256,  // Button D
        2137748889, 1250010585, 2189688281, 3469909657   // Button ALL
    };

    for (int i = 0; i < 20; i++) {
        uint32_t data = testData[i];

        // decode button
        int button = buttonMap[data & 0x0F];
        // decode action
        int action = (data >> actionMap[button]) & 1 ^ actionReverse[button];

        printf("Button %s %s (%u)\n",
               buttonName[button],
               action ? "ON" : "OFF",
               data
        );
    }

    return 0;
}
```

Output of this program should look like this:
```
Button A ON (618491283)
Button A ON (3234986707)
Button A OFF (3359271123)
Button A OFF (3359271123)
Button B ON (273844117)
Button B ON (3744201941)
Button B OFF (38598357)
Button B OFF (2799660949)
Button C ON (3135175067)
Button C ON (3417090523)
Button C OFF (1216381339)
Button C OFF (3980271835)
Button D ON (4629104)
Button D ON (1652398656)
Button D OFF (1786562624)
Button D OFF (2226982256)
Button ALL ON (2137748889)
Button ALL ON (1250010585)
Button ALL OFF (2189688281)
Button ALL OFF (3469909657)
```

The important part of this code is:
``` c
int buttonMap[] =     {  3, 99, 99,  0, 99,  1, 99, 99, 99,  4, 99,  2 };  // 99 is invalid value
int actionMap[] =     { 16, 28, 11,  9, 11 };
int actionReverse[] = {  0,  0,  0,  0,  1 };

// decode button
int button = buttonMap[data & 0x0F];  // This line will only work for valid values (0, 3, 5, 9, 11)
// decode action
int action = (data >> actionMap[button]) & 1 ^ actionReverse[button];
```
Maybe you will want to add some check to handle invalid data (valid values for `data & 0x0F` are `0, 3, 5, 9, 11`).
I hope this helps ;)

rasic, 2023-08-02:
@darkbox I have good news for you. Today I had time to look at the raw data on oscilloscope and now I understand why it isn't working for you and how to fix it :). The remote sends 6x 24bit code and then 4x 32bit code. Your program only shows the 32bit code but outlets only respond to 24bit code. I will write some program that will be able to receive both codes and then transmit the one you need. I will post it here hopefully this week. It will be useful for me too because by comparing codes from both remotes I would be able to learn more about this wireless protocol.

darkbox, 2023-08-02:
@rasic Wow that would be awesome!
Today I have been testing trying different protocols changing them by hand but funny enough I ended with more or less the same protocol 2 of the library xD as well as other random things with no luck so to me what you say makes sense of why doesn't work by just sending those 32bits. I even swapped the receiver and transmitter modules for another just in case was something physical at this point.

darkbox, 2023-08-16:
Hi @rasic, just checking if you did manage to make any advancement :) 
Can I do something to help you? If you have a repository I could take a look. I don't know much _C_ but maybe I can do some testing. I would like to be of more help but to be honest I'm lost at this stage without tooling to check things.

PS: If you are on vacations or something feel free to ignore me ;)

darkbox, 2023-08-25:
Good news! I finally managed to make it work with the help of an oscilloscope I found out that the remote sends two codes, one is 24bits of length and the other one is 32bits. RC-Switch only shown me the 32bits code but no the 24bits code :sweat_smile:.
So **the only one needed is the 24bits one using the protocol number 4** wicht is sent 6 times and then preceded by the 32bits code repeated 5 times with a slightly different timings luckily this one is not needed at all.

Just for illustration here is a capture of the full sequence when the button *A ON* is pressed:
[image]

Summarizing, my remote and plugs use the following codes accompanied by a quick interpretation of mine:
```
BUTTON | ACTION | 24 BITS CODE                  ID   | COMMAND?            | ADDR
A      | ON     | 111010011011101110101100 -->  1110 | 1001 1011 1011 1010 | 1100  -->  ID: 0xE COMMAND: 0x9BBA ADDRESS: 0xC
A      | OFF    | 111011011111010001111100 -->  1110 | 1101 1111 0100 0111 | 1100  -->  ID: 0xE COMMAND: 0xDF47 ADDRESS: 0xC
B      | ON     | 111000100110100111110101 -->  1110 | 0010 0110 1001 1111 | 0101  -->  ID: 0xE COMMAND: 0x269F ADDRESS: 0x5
B      | OFF    | 111001110010001110010101 -->  1110 | 0111 0010 0011 1001 | 0101  -->  ID: 0xE COMMAND: 0x7239 ADDRESS: 0x5
C      | ON     | 111001010100011010001110 -->  1110 | 0101 0100 0110 1000 | 1110  -->  ID: 0xE COMMAND: 0x5468 ADDRESS: 0xE
C      | OFF    | 111010011011101110101110 -->  1110 | 1001 1011 1011 1010 | 1110  -->  ID: 0xE COMMAND: 0x9BBA ADDRESS: 0xE
D      | ON     | 111001110010001110010111 -->  1110 | 0111 0010 0011 1001 | 0111  -->  ID: 0xE COMMAND: 0x7239 ADDRESS: 0x7
D      | OFF    | 111000010000110000110111 -->  1110 | 0001 0000 1100 0011 | 0111  -->  ID: 0xE COMMAND: 0x10C3 ADDRESS: 0x7
MASTER | ON     | 111001101100001000100010 -->  1110 | 0110 1100 0010 0010 | 0010  -->  ID: 0xE COMMAND: 0x6C22 ADDRESS: 0x2
MASTER | OFF    | 111000001001111000010010 -->  1110 | 0000 1001 1110 0001 | 0010  -->  ID: 0xE COMMAND: 0x09E1 ADDRESS: 0x2
```

m1ngr, 2024-06-20:
Hi, I can confirm what @darkbox and other have written.

The remote controller sends 6 * 24bit + 4 * 32bit code within a single button press.

The RC cycles 4 different 24bit values and 2 different 32bit values for each button.

You need to use any 24bit value to control the socket, the 32bit values are irrelevant. 

For example, button "A-On" sends these values:
```
2323804(24)=001000110111010101011100 +  449182419(32)=00011010110001011111101011010011 
3127724(24)=001011111011100110101100 + 2870635923(32)=10101011000110100110110110010011 
2399340(24)=001001001001110001101100 +  449182419(32)=00011010110001011111101011010011
2760988(24)=001010100010000100011100 + 2870635923(32)=10101011000110100110110110010011
```

To received the 24 bits values, you have to slightly modify RCSwitch.cpp, otherwise you'll get only 32bit values.
Change the "**nSeparationLimit**" constant from default 4300 to 2000.
With the 2000, you'll get both 24bits (it corresponds nearly to the default protocol 4) and 32bits.
You can try different nSeparationLimit values, eg. with 1250, it detects only 24bit value, with 2500+ only 32bit values, under 1000 you'll get nothing. Just try it.

For better detection you should add or modify any of existing protocol definition:
```
  { 370, {  1,  6 }, {  1,  3 }, {  3,  1 }, false },
  { 500, {  1, 14 }, {  1,  3 }, {  3,  1 }, false },
```
First line is for 24bit values, second line for 32bit values.

You don't need to modify RCswitch sources if you'll only use it for transmittion. Use the default protocol 4, works fine with following code.
```
  rf433.setProtocol(4);
  rf433.setRepeatTransmit(6);
  rf433.enableTransmit(RF_TX_PIN);
  rf433.send(B_ON_VALUE, 24); 
```
Just select one of four 24bit values (one for "On" and any one for "Off"), there is no logic with value cycling at all. 

I've tested with RCSwitch 2.6.4 and ESP8266. 

The RC cotroller is:
```
Parkside Model: 8505000114
Manufactured: 2022-10
IAN:424191_2204
```

Simplified log / button values I received:

```
A-On
2323804(24)=001000110111010101011100 +  449182419(32)=00011010110001011111101011010011 
3127724(24)=001011111011100110101100 + 2870635923(32)=10101011000110100110110110010011 
2399340(24)=001001001001110001101100 +  449182419(32)=00011010110001011111101011010011
2760988(24)=001010100010000100011100 + 2870635923(32)=10101011000110100110110110010011

A-Off
2277516(24)=001000101100000010001100 + 4032338067(32)=11110000010110001001100010010011
2939068(24)=001011001101100010111100 + 2249025747(32)=10000110000011010110100011010011 
2482812(24)=001001011110001001111100 + 4032338067(32)=11110000010110001001100010010011 
2835500(24)=001010110100010000101100 + 2249025747(32)=10000110000011010110100011010011

B-On
2518005(24)=001001100110101111110101 +  243763093(32)=00001110100001111000011110010101
2592229(24)=001001111000110111100101 + 3732864213(32)=11011110011111101111110011010101
2207493(24)=001000011010111100000101 +  243763093(32)=00001110100001111000011110010101
2692917(24)=001010010001011100110101 + 3732864213(32)=11011110011111101111110011010101

B-Off
3077845(24)=001011101111011011010101 + 1115158229(32)=01000010011101111111011011010101
2625093(24)=001010000000111001000101 + 2679336853(32)=10011111101100110110111110010101
2970565(24)=001011010101001111000101 + 1115158229(32)=01000010011101111111011011010101
2112149(24)=001000000011101010010101 + 2679336853(32)=10011111101100110110111110010101   

C-On
2835502(24)=001010110100010000101110 + 1965533659(32)=01110101001001111010100111011011
2277518(24)=001000101100000010001110 + 1891367323(32)=01110000101110111111100110011011
2939070(24)=001011001101100010111110 + 1965533659(32)=01110101001001111010100111011011
2482814(24)=001001011110001001111110 + 1891367323(32)=01110000101110111111100110011011

C-Off
3127726(24)=001011111011100110101110 + 3868754331(32)=11100110100110001000000110011011
2399342(24)=001001001001110001101110 + 1124737243(32)=01000011000010100010000011011011
2760990(24)=001010100010000100011110 + 3868754331(32)=11100110100110001000000110011011 
2323806(24)=001000110111010101011110 + 1124737243(32)=01000011000010100010000011011011

All-On
2482802(24)=001001011110001001110010 + 3088426457(32)=10111000000101011010010111011001 
2939058(24)=001011001101100010110010 +  593393049(32)=00100011010111100111010110011001 
2277506(24)=001000101100000010000010 + 3088426457(32)=10111000000101011010010111011001 
2835490(24)=001010110100010000100010 +  593393049(32)=00100011010111100111010110011001  

All-Off
2399330(24)=001001001001110001100010 +  555720345(32)=00100001000111111001111010011001
3127714(24)=001011111011100110100010 + 3002138073(32)=10110010111100001111110111011001
2323794(24)=001000110111010101010010 +  555720345(32)=00100001000111111001111010011001
2760978(24)=001010100010000100010010 + 3002138073(32)=10110010111100001111110111011001

```

bero158, 2024-08-13:
> Good news! I finally managed to make it work with the help of an oscilloscope I found out that the remote sends two codes, one is 24bits of length and the other one is 32bits. RC-Switch only shown me the 32bits code but no the 24bits code 😅. So **the only one needed is the 24bits one using the protocol number 4** wicht is sent 6 times and then preceded by the 32bits code repeated 4 times with a slightly different timings luckily this one is not needed at all.
> 

> Summarizing, my remote and plugs use the following codes accompanied by a quick interpretation of mine:
>
> ```
> BUTTON | ACTION | 24 BITS CODE                  ID   | COMMAND?            | ADDR
> A [...]

Thank you for your great work! I've just sniffed these 6x 3 bytes + 4x 4 bytes and was wondering what do they mean. My guess is: There are two different plugs using the same controller (maybe older and newer). Last four bits of 32bit code are button IDs too.