sui77/rc-switch #190: Extra character in binary string

nate123456, 2018-01-10:
Hi,

I'm trying to use this library to send a binary string to an rf transmitter. 

When I send the string and monitor the output via an antenna and recording software, I observe that the returned binary string has an extra zero on the end. The issue occurs with  multiple different binary strings, including ones that normally end in a zero.

i.e. '11110010000100100011011110001111' becomes '111100100001001000110111100011110'.

Here is my sketch: 

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

const int transmitPin = 12; //D6
const int pulseLength = 490;

int counter = 0;

void setup(){
  // put your setup code here, to run once:
  mySwitch.setPulseLength(pulseLength);
  mySwitch.setRepeatTransmit(12);
  Serial.begin(115200);
  mySwitch.enableTransmit(transmitPin);
  delay(300);
}

void loop(){
  // put your main code here, to run repeatedly:
  if (counter < 1)  { //stops it from spamming the binary string
    mySwitch.send("11110010000100100011011110001111"); //off signal
    delay(1000);
    counter = counter + 1;
  }
  delay(1000);
}
```

What could be causing this issue?

Thank you.

Martin-Laclaustra, 2018-01-11:
You do not share here your platforms nor your code at the receiving end and the problem might be there.
You may have a bug converting the final received value to binary. The received code is stored in an unsigned long which can be 32 or 64 bits depending on the platform.

nate123456, 2018-01-12:
Hi,

I do not have code at the receiving end as I am not receiving it programmatically- I am sniffing it using a usb device and decoding it by interpreting the audio file from a recording on the frequency of the transmitter. The raw output of the transmitter contains the extra character. Therefore I have no code at the receiving end to share. 

As far as the platform goes, I'm not sure if you're asking about the IDE I'm using or the hardware, but in either case here's what I've got:

I am using a esp8266 NodeMCU board with a [TX303AT](https://www.appliedwireless.com/index.cfm?fuseaction=product.display&product_ID=42&ParentCat=5) rf transmitter wired in that broadcasts at 303.825 MHz. 

I'm using the Arduino IDE platform.

Hopefully this is what you were asking for. 

Thanks for your help!

Martin-Laclaustra, 2018-01-12:
You are interpreting the sync bit as a data bit. In protocol 1, the "on" part of the sync bit is of the same length as the "on" part of a "0" data bit. The library sends this sync bit at the end of each pulse train.
Please close the issue if you think this answers your doubt.

nate123456, 2018-01-13:
The reason I am interpreting the sync bit as a data bit is because the device I am attempting to communicate with seems to be interpreting it the same way. I conclude this because when I send a binary string to the device it does not understand the signal unless I pre-remove a zero from the end of the binary string. This workaround only works with strings that coincidentally end in a zero, and, unluckily for me, one of the binary strings I need to send ends in a 1.

Is it possible to have the library not append the sync bit to the end of each pulse train? 

Thanks again.

Martin-Laclaustra, 2018-01-21:
Go back to square 1.
1. Explain your whole setting fully (you did not in your previous explanation): which device are you attempting to communicate with? (brand, model...) What response do you see? Which remote control do you have? How many commands?
2. Record the signal of your original remote with https://github.com/sui77/SimpleRcScanner
If you want, share your timings here (the numeric sequence, not the images). How did you previously come to your protocol number and codes? If you do not have a 433MHz receiver (I conclude this because of your comment on sniffing - are you using some SDR device?) then you could share an image of a complete train of pulses including at least 2.5 repetitions. (preferably for a couple of different buttons) You probably got the protocol or codes wrong (or your device uses a strange protocol). Please do not provide your reading of the codes in the audio file, but the raw signal.
3. Record your "synthetic" signal that you generate with rc-switch. You will be able to spot the differences (I assume that it is what you call the "final 0", but believe me, you are probably misinterpreting the problem). Again if you want, share here (preferably the numeric data of SimpleRcScanner, else, the raw image from your setting). Please do not provide your reading of the codes in the audio file, but the raw signal.

nate123456, 2018-01-22:
Hello again,

Thanks for getting back to me.

I think at this point that I'm going to close the issue as it seems we're not effectively communicating with one another. Some of what you have asked me for has already been provided as far as I can tell, which makes me think that I'm not adequately explaining the situation or I'm failing to understand your responses.

I believe I understand what you are thinking must be the cause of the issue where I'm incorrectly interpreting the output of the transmitter, but I'm confident that I'm correctly interpreting the output of my transmitter and of the device I'm trying to replicate. Here's my evidence for this:

When I used the library to control the device via the above code, it did not respond. This means that I either interpreted the raw output of the remote control incorrectly, or I used the library incorrectly, or the library is faulty.  We can eliminate the possibility that I interpreted the remote control output incorrectly, as when I pre-removed the zero from the binary string I sent to the library, the signal I replicated with the transmitter made the device respond. This means that I correctly 'read' the remote output, and correctly read what was being output by the library as being incorrect (i.e. with an extra 0). That is why I disagree that I am reading the output wrong.

The only possibilities that remain are that I am either using the library wrong, or the library is faulty in some way. I assume that I am using the library as is intended as I posted the code I was using in my original post and you didn't seem to have any issue with the code itself. 

Since there hadn't been any response in about 8 days, I figured I wasn't going to get any more help, so last night I wrote my own library that used digitalWrite() and delayMicroseconds to replicate the signal of the remote, which actually gave me control over the output, which this library seemed to be unable to do. It was surprisingly simple to implement, although it took a hour or so to calibrate the timings correctly. Using my own library, I was able to get the correct response from the device, and there was no longer any extra zero being added to the binary string I needed to send.

For those who are having this issue who want to implement your own library as I did, you can pull your output pin high (digitalWrite(transmitPin, HIGH;) then low to have your transmitter transmit the tone, and use delayMicroseconds(n) between high and low to set the duration on the transmission. Don't forget to leave it low for an appropriate time before transmitting again to ensure the proper gap size between transmissions. I found that, for my device, the length of the gap depended on the next bit, so if the next bit was a 1, the gap was short, and if the next bit was a 0, the gap was long. I also discovered that the short gap was the same length as a 0 bit transmission, and a long gap was the same length as a 1 transmission. This meant that I implemented the gap first then the bit transmission when coding it.

Martin, I appreciate your attempts to assist me. I apologize for my inability to adequately communicate my issue with you. I will close the issue, although I don't consider it resolved as the library still seems to fail to send the signal it is provided.

I hope this helps someone else out there who is having this problem. 

Nathan

closed 2018-01-22 by nate123456

Martin-Laclaustra, 2018-01-24:
@nate123456 
Nathan,
given that you wrote so politely, I felt invited to do the same.
The way you describe the "varying" gap hints once more about a possible misinterpretation of your recorded signal. The fact that you can "reproduce" the same signal does not mean that you achieved identifying with total accuracy the beginning or end of each train of pulses.
I must point out that you have not mentioned the brand, model, or type of what you call your "device". (You only described your transmitter)
You solved your need, but if you feel curious on the nature of the problem (I am intrigued what does the signal look like) feel free to post the signal, preferably the numeric timings (automatically recorded with https://github.com/sui77/SimpleRcScanner and spanning at least 2.5 trains of pulses - keep your remote button pressed during the recording period). If you do not have a receiver, you could attach the wav file to your post.
Be aware that there are some remotes that generate signals incompatible with this library. It is also worth to know whether your brand, once you declare it, is one of those, upon inspection of your signal.

jstartt, 2018-12-22:
Hi Nathan,

I'm having the exact same issue as you and the exact same setup you had.  I am using a esp8266 NodeMCU board with a TX303AT rf transmitter wired in that broadcasts at 303.825 MHz.  I'm also using an SDR device, with an audio editor, to view the waveforms.  Two of the codes I'm using end in a "0", so I had no issue with sending those codes.  The third code I need ends in a "1" and RC-Switch seems to add a character that looks like a zero at the end of each string.  It sounds like this is a "sync bit".  Do you know if there is anyway to turn that off?  If not, would you be willing to share the workaround code you used to simulate the RF bit stream?

Thank you,
Jason

jstartt, 2018-12-24:
> Hi Jason!
>
> I'm sorry you're running into the same issues. Hopefully I can help.
>
> It's been quite a bit since I worked on this project, and unfortunately
> the hard drive where I stored the code is now stuck in an all but dead
> computer. I made the mistake of not creating a repository for it at the
> time, so I am unable to pull up the code I used to make it work as it still
> does currently (I was insecure about having my code out there and didn't
> want to pay for a private repository). I have an external HDD adapter on
> the way (your request finally inspired me to go rescue my old files), but
> it won't arrive for a few days. In the meantime, I'll try to recall how I
> implemented my workaround.
>
> I left some description in my last post from January that seems to provide
> the clues you'd need to write your own manual implementation. As far as the
> calls to the 'API' that governs sending the signal, I used the methods
> digitalWrite and delayMicroseconds, so now you have the syntax I used to
> 'write' to the transmitter directly.
>
> As far as the logic that I used goes, I left some description of how to
> implement similar logic- use a digitalwrite high, delay, and digitalwrite
> low, and further delay to simulate a spike as a one or zero as you see in
> your audio editor.
>
> [... 83 lines cut ...]
> I hope I was able to help!
>
> Best of luck,
>
> Nathan

Hi Nathan,

Thank you so much for the thorough response.  I really appreciate your
insight.  I wanted to let you know that I got my project working using
RCSwitch after all.  After reading the notes and MUCH experimenting, I
found that you can "zero out" the initial synch-bit by using
"customprotocol" and changing the synch-bit ratio to "0" as shown in the
"{  0, 31 }" as seen below:

static const RCSwitch::Protocol customprotocol = { 400, {  0, 31 }, {  1,
3 }, {  3,  1 }, false };
mySwitch.setProtocol(customprotocol);

This "0" effectively removes the first bit in the synch-bit (neither a "0"
or "1") and inserts 31 blanks.  Then in the data, you can insert your bit
stream and it will only create the "1" and "0" codes you specify (no extra
timing bits).  For example:

const char* onCode = "0100010010001011001111000";  //onCode is what I used
to turn on my fireplace :)

After I did this, I know longer had the extra "0" at the end of my
bit-stream.

I found this article to be the most useful, but it took me a while to
decipher it.

https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1

Once again, thank you so much for your fast response.  I appreciate your
willingness to offer assistance!

Thanks,
Jason

nate123456, 2018-12-24:
Jason,

I'm glad you were able to figure it out! I sounds like all I needed to do at the time was reconfigure rc-switch properly, which makes sense given the consistent nature of the problem. I suppose I ruled that possibility out originally because the person who was helping me with the issue back in January seemed to think that it was not a rc-switch configuration issue.

Thanks for providing a description of what you did to get it working. I'm sure others will benefit from your discoveries, and therefore will not have to go through the trials and tribulations we endured!

Best of luck,

Nathan

Martin-Laclaustra, 2018-12-24:
@jstartt
You are sending 25 data bits. It is a strange length. One would expect 24. Probably, your last bit is actually the sync bit. You might consider dropping that last data bit and reinserting the 1 in the sync part of the protocol.

mwrierson, 2025-04-06:
> Hi Nathan,
> 
> Thank you so much for the thorough response.  I really appreciate your
> insight.  I wanted to let you know that I got my project working using
> RCSwitch after all.  After reading the notes and MUCH experimenting, I
> found that you can "zero out" the initial synch-bit by using
> 
> "customprotocol" and changing the synch-bit ratio to "0" as shown in the
> "{  0, 31 }" as seen below:
> 
> static const RCSwitch::Protocol customprotocol = { 400, {  0, 31 }, {  1, 3 }, {  3,  1 }, false };
> mySwitch.setProtocol(customprotocol);
> 
> This "0" effectively removes the first bit in the synch-bit (neither a "0"
> or "1") and inserts 31 blanks.  Then in the data, you can insert your bit
> stream and it will only create the "1" and "0" pulses you specify (no extra
> timing bits).  For example:
> 
> const char* onCode = "0100010010001011001111000";  //onCode is what I used
> to turn on my fireplace :)
> 
> After I did this, I know longer had the extra "0" at the end of my
> bit-stream.
> 
> I found this article to be the most useful, but it took me a while to
> decipher it.
> 
> https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1
> 
> Once again, thank you so much for your fast response.  I appreciate your
> willingness to offer assistance!
> 
> Thanks,
> Jason
> […](#)

Jason,

Your good work continues to help people.  This is the fix I needed to get rid of the synch-bit to make my code work. Thank you.