sui77/rc-switch #193: Detect AC113 remote signal

giotms, 2018-01-14:
I need help to recognize a AC113 remote (16 channels with up,  down and center buttons).

Remote probably emits 64 bits of data, but I don't know how editing code. Is there anyone that help me?

G

I tried https://github.com/sui77/SimpleRcScanner.
This is an examples of ch1-up button

208,568,232,184,592,188,624,584,208,568,224,184,596,204,596,592,212,184,612,180,596,584,212,184,616,568,240,564,212,580,212,580,216,564,236,180,596,208,600,184,612,180,612,180,620,180,596,184,624,180,612,180,612,568,212,204,600,180,624,180,612,184,592,200,596,180,620,180,624,184,596,196,596,184,612,564,236,180,596,212,596,580,212,180,616,180,620,568,208,184,624,580,212,564,228,184,596,588,212,596,208,184,608,564,900,4876,600,588,216,176,616,564,228,180,600,176,624,176,632,560,232,560,220,192,600,584,216,180,628,176,600,576,216,580,216,176,624,176,604,588,216,580,216,176,616,176,604,580,220,176,628,176,620,560,232,180,600,584,216,588,220,560,232,560,216,576,220,176,624,176,632,176,600,192,604,176,616,176,624,176,604,204,600,176,616,576,220,180,600,196,604,176,628,176,616,180,616,176,600,180,620,180,628,180,616,176,600,576,216,180,620,180,628,560,216,192,600,180,620,560,236,180,600,588,216,576,220,176,616,560,216,584,220,176,628,576,884,4864,628,556,224,200,604,572,220,176,620,172,604,196,604,584,224,572,224,172,620,556,220,176,624,176,632,556,236,556,224,188,604,172,632,556,248,556,224,184,608,172,620,556,244,172,608,200,604,572,224,172,620,556,224,580,220,584,224,568,224,556,236,172,608,172,628,176,632,172,620,172,608,184,608,172,628,172,632,172,608,572,224,172,620,176,624,172,608,200,604,172,620,172,624,172,604,196,608,176,628,176,620,556,236,172,604,176,624,588,220,172,620,176,608,572,220,172,628,560,248,556,220,188,604,576,220,556,244,172,604,584,904,4864,608,576,224,168,636,552,228,184,608,172,620,172,628,556,224,580,228,172,624,568,224,168,612,188,608,584,224,568,224,168,628,168,612,576,224,580,224,172,624,168,608,572,224,168,632,168,636,556,228,184,608,568,224,556,244,556,224,580,224,568,228,168,624,168,612,192,608,172,636,168,624,168,624,172,608,168,632,168,636,552,244,172,608,184,608,168,632,172,632,172,608,184,608,172,620,172,632,172,608,196,608,572,220,172,624,168,608,580,220,172,636,168,628,556,236,172,608,576,224,584,220,172,624,552,224,572,220,172,628,556

kuchod, 2018-01-17:
Looks like the signal is this:

4876,600,588,216,176,616,564,228,180,600,176,624,176,632,560,232,560,220,192,600,584,216,180,628,176,600,576,216,580,216,176,624,176,604,588,216,580,216,176,616,176,604,580,220,176,628,176,620,560,232,180,600,584,216,588,220,560,232,560,216,576,220,176,624,176,632,176,600,192,604,176,616,176,624,176,604,204,600,176,616,576,220,180,600,196,604,176,628,176,616,180,616,176,600,180,620,180,628,180,616,176,600,576,216,180,620,180,628,560,216,192,600,180,620,560,236,180,600,588,216,576,220,176,616,560,216,584,220,176,628,576,884,

I think you can round every value lower than 400 to 200 and the greaters to 600. 
pulselength = 200
0: 200ms down, 600ms up, 0111
1: 600ms down, 200ms up, 0001
sync bit 1 up, 23 pulses down
and looks like have a tail bit, 4 pulses up 1111

(4876 would be 4676+200)

4876,600,600,200,200,600,600,200,200,600,200,600,200,600,600,200,600,200,200,600,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,600,200,600,200,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,200,600,600,200,600,200,200,600,600,800,

I am not sure it is correct and I dont know how to write the protocol. How does this tail pulse work with this library?

kuchod, 2018-01-18:
check the issue #163, is about a very similar signal

kuchod, 2018-01-18:
you should be able to send the following signal using Martin´s modification to the Suat´s library:

800,4876,600,600,200,200,600,600,200,200,600,200,600,200,600,600,200,600,200,200,600,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,600,200,600,200,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,200,600,600,200,600,200,200,600,600,0

the 800 ms pulse is part of the preamble, not tail.

giotms, 2018-01-19:
I read [#163 ](https://github.com/sui77/rc-switch/issues/163) but I need help to decode, not to send signal
How can I decode this signal?

kuchod, 2018-01-19:
plotting the signal like this

,800,4876,600,600,200,200,600,600,200,200,600,200,600,200,600,600,200,600,200,200,600,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,600,200,600,200,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,200,600,600,200,200,600,200,600,600,200,200,600,200,600,600,200,200,600,600,200,600,200,200,600,600,200,600,200,200,600,600,

it looks like the protocol is something like this

pulselength = 200
sync bit 4 up (800ms), 24 pulses down (about 4676ms)
0: 200ms down, 600ms up, 0111
1: 600ms down, 200ms up, 0001

{200, {4,24}, {1,3}, {3,1}, false}

makes sense?

Martin-Laclaustra, 2018-01-21:
You do not have a good chance to use rc-switch for receiving signals from this remote and triggering responses in your code (is this what you want?)
You have two sync bits, something that the library is not prepared to cope with.
You can easily mimic the signal, but not receive it to detect when it does happen.
The sync part is, in ms, high-low like this:
{600,900}{4900,600}
then you have your 64 bits of data coded as:
{600,200} for 1
{200,600} for 0

kuchod, 2018-02-16:
giotms, you can use this branch

https://github.com/perivar/rc-switch

it supports sync and pause pulses. My remote is an AC123-01 with a very similar code

    {194, {27, 3},  {3, 1}, {1, 3}, {3, 4}, false}, // protocol 11 

and I've been able to make it work.

giotms, 2018-02-19:
I user perivar branch. I tried but it doesn't work. 

Must I copy 
{194, {27, 3},  {3, 1}, {1, 3}, {3, 4}, false}, // protocol 11 
in RCSwitch.cpp file?
Regards
G

kuchod, 2018-02-19:
Yes, perivar´s branch lets you generate the two sync pulses this protocol have, but you must edit rcswitch.cpp to add your protocol.

As your pulses are slightly diferent from mine, try adding this line:

{194, {24, 3}, {3, 1}, {1, 3}, {3, 4}, false}, // protocol 11

try sending this code with the send function:

1010001101001100110010010111110000000001000000000010010010110110

and use an arduino to capture the generated signal to compare it to the posted in your first post and correct the code.

giotms, 2018-02-21:
I tried but I failed to receive and decode signal.

I'm interested in decoding pressed buttons. The remote control has 3 buttons (up down and stop) with 15 different channels.

I would create a sort of bridge from this remote to a home automation controller (like openhab)

Can the branch https://github.com/perivar/rc-switch decode this protocol?

kuchod, 2018-02-22:
I used it to generate the signal, sending the code I manually interpreted, and it works. 
When I try to decode the signal using the ReceivedDemo_Advanced sketch it doesnt work, but the observed signal (declare DEBUG) seems to be valid... it must be very close to work..