sui77/rc-switch #194: Not receiving anything

pkishino, 2018-01-17:
Hello, nice looking library.
I have been trying all evening to get the simple receiver to work with my arduino nano and adafruit feather huzzah together with the https://www.kjell.com/se/sortiment/el-verktyg/elektronik/fjarrstyrning/sandar-och-mottagarmodul-433-mhz-p88905
one that is listed under RF devices
and as remote I am using the nexa neyct-705 remote as well as a generic 4 button 433 remote. The nexa can talk and program all my plugs.
But the receiver never shows anything :(
I have wired it to 5v from arduino and d2
Have tried two receiver modules, both the same, labeled MX-RM-5V
any idea?

kuchod, 2018-01-18:
Your remote may be using a protocol not recognized by this library. Try this one:

https://github.com/Martin-Laclaustra/rc-switch

and use this sketch:

https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino

it scans unknown protocols and outputs raw data. Then you can plot it with 

test.sui.li/oszi/

pkishino, 2018-01-18:
Thanks, I'll try that

pkishino, 2018-01-18:
hmm,tried your suggestion but still shows nothing.
Ive wired the data pin on the receiver to D2 on arduino nano, this should then result in enableReceive(0) correct?
And Ive tried sending using the accompanying sender from a different arduino using sendDemo and still nothing..

kuchod, 2018-01-18:
Also check issue #112, it is about decoding a nexa signal.

kuchod, 2018-01-18:
and this page: http://dephiox.blogspot.com.es/2013/01/decoding-new-nexa-protocol.html

kuchod, 2018-01-18:
Also try this sketch

https://github.com/sui77/SimpleRcScanner/blob/master/htdocs/Sketchbook/SimpleRcScanner.ino

in fact, may be the one I was using at home to get the unsupported data.

chamithchathuka, 2018-01-20:
It works with Martin-Laclaustra protocol receiver branch

finity69x2, 2018-02-03:
I already have hex codes generated by a Broadlink RM Pro for a remote that apparently isn't supported by RC-Switch. Can those codes be used to define a protocol/code to be used in RC-Switch?

chamithchathuka, 2018-02-03:
I have less Idea with RM Pro, 
but as @kuchod  mentioned,

try using the 

[](https://github.com/Martin-Laclaustra/rc-switch)

and use this sketch:

[](https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino
)

Then include the purposed protocol in the RCSwitch library and send the signal received.

Hope you get an idea. If you need more information let us know.

finity69x2, 2018-02-04:
I downloaded the rc-switch protocolless repository as directed above. I uploaded the sketch from above onto a NodeMCU. 

Now when I try to run the code and watch the serial monitor for signs of life I don't see any response to the button push on the remote.

Actually, after lots of trial and error (almost all error unfortunately) I think I've narrowed it down to my hardware setup. I can't figure out which pin to use on the nodeMCU for the receiver interrupt.

The sketch says to use interrupt 0 (pin #2) for an arduino (which I'm pretty sure corresponds to D4 on the nodeMCU).

I'm pretty sure the nodeMCU is never receiving the interrupt from the output of the receiver board. I put a simple serial print at the very beginning of the loop (print "working") and that is running continuously on the screen but when I push a button on the remote nothing happens.

I've tried changing the code to use pins D2, D3 & D4 (GPIO4, GPIO0 & GPIO2 respectively). I also am pretty sure the receiver is actually putting out data because the built-in LED is connected to D4 (GPIO0) and it flickers with the receiver data pin connected to it and when I push the button on the remote the LED goes solid for a split second.

Anyone have any ideas on what might be wrong?

finity69x2, 2018-02-05:
Ok... So here is the next bits of data.

I loaded a sketch of senddemo.ino onto my NodeMCu baord with a TX unit data pin attached to the SD3 pin on the NodeMCU. After loading it I could scan the output of the TX using a rtl-sdr and saw the frequency was transmitting at 433.81mhz. So I know it is working.

now on to the RX end:

I loaded simplrcscanner.ino onto the NodeMCU without modification. I got nothing at all on the serial monitor. And since I have the senddemo.ino running on the TX I should be receiving something. 

looking at the code I notice that there are two pins defined an interrupt pin on GPIO0 & a receive pin on GPIO2. From what I have read the esp8266 (nodeMCU) doesn't handle interrupts the same as the ardiuno. So I was looking around for info on how to connect the interrupt and receiver pins to be used with the SimpleRcScanner sketch and found where someone recommended to use GPIO13 (D7 pin on NodeMCU) as the receive pin ( myswitch.enableReceive(13) ) in the RCSwitch receive demo sketches. 

I tried that by modifying Martin-Laclaustra's ProtocolAnalyzeDemo sketch to set the receive pin to 13 and suddenly I now see the signal received by the TX in the serial monitor of the RX. Success!! Kind of...

Now when I press a button on the remote I still see no response on the serial monitor. 

I am now very confident that all my hardware is working. I can see the remote signal on my rtl-sdr @ 433.910mhz. every time i push the button it sends out a series of four bursts.

My question now becomes, how do I modify the simplrcscanner.ino to be compatible with the nodemcu board? Or should I just give up on that and put the sketch onto a raspberry pi and see if that will work?

If there is any other information that I can provide to (hopefully) get this to work please (PLEASE!) let me know.

chamithchathuka, 2018-02-05:
I am not clear what did you received?
Is that a signal you sent by your self?

But you want to receive the signal sent by the remote. am I correct?

Martin-Laclaustra, 2018-02-06:
Forget:
https://github.com/sui77/SimpleRcScanner/blob/55eb073a85689619b59398251d6b9bb4098eb363/htdocs/Sketchbook/SimpleRcScanner.ino#L9
That value is automatically calculated for your Rx pin (just one pin) in line:
https://github.com/sui77/SimpleRcScanner/blob/55eb073a85689619b59398251d6b9bb4098eb363/htdocs/Sketchbook/SimpleRcScanner.ino#L12
Just define the correct pin as "D1" - "D16"
In your case write:
`static int receiverPin = D7;`

finity69x2, 2018-02-09:
Ok thanks I see now.

omersiar, 2018-03-11:
Hello,

I also have unsupported transmitter, with these tools and audacity, i captured the raw data the transmitter sends.

https://github.com/Martin-Laclaustra/rc-switch
https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino

This is the image representation of raw data.
[image]

This is the sound representation:

[image]

This is the text:

```
84,128,60,12,8,416,208,16,208,68,300,168,360,8,500,32,10968,12,796,20,112,40,396,36,84,16,184,80,132,32,32,116,20,148,72,48,92,116,228,48,28,396,68,192,80,136,44,24,76,280,12,172,52,40,12,972,16,40,60,128,36,136,64,500,24,764,28,84,40,160,52,220,80,312,12,28,64,80,100,12,168,16,20,92,40,36,60,212,40,12,60,32,48,72,96,68,80,76,16,20,208,132,176,20,3316,64,172,388,64,188,16,76,40,20,40,20,60,68,28,16,64,416,32,140,12,176,12,100,72,284,40,52,24,368,44,164,56,208,72,1244,36,12304,12,16,84,276,28,108,8,100,52,76,68,48,152,216,40,16,48,92,96,76,112,40,72,36,296,132,108,12,724,24,8,44,92,52,108,48,640,64,76,52,368,8,164,60,72,72,44,176,48,76,48,80,48,176,100,4764,1508,332,720,336,724,332,720,692,368,688,368,688,372,328,728,684,376,328,724,328,736,324,732,328,724,328,732,684,372,324,728,688,376,328,728,324,732,684,372,324,732,684,372,324,732,684,372,684,376,328,736,324,728,328,732,680,376,320,732,324,732,324,732,684,380,324,728,328,728,324,732,684,372,324,732,324,732,324,732,684,12404,4740,1512,324,728,328,728,324,732,684,372,684,372,684,376,324,732,680,380,324,732,324,728,328,728,324,732,324,732,688,372,328,732,680,380,324,728,328,732,680,376,324,732,680,376,324,728,684,376,684,376,324,732,324,732,324,732,684,376,320,736,324,732,324,736,680,380,324,732,320,736,320,732,684,372,324,732,324,732,320,732,684,12404,4736,1516,324,728,324,740,324,728,684,372,684,376,680,376,324,732,680,380,324,732,324,732,324,728,324,732,324,732,684,376,320,732,684,376,324,732,328,732,684,376,320,736,680,372,324,732,684,372,684,380,324,732,324,732,320,736,680,376,320,732,324,732,324,732,680,384,320,740,324,732,320,736,680,372,328,728,324,732,324,732,684,24332,28,456,36,104,44,244,8,80,12,72,80,80,20,56,52,52,20,52,48,92,64,60,80,72,20,12,20,104,24,44,8,40,72,80,72,88,44,56,32,16,12,84,112,12,172,36,32,68,40,36,88,48,360,92,16,64,12,
Reset your Arduino to scan again...
```

This is the working arduino skecth that i figured out:
Actually opens my garage door, is this can be implementet by this library?

The remote's PCB labeled as **DC184 v1.1** and shell label is **DC115B**

``` c++

#define RF_DATA_PIN 2
#define lowdelay 360
#define highdelay 690 

void setup() {
  pinMode(RF_DATA_PIN, OUTPUT);
}

void loop() {
  sendSokak();
}

void sendSokak() {
  //SYNC
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(12404);

  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(4730);

  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(1490);

  //loop 1 ------------------------
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);

  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);

  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);

  //loop 2 ------------------------
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(highdelay);
[... 178 lines cut ...]
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(lowdelay);
  
  //loop 4 ------------------------
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);
  
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);
  
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(lowdelay);
  digitalWrite(RF_DATA_PIN, LOW);
  delayMicroseconds(highdelay);
  
  //SYNC 6 ------------------------
  digitalWrite(RF_DATA_PIN, HIGH);
  delayMicroseconds(highdelay);
  digitalWrite(RF_DATA_PIN, LOW);
}
```

Martin-Laclaustra, 2018-03-14:
@omersiar have a look at:
https://github.com/sui77/rc-switch/issues/146
https://github.com/sui77/rc-switch/issues/163

UserNAMEPersonThingy, 2018-05-23:
Hello, when I try to compile ProtocolAnalyseDemo.ino, it fails at getRecievedLevelInFirstTiming. Help would be appreciated.

minusdavid, 2018-08-04:
Yeah according to https://github.com/arduino/Arduino/issues/3929 it looks like maybe just an ancient IDE courtesy of the Debian distro :/

minusdavid, 2018-08-04:
Didn't want to uninstall the Debian packages and didn't want to clutter up my desktop, so I took inspiration from Jessie Frazelle (https://blog.jessfraz.com/post/docker-containers-on-the-desktop/) and found a Dockerfile from Maël Auzias (https://stackoverflow.com/questions/34017038/running-gui-in-docker-no-ssh-no-vnc) that let me run up the latest version of the Arduino IDE inside a Docker container. 

I updated Auzias's Dockerfile to use the latest Arduino IDE code and added the user to the dialout group and used --device /dev/ttyACM0, and it worked perfectly. I got my Arduino IDE 1.8.5 running on my Debian host (from a Ubuntu 14.04 based container) over a X11 unix socket connection.

I was able to compile the code from https://raw.githubusercontent.com/Martin-Laclaustra/rc-switch/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino and uploaded it to my Arduino UNO, and the results are fantastic!

With rc-switch, I was getting absolutely nothing on the serial monitor 99.99% of the time. With @Martin-Laclaustra's fork, I get output every time I click my doorbell button.

Now to try plotting it using http://test.sui.li/oszi/ like @kuchod suggested. 

Thanks so much for the useful information, people! 

Hopefully this comment will help anyone working with an old IDE... Docker doesn't have to just be for work or servers...

Martin-Laclaustra, 2018-08-05:
You probably can skip the plotting chapter. Just read the output of the ProtocolAnalyzeDemo.ino and use/test the protocol proposed in it.

minusdavid, 2018-08-05:
Thanks, @Martin-Laclaustra. This is my first time working with RF hardware, so I've been reviewing the rc-switch source code to see what the protocol even signifies. I think that I've nearly worked it all out, so I plan to test the data received from the doorbell button on the RF receiver with a transmission from a RF transmitter to the doorbell receiver. It was neat plotting out the data to see how it mapped to a binary format. 

I was wondering... do you have a pull request open with rc-switch to incorporate your changes? They seem really useful. I'm trying to work out exactly what it is about your changes (as distinct from the vanilla rc-switch) that make it work for me.

This is the typical output of ProtocolAnanlyzeDemo.ino:

`
Received 2817 / 12bit Protocol: 6
data bits of pulse train duration: 12532
proposed protocol: { 348, { 15, 2 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
5324 664 
764 300 416 612 796 232 904 148 468 600 424 612 428 616 396 668 
348 688 348 692 348 692 720 31
`

That proposed protocol does seem pretty close to Protocol 6 (I haven't worked out whether it's actually in the thresholds but I'm guessing maybe it is if I'm getting that output). 

I'm getting about 5-6 repetitions. Anyway, I shouldn't use this space for that...

I guess I'm just concerned about using your fork in the long-term because it hasn't been updated since May 2017 while the upstream rc-switch was updated in November 2017. I suppose the latter is still a while ago so I imagine it's pretty stable...

minusdavid, 2018-08-05:
The data received using the code in https://raw.githubusercontent.com/Martin-Laclaustra/rc-switch/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino is definitely correct. I was just able to spoof the purchased doorbell button and get the purchased doorbell receiver to activate. 

Set the protocol to 6, changed the pulse length to 350, and set the repeat transmit to 6. 

Anyway, I'll stop spamming everyone. I'm just so excited that this worked.

hycday, 2019-02-13:
trying to decode a signal from a Flamingo, model SF-500S, but i am not even receiving anything...any help is welcomed :)

minusdavid, 2019-02-13:
I'd suggest reading the comments above. 

Play with https://raw.githubusercontent.com/Martin-Laclaustra/rc-switch/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino instead and you'll probably have better luck.

hycday, 2019-02-13:
this is what I did, but nothing appears on the serial monitor.... using another brand/remote, it works. 
As if my Flamingo SF-500S is not detected.

minusdavid, 2019-02-13:
Your Flamingo SF-500S looks like it's 433.92 which is the same as my doorbell.

But I'm no expert on RF and I haven't looked at the code for a while. 

If it were me, I'd start fiddling with the library. Adding extra diagnostics at the earliest steps.

But hopefully one of the people who know more about RF and this library will comment.

hycday, 2019-02-14:
yes it's 433.92 MHz.
using the above code, I get mySwitch.available() = false with the SF-500S.
I know the code is good because using anoter brand/remote, i get a signal.

So I believe the available() function is "too restrictive" or not detecting anything from the SF-500S.
How would I need to modify the library to allow it to detect further signals then ?

minusdavid, 2019-02-14:
Not sure off the top of my head. I know I've gone into the library code and just messed around.

I work as a software engineer though so I'm used to troubleshooting and diagnosing things through a bit of trial and error when there's no obvious faults.

Martin-Laclaustra, 2019-02-15:
@hycday
You have several problems that make this library unsuitable for your switch.
To set you on track, this is are quick hints from observing "A-ON":
This is the head of you signal (2 sync bits + 4 first data bits, each bit one timings pair:on-off): 

243 | 10782 | 237 | 2584 | 250 | 1309 | 245 | 304 | 250 | 1308 | 246 | 306
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
The signal repeats each 132 timings, that is 2 sync bits 64 data bits.
The pulse length is ~260. The syncs are aprox. (1,41) and (1,10), and data are encoded as (1,5) and (1,1) to separate 1s and 0s.
The problems are:
- 64 bits (this can be solved by modifying the library code).
- 2 sync bits (this can not be solved).

With this library you can only expect to mimic the signal, but never to receive it. To achieve this you should use my branch https://github.com/Martin-Laclaustra/rc-switch/tree/transmittimingsarray
Search around a little bit in issues and pull request to find out what to do.
Best luck!

hycday, 2019-02-15:
thanks Martin for your answer. So I found out some elements of answers indeed (in case other are also searching, I believe this comment is good enough : https://github.com/sui77/rc-switch/issues/237#issuecomment-399211046). I will dig more into that to know how to use the info you mentionned and try to understand how to use the transmittimingsarray branch so that i could send arrays.

I now understand that my issues is due to the 2 sync bit, however, I have no idea what that means....and I have no idea how you got all that info by observing A-ON :/  I would really love to understand how you did that, so that I can replicate the same logic for the rest, and also to know what I am doing rather than just "copy paste thanks bye". 

I tried to understand; and I wonder if my "logic" is good.
so if i clean the A-ON signal by removing all the "duplicates" (easy to do via the visual graph), I have : 
```
244,10770,249,2579,248,1313,246,301,251,1308,246,302,251,1318,238,307,244,300,246,1315,246,1306,249,307,246,297,250,1311,248,1308,247,301,252,300,244,1319,242,301,245,1319,243,1311,243,314,239,1312,243,306,247,1306,248,306,244,1314,244,306,247,303,243,1313,246,305,241,1322,239,1308,248,307,244,307,240,1313,247,302,245,1316,245,302,242,1313,246,309,239,1319,242,320,225,1311,251,305,240,1314,246,300,246,1312,248,313,231,1315,247,304,242,1315,245,303,244,1316,243,303,243,1312,249,1315,240,312,239,305,242,1318,243,300,246,1323,237,302,245,1312,246,1312,245,306
```
1- I am not sure how to find the real beginning of a signal... why does it start with "244, 10770, 249..." and not one data bit before or after ? how to know that ? I am asking because then it completely changes the way I look at the graph and it can make it "mirrored". I looked at the pattern, and went from one "10000" to another "10000" and basically, in order to keep the minimum to do a loop, it had to start with 244, 10770...is that a good logic or is there a better way to know ?

2- lets say, considering point one is understood, and taking the head signal data bits that you quoted on above comments as example, then are you saying that it is a 2 sync bit signal because of the 10782 and 2584, which are longer than the other 0 and 1 bits that would be around 1300 and 300 ? (also this is visible in the graph because the first 2 are longer than all the rest ?) (see first 2 red lines in image in point 3 below)

3- with that in mind, i then have 2 sync bits, followed by 64 data bits (2+64 red lines in the above image)
[image]. Ok I think this is clear.

4- then here, i dont get how you found the "every 132 timing", is it 2+64 = 66, x2 = 132 ? which I can also get by counting how many comma separated numbers i have in the signal ? 

5- then you got the 260 pulse by looking at the lowest number (average) and you got that the bit sync are (1, 41) and (1,10) by doing 10770/260 = 41 and 2585/260 = 10 approx. then same for 0 and 1. but how do you know if 0 is (1,1) and 1 is (1,5) or 0 is (1,5) and 1 is (1,1) ? 

So I think I get it better if this is it. I only need to understand how to use these info to create a new protocol and send my signal A-ON (1010100110011001011010101001011001010101010101010101011001010110)
I tried "normalizing" the signal and got 
```
260,10660,260,2600,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,1300,260,260
```
Then I tried the below, but it didnt work...
```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

unsigned int data[]   = {260,10660,260,2600,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,1300,260,260,260,260,260,1300,260,260,260,1300,260,260,260,1300,260,1300,260,260, 0};

void setup() {
  mySwitch.enableTransmit(13);  
}

void loop() {
  mySwitch.send(data);
  delay(200);
}
```
data being a starting-low zero-terminated array of timings...not sure if I need to do anything else now...but no success till now (but at least i have a feeling i understand more)

181221, 2019-04-24:
@SebastianGrans
I got the same remote sockets that you have, but I am not able to get it to work. How did you get it to work? Can you post your codes, maybe I can try them and see if they work for me to?

SebastianGrans, 2019-04-26:
> @SebastianGrans
> I got the same remote sockets that you have, but I am not able to get it to work. How did you get it to work? Can you post your codes, maybe I can try them and see if they work for me to?

I tried to document everything I did and it should be available here: [http://sebastiangrans.github.io](http://sebastiangrans.github.io)

Good luck :)

vegovs, 2020-04-10:
@SebastianGrans I also got the same remote sockets that you have(Luxorparts 50987 with BQ31081 chip), and i have decoded the signal, but my wall-sockets wont react when i send the code from my RPi even tho the signal seems identical! 

This is the recording of the remote: (A On) 

Decimal: 3450469075 (32Bit) Binary: 11001101101010011111101011010011 Tri-State: not applicable PulseLength: 738 microseconds Protocol: 2
Raw data: 7384,1412,676,1416,676,368,1708,368,1720,1400,680,1412,672,372,1712,1412,672,1416,676,368,1712,1412,676,368,1712,1412,676,368,1716,360,1700,1420,680,1412,688,1404,680,1408,688,1404,676,1416,672,372,1724,1396,680,368,1708,1412,684,1408,692,356,1712,1408,672,372,1728,336,1736,1388,696,1392,708,

This is the recording of the RPi signal: (A On) 

Decimal: 3450469075 (32Bit) Binary: 11001101101010011111101011010011 Tri-State: not applicable PulseLength: 749 microseconds Protocol: 2
Raw data: 7492,1484,884,1464,928,120,416,156,1644,712,1628,1464,876,1476,912,704,1624,1464,916,1440,876,740,1608,728,1636,1456,888,728,1616,736,1652,1436,896,1456,892,1412,960,1440,932,1420,908,1456,924,704,1640,1112,176,716,1628,1456,896,1456,904,708,1628,1460,908,704,1628,724,1632,1456,892,1456,904,

Do you have any idea what the issue could be?

closed 2021-03-11 by pkishino

ThFischer, 2022-09-19:
Hello,
I try to receive data from a "TFA Dostmann 30.3249.02" (thermo/hygrometer).
The payload is similar to the [TFA 30.3221.02 protocol](https://github.com/RFD-FHEM/RFFHEM/blob/a1222ecd68e1febeca894ebdabfe71c22ae7fa06/FHEM/14_SD_WS.pm#L591) which I could validate with a LogicAnalyzer.
The problem is: Neither rc-switch's ReceiveDemos nor  the ProtocolAnalyzeDemo outputs any data (other transmitters are well recognized).

The signal looks like this:
[image]

1. frame-sequence of either 3 or 4 impulses ~750µs High / 700µs Low
2. payload
3. frame sequence of 4 impulses ~750µs High / 700µs Low
4. payload
5. ...

The payload is sent 4 times, at the end no frame sequence is sent.

Is there any chance to get this signal recognized by rc-switch?

Thanks in advance
Thomas