sui77/rc-switch #196: Cannot set protocol 0

chamithchathuka, 2018-01-20:
This is the signal I want to send 

`Decimal: 13930215 (24Bit) Binary: 110101001000111011100111 Tri-State: not applicable PulseLength: 294 microseconds Protocol: 0
Raw data: 6014,831,857,313,859,319,337,836,857,319,336,835,857,319,335,836,340,842,849,323,335,842,329,843,330,846,845,330,843,329,845,330,325,843,850,323,849,329,842,330,334,838,331,843,847,328,845,327,844,`

but I have received 

`Decimal: 13930215 (24Bit) Binary: 110101001000111011100111 Tri-State: not applicable PulseLength: 358 microseconds Protocol: 1
Raw data: 10953,1070,357,1076,364,368,1065,1074,357,370,1063,1076,360,371,1060,377,1059,1078,352,371,1058,374,1054,382,1054,1082,346,1081,346,1077,357,373,1051,1082,344,1082,350,1081,345,374,1051,380,1050,1082,347,1083,341,1082,341,`

The code I used to send commands is below.

`/*
  Example for different sending methods
  
  https://github.com/sui77/rc-switch/
  
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);
  
  // Optional set protocol (default is 1, will work for most outlets)
   mySwitch.setProtocol(0);

  // Optional set pulse length.
  mySwitch.setPulseLength(287);

  
  // Optional set number of transmission repetitions.
   //mySwitch.setRepeatTransmit(1);

   pinMode(13,OUTPUT);
  
}

void loop() {

  /* Same switch as above, but using binary code */
  
   digitalWrite(13,HIGH);
   delay(1000);  
   mySwitch.send("110101001000111011100111");
   Serial.println("done");
// 
 

 digitalWrite(13,LOW);
  

  
  delay(3000);
}`

Martin-Laclaustra, 2018-01-21:
There is no protocol 0. That means that you have to include your personalized protocol.
You can get it using:
https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino
if you have an Arduino or ESP8266
or
https://github.com/Martin-Laclaustra/433Utils/blob/verbosesnifferwithprotocollessreceiver/RPi_utils/RFSniffer.cpp
if you have a RPi.

chamithchathuka, 2018-01-21:
Hi @Martin-Laclaustra ,

Thanks very much for the quick reply.

This is the was the **output** of the AnalyzeDemo

> Received 13930215 / 24bit Protocol: 0
> data bits of pulse train duration: 28132
> proposed protocol: { 293, { 21, 3 }, { 1, 3 }, { 3, 1 }, true }
> 
> first level down
> 6011 829 
> 858 313 864 311 347 823 864 310 347 824 862 311 348 823 349 823 
> 866 308 349 822 350 822 350 823 861 311 863 310 858 313 343 828 
> 859 315 859 313 856 316 352 818 350 822 863 311 860 312 866 306 
> 
> 

I have no idea of including personalized protocol. could you please mention link for documentation or guide me how to set personalized protocol.

Martin-Laclaustra, 2018-01-23:
Edit your local file "RCSwitch.cpp" in this place:
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L81
use the number of protocol depending on which line you do add yours.

Please provide here the brand, model, and may be a photograph of your device, so that other users can benefit from your efforts.

chamithchathuka, 2018-01-29:
Hey @Martin-Laclaustra,

It worked. Thank you.
I was a Chinese branded switch which does not contain any name on it.
here I have attached some pictures of the switch so it will be helpful for others.
[image]

closed 2018-01-29 by chamithchathuka