sui77/rc-switch #199: trying to get an unsupported remote decoded

finity69x2, 2018-02-04:
I'm trying to get this remote decoded to add support for it.

I haven't been able to get the protocolless analyzer to work yet and hopefully soon that will be addressed in another issue thread. Until then maybe someone can help by going this route.

I've managed to get some additional information.

I captured a button on the remote using a rtl-sdr dongle with SDSharp and brought it into audacity.

Here is the pulse stream:

[image]

the stream above repeats 3 more times on every button press.

And I analyzed the raw signal using rtl_433. Here are the results:

```

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Detected OOK package
Analyzing pulses...
Total count:    5,  width: 142230               (568.9 ms)
Pulse width distribution:
 [ 0] count:    5,  width: 23318 [23317;23321]  (93272 us)
Gap width distribution:
 [ 0] count:    4,  width:  6409 [6298;6745]    (25636 us)
Pulse period distribution:
 [ 0] count:    4,  width: 29728 [29615;30062]  (118912 us)
Level estimates [high, low]:  15890,     50
Frequency offsets [F1, F2]:   -5179,      0     (-19.8 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...

Detected OOK package
Analyzing pulses...
Total count:    4,  width: 112619               (450.5 ms)
Pulse width distribution:
 [ 0] count:    4,  width: 23315 [23313;23322]  (93260 us)
Gap width distribution:
 [ 0] count:    3,  width:  6452 [6300;6755]    (25808 us)
Pulse period distribution:
 [ 0] count:    3,  width: 29768 [29614;30069]  (119072 us)
Level estimates [high, low]:  15913,     62
Frequency offsets [F1, F2]:   -4160,      0     (-15.9 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...

Detected OOK package
Analyzing pulses...
Total count:    4,  width: 112616               (450.5 ms)
Pulse width distribution:
 [ 0] count:    4,  width: 23316 [23315;23321]  (93264 us)
Gap width distribution:
 [ 0] count:    3,  width:  6450 [6300;6749]    (25800 us)
Pulse period distribution:
 [ 0] count:    3,  width: 29767 [29615;30064]  (119068 us)
Level estimates [high, low]:  15974,     49
Frequency offsets [F1, F2]:   -4017,      0     (-15.3 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...

Detected OOK package
Analyzing pulses...
Total count:    4,  width: 112618               (450.5 ms)
Pulse width distribution:
 [ 0] count:    4,  width: 23316 [23315;23320]  (93264 us)
Gap width distribution:
 [ 0] count:    3,  width:  6450 [6300;6751]    (25800 us)
Pulse period distribution:
 [ 0] count:    3,  width: 29767 [29615;30066]  (119068 us)
Level estimates [high, low]:  15879,     54
Frequency offsets [F1, F2]:   -4610,      0     (-17.6 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...

Detected OOK package
Analyzing pulses...
Total count:    4,  width: 112619               (450.5 ms)
Pulse width distribution:
 [ 0] count:    4,  width: 23317 [23315;23322]  (93268 us)
Gap width distribution:
 [ 0] count:    3,  width:  6450 [6299;6752]    (25800 us)
Pulse period distribution:
 [ 0] count:    3,  width: 29768 [29615;30068]  (119072 us)
Level estimates [high, low]:  15975,     51
Frequency offsets [F1, F2]:   -4576,      0     (-17.5 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...

Signal caught, exiting!

User cancel, exiting...

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
*** signal_start = 693747, signal_end = 855971
signal_len = 162224,  pulses = 259
Iteration 1. t: 11739    min: 163 (257)    max: 23316 (2)    delta 25922
Iteration 2. t: 11739    min: 163 (257)    max: 23316 (2)    delta 0
Pulse coding: Short pulse length 163 - Long pulse length 23316

Short distance: 9, long distance: 147, packet distance: 6466

p_limit: 11739
bitbuffer:: Number of rows: 5
[00] {1} 80 : 1
[01] {1} 80 : 1
[02] {101} 00 00 00 00 00 00 00 00 00 00 00 00 00
[03] {78} 00 00 00 00 00 00 00 00 00 00
[04] {78} 00 00 00 00 00 00 00 00 00 00
*** signal_start = 10872469, signal_end = 11005079
signal_len = 132610,  pulses = 4
Iteration 1. t: 23318    min: 23317 (1)    max: 23319 (3)    delta 0
Distance coding: Pulse length 23318

Short distance: 6296, long distance: 0, packet distance: 6445

p_limit: 23318
bitbuffer:: Number of rows: 4
[00] {0} :
[01] {0} :
[02] {0} :
[03] {0} :
*** signal_start = 12211687, signal_end = 12373912
signal_len = 162225,  pulses = 5
Iteration 1. t: 58296    min: 0 (0)    max: 116593 (5)    delta 653833608
Iteration 2. t: 58296    min: 116593 (5)    max: 0 (0)    delta 1418051522
Iteration 3. t: 58296    min: 116593 (5)    max: 0 (0)    delta 0
Distance coding: Pulse length 58296

Short distance: 6296, long distance: 0, packet distance: 6407

p_limit: 58296
bitbuffer:: Number of rows: 5
[00] {0} :
[01] {0} :
[02] {0} :
[03] {0} :
[04] {0} :
*** signal_start = 15964988, signal_end = 16127487
signal_len = 162499,  pulses = 5
Iteration 1. t: 58298    min: 0 (0)    max: 116597 (5)    delta 654439898
Iteration 2. t: 58298    min: 116597 (5)    max: 0 (0)    delta 1419917042
Iteration 3. t: 58298    min: 116597 (5)    max: 0 (0)    delta 0
Distance coding: Pulse length 58298

Short distance: 6296, long distance: 0, packet distance: 6475

p_limit: 58298
bitbuffer:: Number of rows: 5
[00] {0} :
[01] {0} :
[02] {0} :
[03] {0} :
[04] {0} :
*** signal_start = 18358955, signal_end = 18491568
signal_len = 132613,  pulses = 4
Iteration 1. t: 46637    min: 0 (0)    max: 93275 (4)    delta 1142603764
Iteration 2. t: 46637    min: 93275 (4)    max: 0 (0)    delta 220582066
Iteration 3. t: 46637    min: 93275 (4)    max: 0 (0)    delta 0
Distance coding: Pulse length 46637

Short distance: 6297, long distance: 0, packet distance: 6445

p_limit: 46637
bitbuffer:: Number of rows: 4
[00] {0} :
[01] {0} :
[02] {0} :
[03] {0} :
Signal caught, exiting!

User cancel, exiting...

c:\rtl433>^X
```

Hopefully this will give some more information to help me make some progress on this.

Martin-Laclaustra, 2018-02-06:
I have bad news.
That protocol (from the view in audacity) is not compatible with rcswitch.
The most you can aspire to is to write code to "play" the signal when you need. Receive it to trigger actions would be much more complex.
If you want to do the first task, you need to use [SimpleRcScanner](https://github.com/sui77/SimpleRcScanner).
With the timings, you can follow:
https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310

Get your hardware working first! You can assign the correct value to _receiverPin_, using the macro definitions "D1" - "D16", if you use Arduino IDE. For example:
`static int receiverPin = D4;`

robbymeier, 2018-02-08:
Ah, that signal looks very similar to my remote (Hunter Douglas ceiling fan). RC Switch doesn't print anything in serial monitor when I press a button on the remote. Why can RC Switch not handle this? Is it because of the preamble and that 'silence' between the preamble and the actual code? Also is there somewhere a description of the different protocols RC-Switch supports? That would also help me understand what the limitations are.

Martin-Laclaustra, 2018-02-08:
> Is it because of the preamble and that 'silence' between the preamble and the actual code?
Exactly
> Also is there somewhere a description of the different protocols RC-Switch supports?
Single gap (sync bit) between repetitions of a set of bits, each bit encoded as two levels, with different duration of the levels for 1s and 0s.

robbymeier, 2018-02-09:
Thanks for the response. Actually, I wonder why the preamble and pause is an issue. This is really just to tell the receiver to be ready for a signal. Then the signal comes. Everything else before can be ignored. Can this be done with RC-Switch. If so, is there any issue with the type of signal my remote sends out? Basically it just sends I think 66 bits where 0=100 and 1=110 with each of the 3 digits in 100 and 110 having the same length. 
Here is an image of one such sequence. All to the left is the preamble (AAAAAA), followed by a pause and then the actual signal that I need to detect.
[image]

mttstt, 2019-01-28:
This is like my segnal ! 
I solved without library in raspberryV2: https://github.com/mttstt/remote-control-shutters.
Works like a charm.
Now, I am porting on esp8266

AKTanara, 2021-08-12:
Your code is just for transmitting!