sui77/rc-switch #206: Question about Intertechno ITT-1500 and ITR-1500

ovisopa, 2018-02-24:
First of all, I thank you for this awesome library ! I started using it a week ago with some chinese 433MHz relays and remotes and everything works just fine.

As now I included 433MHz devices in my Smart Home project, I purchased some ready made 433 socket switches, I didn't chose the cheapest ones, I got a set of 3 ITR-1500 with one remote ITT-1500 , and also a separate socket switch that is for a higher load 3kw, this one with no remote.

Now my 433MHz "gateway" using a ESP8266 is always one, but it's not receiving any signal from the Intertechno remote, no matter how long I press, how should I find out what parameters I should use for the transmitter (family, group, device) ? 

Any help is appreciated.

Thanks again.

wochinge, 2018-02-27:
Might be related to https://github.com/sui77/rc-switch/issues/178

ovisopa, 2018-02-27:
@wochinge It's been a few days I'm googling about those intertechno switches and rc-switch library, and the #178  never showed up in results, but it's definitely the same issue. I will try to test the code provided as solution in that page, and post back

Thank you !!

wochinge, 2018-02-27:
@ovisopa Maybe you can tell how it worked out for you as I am struggling with the same problem.

ovisopa, 2018-02-27:
**Still struggling ?** The code provided in thread #178 did not work for you ? I still haven't tested it yet as I don't have the transmitter with me now

ZweiEuro, 2020-07-22:
@ovisopa I am one of the devs of the gateway, what exactly where you trying to do ?