sui77/rc-switch #210: Using with WeMos D1 mini

eneski, 2018-03-12:
Hello, I want to use this library with my WeMos D1 mini by first receiving the signal and then using the send sketch. How do I connect the modules to my board, like I mean which pin is used as an interrupt pin?

Attila-FIN, 2018-03-13:
Hi,
As far as I know, interrupts might be attached to all the GPIOs of the ESP8266, except the GPIO16.
I use D2 for TX and D3 for RX. Not important, but board looks like that:
https://photos.app.goo.gl/Rmwu9i9HYfaqzMG73

The Relevant code:
```
#define RF_EMITTER_PIN 4
#define RF_RECEIVER_PIN 0 
  mySwitch.enableTransmit(RF_EMITTER_PIN);
  mySwitch.enableReceive(RF_RECEIVER_PIN); 
```

1technophile, 2020-01-05:
Here is a [list ](https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1864843263)of ESP8266 compatible pins for emitting and receiving RF signals.

closed 2020-01-05 by 1technophile