sui77/rc-switch #213: Unsupported rc unit- finding codes

lrv209, 2018-03-22:
From reading other issues I can see that others have found that the RC-Switch ReceiveDemo-Advanced sketch wont capture the codes for some RC Transmitters? I have successfully used the sketch to capture the codes for a Watts Clever RC operating on 433.92mHz but I also have an Arlec unit (RC213) supposedly operating on the same frequency but am unable to capture the sent codes. I have tested the sender unit and it works fine and sends a signal that appears binary in shape when viewed on an oscilloscope although I am pretty new to all this so hoping I am seeing and saying things correctly. (I believe the binary signal may be 32bit?/). The Oscilloscope signal is hard to capture as there is also a lot of background noise being captured by the Rx so very hard to get a steady signal and doubt my ability to convert that image to a binary number anyway although I believe it can be done. I have opened the Arlec RC213 Tx and it has RC-10-V1-2 and the number 4801 printed on the board. The transmitter puck has HD R433M printed on it and the 14 pin chip has no ID at all.
I have seen a comment from Martin Laclaustra and do, I think, have the latest version of RC-Switch (2.6.2) downloaded today but cant see the ProtoAnalyzeDemo referenced and while I have found the ProtoAnalyzeDemo sketch it wont compile as it fails at Line 85 "Serial.println((mySwitch.getReceivedLevelInFirstTiming() == 0) ? "down" : "up" );".
If anyone has any suggestions for a rank beginner on all the things I am probably doing wrong please help with whatever you can suggest. It would be great to get these codes as my whole house is kitted out in Arlec RC units.

lrv209, 2018-03-27:
I am still working on a solution but have realised some of the information in my pot was ambiguous and naive so closed the post

UserNAMEPersonThingy, 2018-05-23:
Have you worked this out yet. Mine doesn't compile either.

lrv209, 2018-05-23:
Hi, I tried all the various forks hoping that someone would have come up with a solution to capture the Arlec codes but never was able to get them.  For the moment I have put it in too hard basket. Didn't have any compiling problems that I recall though so not sure what problem you are experiencing. Sorry I can't be of more help.

closed 2018-05-23 by lrv209

dinukasal, 2018-06-08:
You have to switch to the branch named `protocollessreceiver`

lrv209, 2018-06-10:
Thanks Dinukasal, I had previously tried the protocollessreceiver branch by Martin  Laclaustra if that is what you are referring to but had no joy unfortunately. I fear the rc available from Arlec in Australia must be using a protocol that just can't be read by Rc-switch in any of its branches  although it works fine with older versions of Watts Clever rc units. Haven't however tried it on the newer Watts Clever products. Still in too hard basket for me at the moment although it would be good to harvest the signal outputs from the Arlec rc. Regards

nzbart, 2018-07-09:
I've got the same Arlec sockets, and am able to get a reliable data stream, but can't figure out which branch or code to use to get rc-switch to work.

What I've done so far:

1. Captured the raw data stream:
```
404,744,400,748,400,752,392,232,924,228,916,752,400,748,404,744,404,740,400,232,916,232,924,228,10316,236,920,748,388,760,400,744,400,228,920,752,416,732,412,216,916,756,400,748,392,232,924,744,408,740,404,224,936,216,916,236,924,220,916,236,920,228,932,220,916,752,396,752,408,740,396,752,400,748,400,224,920,232,920,748,412,736,408,744,404,740,412,216,932,220,924,224,10324,228,912,756,392,756,396,748,408,224,916,756,400,744,400,228,928,744,392,760,396,232,920,752,396,752,400,224,932,220,924,224,920,228,920,232,916,232,920,228,916,756,396,756,400,744,408,740,400,748,396,232,920,232,916,748,412,740,408,740,400,748,400,228,924,224,924,228,10316,232,920,748,400,748,400,748,400,224,924,752,400,748,392,232,928,744,400,748,400,228,944,732,396,748,400,228,920,228,924,224,940,212,916,232,924,228,920,224,916,760,396,752,400,748,396,748,404,748,400,224,920,232,924,744,408,740,400,748,412,736,400,228,924,224,924,228,10312,236,928,744,392,756,412,736,400,224,920,756,404,740,400,228,928,7 44,400,7
```
[image]

It appears that the bit length is 1150 microseconds, 0 bit is 225/925, and 1 bit is 745/405 microseconds.

2. Made a stab at coming up with the protocol configuration, which looks odd compared to the existing configurations:
```
{ 58, { 0,  180 }, {  4,  16 }, {  13,  7 }, true }
```
3. Added this configuration and tried rc-switch.
4. Tried the protocollessreceiver branch mentioned above.

Note that this is the first time I have used the library, so could be committing some rookie mistakes. Any help would be greatly appreciated!

Priva28, 2018-07-09:
If you have a raspberry pi. Use pilight raw commands.

nzbart, 2018-07-09:
Thanks, @Priva28 . I neglected to mention that I'm using an Arduino.

Priva28, 2018-07-09:
That’s alright. I just thought I’d mention it as it’s how I got mine working.

nzbart, 2018-07-12:
@lrv209 I've added this wiki page detailing how I got my Arlec devices to work:

* [Description of socket protocols from different brands and models](https://github.com/sui77/rc-switch/wiki/Description-of-socket-protocols-from-different-brands-and-models#arlec-rc210-aunz-240v-socket-remote-controlled-power-outlets)