sui77/rc-switch #214: Geeetech 433mhz receiver never not working

travisbrkr1234, 2018-03-26:
I have been testing this library on an arduino uno using a [Geeetech 433mhz receiver](http://www.geeetech.com/433mhz-superheterodyne-3400-rf-link-kits-p-167.html) without luck. Any help is appreciated.

So far I have tested various pins on the duino to see if the receiver is working. From what I can tell it is(used just the digitalRead) but am having no luck getting into the block under:
```
mySwitch.available()
```
Currently the pins on the receiver look like:
+5V - 5v on arduino
GND - ground on arduino
DATA - digital pin 2 on arduino (tried pins 1-11)
Is there anything special with other types of receivers that I am missing? I have tried power on the antenna side as well.

jpascher, 2018-04-09:
I also have the same problem, so far. just for testing the right interrupt pin on the Demos i used the following skatsh: https://techtutorialsx.com/2016/12/11/esp8266-external-interrupts/
this proves that the use pin for input is working with the receiver module quite well.

Martin-Laclaustra, 2018-04-10:
Start with testing the signal you receive with this:
https://github.com/sui77/SimpleRcScanner
Post the timings here (not the image) if you still have problems.

jpascher, 2018-04-21:
I want to let other know that if the protocol is not recognized nothing will be display! Makes sure with other tolls that you use the right pins. I used instead the very rudimentary method with audacity (https://hackaday.com/2012/07/02/decoding-rf-link-using-a-pc-soundcard/) to record the bit buttern. then i have forked an other project to be able to use old eumig outlets, if some one also wants to adapt  some other protocols have a  look how this can be done. Not all Protokolls used are standard. Reverse Engineering my be needed.  rc-switsh library also may not be usable for recording, besaus the protocols Implement may not fit as in my case with this older eumig outlets. https://github.com/jpascher/EspRfRemote

travisbrkr1234, 2018-04-26:
Thanks for your responses :) I haven't had a chance to check out the SimpleRcScanner yet, but will post back once I do.