sui77/rc-switch #218: Sniffing any protocol

bepehr, 2018-04-16:
Hi
Thank you for your good library , i did test it with some common remotes and it works very good but with Car remotes or some home security remote keys it wont detect protocol to capture signal , i did my tests and receiver works and library gets the interrupts.
How can i edit library so that its start to capture signals after any sync period for specific time periods like 1 sec to capture any protocol ?

bepehr, 2018-04-17:
Yes , when it could sniff raw data then it should able to send its saved data .

Martin-Laclaustra, 2018-04-19:
Check https://github.com/sui77/rc-switch/pull/170 and https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310

flatsiedatsie, 2018-08-08:
I've built this in another project. It works great.
https://github.com/flatsiedatsie/433Switches