sui77/rc-switch #222: EMW100R (Everflourish \ COTECH \ CO/TECH \ SMJ) codes not able to be scanned

frownbreaker, 2018-04-21:
I've used the RCSwitch lib to build many projects using 433mHz transmitters all working fine. Great library! I've a couple of remotes that I could never get to work: The receive demo had no output :(

At first, I thought these units used another frequency like 315mHz but when I opened one of them it was a 433mHz transmitter. So I think it might be the encoding/chipset being used.

The remote uses EMW100R board and looks like the one in this blog (scroll down for the insides of the remote)

Blog post for photos of insides of remote 
http://kinavara.blogspot.co.uk/2015/01/remote-control-power-shitch.html

I saw this post that has this transmitter's code documented Code for EMW100R if this helps?
https://github.com/chrols/emw100

Is there any example code I could use in a receiver for transmissions from an EMW100R based transmitter? Of course, my hope would be to use the captured codes to transmit compatible codes.

Martin-Laclaustra, 2018-04-22:
Looking at your links, it seems possible to use rcswitch, and at least it is worth trying.
Capture your codes with this:
https://github.com/sui77/SimpleRcScanner
Post the timings here (not the image)

Please keep the button pressed enough time to allow several repeats of the signal in your capture.

frownbreaker, 2018-04-22:
Thanks for the help. I'm not sure what all the numbers mean are they they duration of the signal when its high then low in milliseconds? 

Anyway I **_got_** data and the normal RFSWITCH advanced recived demo that I normally use to "capture" 433RF codes would not produce any data for this remote, this app seems to detect things! Which is progress, many thanks! Not sure what to do next assume some analysis to work out the encoding being used, and perhaps some code to reproduce this scanned signal? Though I'm clueless about how to do this! I suppose some code to modulate the output of a 433mHz transmitter using these numbers would also work as a sort of "capture / playback" approach?

I connected the data pin of the 433mHz receiver to Pin2 of an Arduino Uno and VCC / GND of the reciver to the 5V / GND (It's a 5-12v tolerant board)

At first I ran the program and was shocked to get output even when I did not press any buttons! Then I looked at the sketch code and saw that this scanner counts down, reads the data coming in and outputs it before terminating. I also reread your comment _**"Please keep the button pressed enough time to allow several repeats of the signal in your capture."**_ 

I ran the program and pressed the button when the countdown got to 2 and held the button down until the output on the serial port had stopped. During this time the red LED on the remote was flickering and the remote was within a few cms of the receiver. 

When I did this I got a bunch of numbers. Each time it looked very different when I looked at the numbers. I tired putting the numbers in the spreadsheet but nothing seemed to happen (I was using Excel to open the ODS file) but I clicked the DEMO button on the github page, entered the numbers on the link to the demo page here http://test.sui.li/oszi/ and the online tool generated a waveform. I pasted in the second set of numbers which looked different as did the waveform. I noticed a "silent" part of the waveform and used the horizonal scroll bar to align the waveform start and gradicule (scale markings) and voila the waveforms appeared to match! I added a third set of serial caputures and then 3 more for the second button on the remote. 

This is the output Capture 1-3 are the First OFF button and 4-6 the First ON button on the top of the remote.

I used the viewer here http://test.sui.li/oszi/ and although the numbers looked different the I could align the waveform that the web page generated using the numbers from the serial captures.

_I had one scan where the waveform that the web page generated using the numbers looked odd, I discarted the data and rescanned this scan looked like the previous scans, perhaps it was some interference._

**Serial Capture #1**
```
5... 4... 3... 2... 1... 
1200,544,832,384,1380,428,756,472,1324,476,740,476,1312,492,1320,508,692,524,704,516,1272,528,700,524,1268,536,1268,588,584,628,596,624,1176,620,604,620,1176,620,608,612,1184,612,612,612,1188,604,16964,648,584,624,592,620,592,620,596,616,1208,612,600,648,1144,668,552,620,1244,576,632,580,1232,588,624,592,1232,588,624,592,1228,592,620,608,620,580,1220,580,640,588,1196,604,620,604,1196,604,620,600,1192,604,1196,656,556,648,1176,636,584,624,600,620,1180,616,608,616,1188,608,616,608,1184,612,612,608,1188,612,1208,608,604,612,612,608,1188,612,616,604,1192,604,1224,596,612,604,616,604,1196,600,620,604,1192,604,624,600,1196,604,624,596,1196,604,16952,672,544,660,560,652,564,644,568,648,1172,644,572,640,1180,640,576,636,1184,636,580,632,1188,628,588,624,1196,624,592,624,1192,624,592,624,600,620,1176,620,608,612,1184,616,608,612,1184,612,612,612,1188,608,1212,608,604,612,1208,608,608,604,620,604,1192,604,620,600,1204,592,628,596,1208,584,644,592,1184,612,1208,612,608,604,620,604,1196,600,624,596,1200,600,1220,600,612,600,1220,600,616,604,616,600,1200,596,1224,596,616,600,624,596,1204,592,16956,668,552,660,556,652,564,648,572,640,1176,644,572,640,1180,636,580,632,1188,632,584,628,1192,628,588,620,1196,628,588,624,1200,616,596,616,608,616,1180,616,604,620,1180,616,608,612,1188,612,616,604,1188,612,1208,612,604,608,1212,608,608,604,616,608,1188,608,620,600,1196,604,620,600,1196,604,624,596,1200,600,1220,600,612,600,624,600,1196,600,624,600,1200,596,1220,600,616,600,1220,600,612,600,624,596,1200,600,1220,600,616,596,628,592,1200,600,16952,680,540,664,548,660,560,648,572,640,1180,636,576,640,1184,636,576,636,1188,628,588,620,1196,624,596,616,1204,616,604,600,1224,596,620,580,660,564,1228,572,652,568,1228,572,648,580,1212,592,624,596,1188,616,1212,600,628,596,1208,620,584,628,596,624,1176,620,604,616,1180,616,612,612,1184,612,612,608,1192,608,1212,608,608,604,620,600,1196,604,620,600,1196,604,1216,604,612,596,1220,604,612,600,624,600,1196,600,1220,600,612,600,624,600,1200,596,16956,672,544,664,556,652,564,644,572,644,1176,640,576,636,1184,636,580,632,1188,628,588,628,1192,624,592,620,1200,620,592,620,
Reset your Arduino to scan again...
```

**Serial Capture #2**

```
5... 4... 3... 2... 1... 
1216,604,604,608,1212,608,604,608,616,608,1192,604,616,604,1196,604,620,600,1196,604,620,600,1200,596,1220,600,616,596,624,600,1196,604,620,600,1200,596,1220,604,612,600,1220,600,612,600,624,596,1200,600,1220,600,612,600,624,600,1196,600,16944,684,544,660,556,652,564,648,568,644,1176,644,572,640,1180,636,580,632,1188,632,584,628,1192,624,588,624,1196,624,592,620,1200,620,596,616,604,616,1184,612,612,612,1184,612,608,612,1188,612,612,608,1188,608,1208,612,608,604,1216,604,608,604,624,600,1192,604,620,60 [... cut ...] 20,604,616,1180,620,608,612,1180,620,604,616,1184,616,608,612,1184,612,1208,612,604,608,1208,612,604,608,616,608,1188,612,612,608,1192,604,616,608,1188,608,616,608,1192,604,1212,608,608,604,616,608,1192,604,620,600,1196,604,1216,604,612,600,1216,604,608,604,1216,604,612,604,1216,604,608,604,1216,604,612,600,16740,676,536,668,548,660,560,648,568,648,1172,644,572,640,1180,640,576,636,1184,636,584,628,1188,632,584,628,1192,624,588,628,1192,624,592,620,
Reset your Arduino to scan again...
`

[image]

Martin-Laclaustra, 2018-04-23:
> Are they the duration of the signal when its high then low in milliseconds?

yes

> I've assumed ASK

It is ASK, in particular [OOK](https://en.wikipedia.org/wiki/On-off_keying), with [RZ](https://en.wikipedia.org/wiki/Return-to-zero) line coding. So you are interpreting more bits that there really are. You need up+down for each bit.

> I suppose some code to modulate the output of a 433mHz transmitter using these numbers would also work as a sort of "capture/playback" approach?

For resending the captured signal, as a last resort, check https://github.com/sui77/rc-switch/pull/170 and https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310

> Is there a program that will take the numbers and convert them into a string of 1s and 0s?

I wrote a branch of this library that does this, but for coding that allocate the same duration to 1s and 0s, which is not your case. So it must be done manually.

Your signal has 1 sync bit + 52 data bits (check this repository's wiki). That is quite a long train of pulses and you will need to modify the source code of the library (cpp + h files) to hold the data (currently it uses a 32-bits int, you need a 64-bits one) for both, sending and receiving. I have written about it in other issues. Please search for it. Its fairly doable.

Now, your signal:
The basic unit of time, called pulselength in the library, is 605 ms for you. I just sorted your numbers, averaged each cluster and assumed simple natural number ratios between them. These ratios were 1, 2, and 28 for your device. Data bits are paired (on-off) with timings (1,1) and (1,2). Lets assume them as 1s and 0s respectively. This is arbitrary and we could have assigned them the other way round. Just stick to this decision when decoding your signal.
According to that, your protocol should look like this:
{ 605, { 1, 28 }, { 1, 2 }, { 1, 1 }, false }

Finally, your codes. Read them from the gap and discard the last bit (it is the sync bit of the next train). I interpreted the first one for you:
1111010101010101101010100101101010100110100101100110
Its: 4315972465486182 decimal.
(Some of the repeats were not equal. Check all repeats and try which ones work)

frownbreaker, 2018-04-24:
Thanks for such a complete reply! I'm reading your reply carefully and trying to do what you have suggested.

frownbreaker, 2018-04-24:
I pasted the data into a spreadsheet and get the same number (except) for result  No.1 which I'll disregard
The other results from the data are binary for 4315972465486182 which is the same result you got!

1. 1111010101010101101010100101101010100110100110101010 (Suspect)
2. 1111010101010101101010100101101010100110100101100110 (Match)
3. 1111010101010101101010100101101010100110100101100110 (Match)
4. 1111010101010101101010100101101010100110100101100110 (Match)
5. 1111010101010101101010100101101010100110100101100110 (Match)

I'm looking at sample 4 which is the OFF command..

frownbreaker, 2018-04-24:
> Your signal has 1 sync bit + 52 data bits (check this repository's wiki). That is quite a long train of pulses and you will need to modify the source code of the library (cpp + h files) to hold the [...]

Ok now I'm armed with my 2 integers that represent the 52-bit data I need to modifiy the library to use 64 bit int rather than 32-bit. I could create a send64 command so the name does not class with the existing 32bit send function?

Martin-Laclaustra, 2018-04-25:
at https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L82 replace or add your protocol:
{ 605, { 1, 28 }, { 1, 2 }, { 1, 1 }, false }

The size must be changed at:
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.h#L61
(calculate to accommodate 64 bits)
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.h#L80
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L487
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.h#L90
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.h#L170
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L573
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L609
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L90

I think I did not miss any of them. This was only a quick review of the code...
Changing these to long long (depending on the platform) would enable both, sending and receiving.

frownbreaker, 2018-04-28:
I've uploaded my changes here https://github.com/frownbreaker/rc-switch

ReceiveDemo_Advanced (No Output) When I run the example ReceiveDemo_Advanced it receives protocol1 but not my new protocol 7 (no output) 

ReceiveDemo_Simple (needed minor to print some output only one number matches manual decode)
If I compile the example ReceiveDemo_Simple Serial.print conversion is flagged as ambiguous. I tried `Serial.print(static_cast<  long long >(mySwitch.getReceivedValue()) );` but that would not compile though `Serial.print(static_cast<  long >(mySwitch.getReceivedValue()) );` does. I put in some code to provide the output.

Sample output using Remote that I want to capture / transmit with
```
Received (Cast long) -1515558486 Received (hack) 4315972465486250 / 52bit Protocol: 7
Received (Cast long) -1515558486 Received (hack) 4315972465486250 / 52bit Protocol: 7
Received (Cast long) -1515558571 Received (hack) 4315972465486165 / 52bit Protocol: 7
Unknown encodingReceived (Cast long) -1515558571 Received (hack) 4315972465486165 / 52bit Protocol: 7
Received (Cast long) -1515558486Received (hack) 4315972465486250 / 52bit Protocol: 7
Received (Cast long) -1515558486Received (hack) 4315972465486250 / 52bit Protocol: 7
Received (Cast long) -1515558571Received (hack) 4315972465486165 / 52bit Protocol: 7
```
Modifed ReceiveDemo_Simple 

```
RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
}

void loop() {
  if (mySwitch.available()) {
    
    int value = mySwitch.getReceivedValue();
    
    if (value == 0) {
      Serial.print("Unknown encoding");
    } else {
      Serial.print("Received (Cast long) ");
      Serial.print(static_cast<  long >(mySwitch.getReceivedValue()) );
      Serial.print("Received (hack) ");
      uint64_t xx = mySwitch.getReceivedValue()/1000000000ULL;
      if (xx >0) Serial.print((long)xx);
      Serial.print((long)(mySwitch.getReceivedValue() - xx*1000000000));
      Serial.print(" / ");
      Serial.print( mySwitch.getReceivedBitlength() );
      Serial.print("bit ");
      Serial.print("Protocol: ");
      Serial.println( mySwitch.getReceivedProtocol() );
    }

    mySwitch.resetAvailable();
  }
}
```

frownbreaker, 2018-04-28:
# Send Test
No luck with the send code. The receiver does not respond even standing next to the receiver.

Code from Send test 
```
#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
void setup() {
  Serial.begin(9600);
   // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10); 
}

void loop() {
  mySwitch.setProtocol(1); // Test with working reciver just to validate Transmitter is working
  mySwitch.setPulseLength(175);
  mySwitch.setRepeatTransmit(7);
  mySwitch.send("010001010001010111001100");
  delay(2000);
  // Test code for new protocol 7
  mySwitch.setPulseLength(605);
  mySwitch.setProtocol(7);
  mySwitch.setRepeatTransmit(7);
  mySwitch.send(4315972465486250 ,53);
  delay(20000);
  mySwitch.send(4315972465486165 ,53);   
}
```

Martin-Laclaustra, 2018-04-29:
Change:
`mySwitch.send(4315972465486250 ,52);`

If that correction does not solve the problem, the following procedure might help:
1. Run your code to produce your "synthetic" codes and record them with SimpleRcScanner.
2. Compare the recordings of the "native" signal with those of the "synthetic" one.

Surely you will be able to spot why it does not work.

NOTE: @sui77 I think the procedure in the discussion of this issue should be abstracted into the wiki.
@frownbreaker Maybe you could write a page with a summary into this repository's wiki? (or two, one for decoding new remotes and another for adapting rcswitch for remotes with long codes).

frownbreaker, 2018-04-29:
Many thanks, that helped! I forgot the Sych bit is covered by the library

I wrote a programme to toggle the switch on and off every 20 seconds the OFF command works so I'll use your procedure of comparing the original signal to make synthetic one. 

The receiver for this remote controls the Pond fountain outside, so testing may take longer than normal ;)
Regarding your suggestion for the wiki I've added the 2 section under background info
https://github.com/sui77/rc-switch/wiki comments welcome!