sui77/rc-switch #223: Need to Send a free sequence of pulses

jafal1, 2018-04-22:
hi  

how i can Send a free sequence of pulses and how i can add a protocol  to receive them

thanks in advance

Martin-Laclaustra, 2018-04-22:
Check https://github.com/sui77/rc-switch/pull/170 and https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310
While sending a free sequence is quite simple, you will only be able to detect sequences that conform to a certain protocol (1 sync bit + data bits encoding both 0 and 1 as high-low signal events with different timings).
Your question is vague. If you want to use and arbitrary code, just choose one of protocols already in the library and use the function send(). It will be detected at the other end.

jafal1, 2018-04-22:
hi 
thanks for answer

what is the max pulse frequency  the library can deal with it ?

 max pulse train  i mean or the max code frequency 

and  what the true or false  at the end of protocol refers to ?

do you know that  your library  is a magic its a great work and effort from you