sui77/rc-switch #227: ESP32 support

X-Dron, 2018-05-04:
PLS add ESP32 support to the interrupt handler.
#ifdef ESP8266
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define RECEIVE_ATTR ICACHE_RAM_ATTR
#elif ESP32
	#define RECEIVE_ATTR IRAM_ATTR
#else
    #define RECEIVE_ATTR
#endif
See
https://github.com/espressif/arduino-esp32/issues/855
Other cases of using #ifdef ESP8266  I did not test.

ttww, 2018-05-07:
Vote 👍

pemca, 2018-07-11:
Hi,

Working fine 👍 

I changed locally in RCSwitch.cpp To

#ifdef ESP8266
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define RECEIVE_ATTR ICACHE_RAM_ATTR
#elif ESP32
    #define RECEIVE_ATTR IRAM_ATTR
#else
    #define RECEIVE_ATTR
#endif

FROM
/*
#if defined(ESP8266) || defined(ESP32)
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define RECEIVE_ATTR ICACHE_RAM_ATTR
#else
    #define RECEIVE_ATTR
#endif
*/

And in my ESP32 Code 
#
#include <WiFi.h>
#include <PubSubClient.h>
#include <RCSwitch.h>

#define ESP32

And it works fine
/Peter

mattjarvis, 2019-12-10:
This sorts it! Cheers!

1technophile, 2020-01-05:
The current code in master branch supports already ESP32

closed 2020-01-05 by 1technophile