sui77/rc-switch #228: (Conrad) RS-200 switch is too noisy?

nothingTVatYT, 2018-05-15:
I have two sets of remotes. The newer one is working fine with RCSwitch (i.e. the Arduino module is working fine) but the older one which is labeled as "Europe Supplies Ltd. RS-200 Transmitter" and as far as I remember sold by Conrad Electronic won't trigger any output in the ReceiveDemo (_Simple and _Advanced) projects.
So I tried the protocolless branch and the SimpleScanner and found lots of really small values. Captured timings go as low as 12 with lots of 20 - 30.
From what I found on microcontroller.net the signal length should be 650 or 700.

So it looks like the sender is emitting a lot of noise - maybe too much for RCSwitch to detect the protocol. I tried to set the protocol to 2 and send some commands but the receiver does not react at all. In the end I would be happy to get the sending part working - the receiving is not that important if I could figure out the commands without a clean capture.

Has anybody gotten the RS-200 working?

nothingTVatYT, 2018-05-16:
```
16,36,28,760,36,68,28,24,24,24,20,56,36,16,72,40,24,60,16,640,28,48,16,96,20,32,32,60,32,80,24,568,20,72,8,96,16,308,16,256,12,72,8,88,40,36,76,24,284,28,32,76,20,136,24,60,24,28,88,12,1100,24,48,36,64,32,1108,40,28,24,1160,28,888,12,32,64,20,168,20,32,652,20,28,588,48,72,36,20,64,16,36,16,24,44,28,48,28,652,24,76,40,36,16,28,40,12,24,44,68,32,24,36,24,52,20,48,20,68,16,36,20,96,32,96,20,260,16,76,36,72,88,16,32,32,76,48,24,32,24,32,20,68,12,352,24,164,24,60,32,64,24,808,32,76,36,52,28,40,20,60,28,3220
,32,104,32,28,24,916,24,48,24,20,540,28,48,100,16,220,28,44,24,20,40,20,16,32,44,12,160,36,60,12,32,36,44,68,12,212,28,64,8,12,20,44,24,12,28,16,312,36,60,16,40,12,24,476,20,16,40,12,508,12,1096,20,60,28,36,40,20,36,40,44,16,200,24,112,40,88,72,20,36,20,32,24,108,16,68,24,52,8,120,28,88,8,36,28,32,100,16,36,76,52,28,56,24,32,20,12,8,28,36,36,88,20,252,8,76,20,56,20,40,12,28,32,908,24,32,52,24,512,36,56,24,44,28,48,20,24,24,120,32,6840,20,176,12,124,12,2620,16,88,8,76,20,64,48,8,40,52,20,32,24,8,64,24,44,72,92,12,44,68,40,56,8,64,60,16,20,16,32,8,60,20,32,56,16,52,24,208,8,644,36,124,28,68,20,40,24,540,100,24,580,40,20,40,20,472,32,28,32,16,192,16,392,20,20,56,24,420,56,100,12,168,16,40,8,68,16,88,32,100,8,2140,52,16,36,8,252,20,48,556,16,9780,24,188,24,44,32,84,32,36,44,16,52,40,32,56,24,36,28,44,12,260,20,56,20,60,40,44,36,12,36,8,32,128,20,8,80,108,12,392,44,8,28,16,44,28,152,24,48,72,32,44,28,56,24,120,12,284,20,60,20,40,20,204,16,716,44,16,20,8,28,20,104,24,16,24,12,100,28,72,36,48,48,12,64,32,56,4812,12,84,24,20,8,28,72,476,20,68,72,20,

```

ddTech, 2018-05-17:
Funny, I'm currently working with exactely the these sockets, published a longer post in a German forum tonight and was just about to post some informations here. 

Took me a few days but they're working perfectly well now. I was struggling with the timings first as well, but the reason was the receiver. These cheap MX-RM-5V receivers really are bad.  I've "hacked" one of the sockets, which is very easy, and used it as a receiver and got consistent results from then on. Paralell I've ordered one of these RXB6 Superheterodyne receivers and that works like a charm as well. 

this is what my Saleae was seing with the MX-receiver: 

[image]

No consistency at all. Because this is, what it sees: 

[image]

[image]

Then, while waiting for the RXB6, and because I wanted to see, what the socket actually "sees" (not sure if I had a faulty transmitter), I "hacked" the socket and read directly from it. This is, what the socket sees.
And that's what I saw then:  

[image]

a bit wavy, as modulated onto the mains cycle but perfectly consistent and usable.

Finally with the RXB6 I got this: 

[image]

But You don't need to decode it. Just give me a few hours to write down my findings then You can used that information to address these sockets.

regards

Frank

ddTech, 2018-05-17:
I've added a new "issue" for the full documentation. See #231. That should get You going.

regards

Frank

nothingTVatYT, 2018-05-17:
Many thanks, I'll try it.

nothingTVatYT, 2018-05-18:
I tried it but the socket does not react in programming mode i.e. when it's plugged in and the socket button pressed. It will stay in the learn mode.
Were you able to "teach" a socket in this mode using the code posted?

ddTech, 2018-05-18:
yes, I just did it a few hours ago about 20 times. 

When You plug it in, it is flashing slowly. This is not the program mode. It's the "I am not yet assigned" - mode.
You need to press the button on the socket for a few seconds. It then should start to flash faster. This is program mode and it will learn the next ON - signal it gets. 

Can You try this.

regards 
Frank

nothingTVatYT, 2018-05-18:
OK, so it should work and something else is wrong here.

Yeah, I know the teach mode. It's working with the handheld remote but not with the module. That's why I asked.

ddTech, 2018-05-18:
OK, Maybe Your sender has a different code. 
If You could read it, I could tell You which code Your sender sends
I had trouble with my sender first too. They are old and the contacts did not work properly. 
When put in the batteries after around 30secs without, it prompts You for the code by flashing. 
After entering the four digit key, the flashing ends. But due to bad contacts it did not end after four digits, It either needed more or less and so it was not clear what actually the code was. No problem, if You just work with handheld and socket, but if You want to control it, You need to know what the receiver expects. 

Maybe You try to recode and relearn the sender.

ddTech, 2018-05-20:
I have created a list of all possible code-combinations and a simple demo that utilizes the current RC-Switch class. Maybe this works better for You. 

See my post from today in #231 

Frank

ddTech, 2018-05-27:
good to read that. 

thanks for the response. 

Frank