sui77/rc-switch #236: Bearware WDP 302966 Protocol ?

H4R0, 2018-06-06:
I bought the following outlets: Bearware WDP 302966 https://www.amazon.de/dp/B07BQRRM4T

I like them as they are really small in comparison to all others.

Now i would like to control them using rf-switch, sadly they are not supported by default.

I recorded the pulses using SimpleRcScanner but how can i convert these to the Protocol ?

Outlet Channel A1 On:
1205,289,267,241,1206,287,214,291,1209,292,207,295,1222,287,1208,278,227,286,1201,286,224,290,276,242,1199,281,1204,283,225,288,1202,282,230,283,222,279,1214,284,1201,285,232,282,276,223,1225,284,1199,289,227,284,1199,287,227,282,1206,286,223,285,1204,285,229,281,1224,281,228,281,1202,285,227,281,1207,280,233,279,1210,280,246,284,1206,279,234,279,237,259,1211,289,219,278,1212,287,1202,283,228,283,1203,285,229,282,1206,279,233,278,1214,281,226,280,1208,277,235,274,8936,270,210,293,1205,287,1195,287,224,288,1199,290,224,288,217,291,1222,286,215,290,1206,288,218,281,1215,284,218,289,1210,285,219,285,1221,289,1217,269,239,274,1201,287,222,287,219,286,1213,283,1203,288,226,285,1200,285,225,288,219,283,1225,275,1200,284,230,280,224,281,1232,283,1205,279,234,280,1203,288,224,285,1205,280,232,278,1206,284,230,286,1203,282,231,282,1205,288,226,282,1201,285,228,286,1218,268,246,281,1205,284,238,271,226,277,1216,282,224,277,1217,283,1201,284,226,284,1221,268,233,281,1206,276,234,282,1205,284,229,283,1206,279,231,280,8920,278,216,284,1211,290,1193,289,274,246,1194,285,226,287,217,293,1202,293,212,286,1210,287,215,290,1206,290,215,290,1201,297,214,285,1232,279,1199,285,231,284,1199,289,224,285,217,292,1208,285,1200,286,280,241,1196,280,229,285,219,287,1196,287,1200,287,221,293,210,289,1226,287,1204,283,334,203,1195,283,225,287,1204,283,228,282,1208,281,228,286,1200,285,229,283,1203,284,280,238,1200,282,228,294,1190,286,244,285,1200,287,225,286,221,283,1211,285,221,286,1260,241,1197,284,226,285,1206,280,231,280,1208,283,227,282,1205,281,233,284,1198,285,274,241,8915,281,210,344,1159,290,1190,289,225,289,1198,289,221,290,215,288,1206,290,213,290,1206,289,217,348,1157,287,208,287,1209,287,221,283,1224,292,1196,290,224,285,1198,293,216,294,213,342,1164,284,1194,282,231,283,1203,284,225,288,220,282,1214,286,1201,285,228,285,221,334,1184,282,1199,285,229,289,1197,285,223,290,1203,284,226,287,1197,291,223,286,1213,261,228,291,1194,287,226,286,1201,285,229,281,1205,283,246,282,1204,285,226,283,222,394,1124,277,219,282,1216,270,1199,289,222,289,1197,285,228,282,1205,284,228,286,1210,282,228,280,1205,282,228,283

Outlet Channel 1A Off:
1208,290,216,286,1222,273,216,287,1210,277,224,286,1225,285,1199,285,227,286,1198,289,225,287,217,283,1227,271,1202,286,236,270,1206,285,227,284,219,285,1210,287,1197,286,226,285,221,284,1243,274,1198,284,232,274,1207,280,231,281,1206,280,230,283,1206,281,233,278,1222,267,226,286,1204,278,235,275,1207,284,229,285,1202,284,245,279,1206,281,232,282,221,281,1225,267,224,279,1217,262,227,278,1229,271,1207,276,234,277,1208,282,229,283,1224,277,232,277,1206,279,231,278,8925,254,220,278,1215,286,1199,283,227,285,1201,289,223,286,219,284,1232,279,218,285,1209,287,219,285,1204,290,217,291,1203,286,216,286,1228,284,1202,303,208,284,1203,281,232,281,223,278,1228,273,1198,285,231,280,1202,285,228,284,219,282,1234,264,1200,284,226,287,220,280,1236,283,1204,283,229,279,1203,283,227,288,1204,297,216,279,1201,283,232,283,1204,279,234,279,1205,281,234,277,1204,280,231,281,1206,294,237,280,1207,280,229,283,221,282,1212,281,225,279,1226,271,225,276,1217,280,1202,295,223,276,1207,278,234,275,1210,279,230,277,1211,278,234,277,8921,272,215,289,1207,288,1197,287,226,287,1210,275,221,290,215,286,1212,268,219,282,1214,285,219,284,1210,289,216,283,1210,289,216,289,1226,283,1196,283,230,287,1199,285,227,295,211,284,1210,283,1201,285,228,280,1222,287,220,287,216,286,1213,279,1203,283,232,278,224,280,1235,280,1203,284,229,282,1217,273,226,283,1201,278,238,282,1198,290,222,284,1202,285,229,282,1204,285,228,284,1217,267,233,278,1204,277,254,279,1206,279,230,279,225,282,1226,270,221,281,1216,280,223,285,1286,215,1197,277,229,284,1204,286,227,283,1203,281,232,280,1204,285,227,280,8916,274,210,293,1214,276,1198,291,223,290,1195,289,221,292,214,288,1208,292,212,288,1207,286,216,290,1219,275,214,287,1208,289,220,276,1230,286,1197,290,233,276,1202,286,224,285,220,288,1223,272,1197,291,222,290,1195,287,221,292,214,290,1206,287,1199,287,222,290,215,291,1240,270,1199,282,229,284,1203,270,231,283,1201,286,226,286,1200,285,228,281,1249,257,226,282,1202,287,222,288,1198,289,223,288,1198,286,244,284,1198,296,217,285,218,289,1223,269,222,282,1213,282,220,277,1220,281,1203,285,229,278,1207,283,226,284,1219,268,230,279,1204,289,220,288

Martin-Laclaustra, 2018-06-07:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2

H4R0, 2018-06-08:
@Martin-Laclaustra thank you for your reply.

I followed the instructions but i can't get it working. 

Receive_Demo does not print anything with my protocols.
For clarification, I have other outlets (intertechno) which work well with the default protocols

I tried the following protocol without success should be right though
{ 250, { 1, 36 }, { 1, 5 }, { 1, 1 }, false }

also the following with another pulse length does not work:
{ 300, { 1, 30 }, { 1, 4 }, { 1, 1 }, false }

Did i something wrong ?

The scans i used can be found here: https://pastebin.com/tb6ivj7m

H4R0, 2018-06-08:
@Martin-Laclaustra Thank you got it working.

closed 2018-06-08 by H4R0

janbollen, 2018-12-22:
@H4R0: I have the same problem, and tried both you suggestions. None of the seem to work..

If I add Protocol(100, 1, 40, 1, 6, 1, 0.5) to the rpi-rf, and send "./rpi-rf_send -t 7 10": neither the switch (learning mode) nor the  ./rpi-rf_receive seems to react.. Any tips?

H4R0, 2018-12-22:
@janbollen  i don't use the outlets anymore as they were unreliable with the python implementation. The timing is really important and the python implementation on the pi zero was randomly slowed down if the cpu load was over 20% (process / thread priority had no effect). Only the c solution would work reliable. My pi had a high cpu load all the time so the python implementation didn't work.

i don't even know if the receive (rpi-rf_receive) worked at all, so don't bother with it.

"./rpi-rf_send -t 7 10" <- where do you take the "10" from ? 

You have to send the corresponding code which you have to find out first, which i did using SimpleRcScanner.

Try the following:

Unpair / Unsync the outlet using the manual for all buttons that had been paired before
Sync the outlet with A1 (Remote Control set to A and Pair Button 1)

Make sure nothing runs in background of your pi, there should be no cpu load going on. (1-3% cpu usage is fine)

To turn outlet A1 On you have to send "10856588147783199061"
To turn Off an outlet send the code+256 = "10856588147783199317"

# turn on A1 (repeat this 5 times):
"./rpi-rf_send -t 7 10856588147783199061" 

# turn off A1:
"./rpi-rf_send -t 7 10856588147783199317" 

As you can see the code is quiet large which is the problem, bearware uses a 64 bit protocol and every bit has to be send perfect in time. if there is a small delay of 10-20ms the outlet will not respond.

Other Codes:
captured rf codes bearware:
# codes have been captured manually by SimpleRcScannerrpi
# converted to high / low sequence image and bits read manually
on codes (off + 256)
A1 1001011010101010010110010110011001010101010101010110100101010101 = 10856588147783199061
B1 1001011010101010010110010110011001010101010101010110100101100101 = 10856588147783199077
C1 1001011010101010010110010110011001010101010101010110100110010101 = 10856588147783199125
D1 1001011010101010010110010110011001010101010101010110100110100101 = 10856588147783199141

D1 1001011010101010010110010110011001010101010101010110100110100101 = 10856588147783199141
D2 1001011010101010010110010110011001010101010101010110100110100110 = 10856588147783199142
D3 1001011010101010010110010110011001010101010101010110100110100111 = 10856588147783199143
D4 1001011010101010010110010110011001010101010101010110100110101000 = 10856588147783199144

janbollen, 2018-12-23:
Thanks for your tips and codes.

Really strange that even after reset and pairing with A1 "./rpi-rf_send -t 7 10856588147783199061" does not turn the socket on..

What are your changes to the file "rpi_rf.py"?

I just changed the protocols to:
```
PROTOCOLS = (None,
             Protocol(350, 1, 31, 1, 3, 3, 1),
             Protocol(650, 1, 10, 1, 2, 2, 1),
             Protocol(100, 30, 71, 4, 11, 9, 6),
             Protocol(380, 1, 6, 1, 3, 3, 1),
             Protocol(500, 6, 14, 1, 2, 2, 1),
             Protocol(200, 1, 10, 1, 5, 1, 1),
             Protocol(100, 1, 40, 1, 6, 1, 0.5))
```

Out of the logger (on debug) I can see its selecting protocol 7 and sending "1001011010101010010110010110011001010101010101010110100101010101":
> rpi_rf: Using GPIO 17
> rpi_rf: TX enabled
> rpi-rf_send: 10856588147783199061 [protocol: 7, pulselength: default, length: default, repeat: 10]
> rpi_rf: TX code: 10856588147783199061
> rpi_rf: TX bin: 1001011010101010010110010110011001010101010101010110100101010101
> rpi_rf: TX disabled
> rpi_rf: Cleanup

Fyi: hardware seems to work since other protocols are working fine (tested protocol 1 with another 433mhz appliance)

H4R0, 2018-12-23:
@janbollen did you also patch rpi_rf.py to support 64bit protocols ? otherwise it does stop sending after 32 bits.

i uploaded my patched rpi_rf.py to pastebin. (probably older version)

https://pastebin.com/VueBhn21 
there are 5 protocols added, test them all.

janbollen, 2018-12-23:
Thanks again for your quick feedback and sharing your changes!

I used your file, uncommented the protocol section and ran those:

> ./rpi-rf_send -t 1 -l 64 10856588147783199061 
> ./rpi-rf_send -t 2 -l 64 10856588147783199061 
> ./rpi-rf_send -t 3 -l 64 10856588147783199061 
> ./rpi-rf_send -t 4 -l 64 10856588147783199061 
> ./rpi-rf_send -t 5 -l 64 10856588147783199061 
> ./rpi-rf_send -t 6 -l 64 10856588147783199061 
> ./rpi-rf_send -t 7 -l 64 10856588147783199061 
> ./rpi-rf_send -t 8 -l 64 10856588147783199061 
> ./rpi-rf_send -t 9 -l 64 10856588147783199061 
> ./rpi-rf_send -t 10 -l 64 10856588147783199061

Output look fine: 

> rpi_rf: Using GPIO 17
> rpi_rf: TX enabled
> rpi-rf_send: 10856588147783199061 [protocol: 6, pulselength: default, length: 64, repeat: 10]
> rpi_rf: TX code: 10856588147783199061 64
> rpi_rf: TX bin: 1001011010101010010110010110011001010101010101010110100101010101
> rpi_rf: TX disabled
> rpi_rf: Cleanup

Also double checked if the socket reacts on the A1 on / off, and it does..

No clue anymore.. Guess will have to look for other small 433 mhz sockets..

H4R0, 2018-12-23:
Hmm odd should work..

You should verify what the pi sends using SimpleRcScanner.

Do you have 2 pi's to test ? One to send with the custom protocol another one to receive.

Compare the capture of the pi and the original remote control of the outlets.

Check that the Pi sends the whole 64bits and verify that the timing is correct otherwise do adjustments to the protocol.

Maybe the codes are not the same ?

Getting the outlets to work takes a lot of time...

Im using the old Brennenstuhl as they are reliable and also fit side by side but are quiet large in height.

garglkarg, 2019-01-30:
using the C library on my raspi the protocol ({ 250, { 1, 36 }, { 1, 5 }, { 1, 1 }, false }) above won't work for me, I changed code to use long long, and used a signalLength of 64 `mySwitch.send(data.targetSignal, signalLength);` but I can't send or receive anything (in contrast to another rf-set which works fine with protocol 1 and signal length 24). 
And yes I need the c code to execute the commands not python.

codingcatgirl, 2022-02-07:
Hey, did any of you manage to figure this out? I'm trying to get it to work with esphome, which seems to be using rc_switch in the background.

garglkarg, 2022-02-08:
> Hey, did any of you manage to figure this out? I'm trying to get it to work with esphome, which seems to be using rc_switch in the background.

No, sorry - switched to the Brennstuhl devices (and denied the above int-switches ;) )

H4R0, 2022-02-24:
Meanwhile I completely switched from 433MHz sockets to wifi sockets with build in power meter.

They cost about 7€ per socket and can be flashed with other open source software like tasmota.

Works great together with home assistant / grafana dashboards.