sui77/rc-switch #237: Decode AC123-01D

onepas, 2018-06-19:
This code recorded from AC123-01D remote used https://github.com/sui77/SimpleRcScanner
Please help me decode 
@Martin-Laclaustra

300,572,300,580,304,580,300,584,304,568,300,256,620,260,628,572,296,260,628,252,620,264,620,584,300,580,300,580,296,580,296,260,624,580,296,256,620,264,620,260,616,264,616,264,616,264,616,260,616,268,608,272,620,588,288,264,612,264,612,268,616,260,616,264,616,268,616,264,616,264,616,264,612,268,612,592,288,264,612,268,616,264,616,588,292,588,284,272,612,268,612,268,616,580,300,588,288,588,300,580,292,584,268,588,5004,4988,640,584,296,268,612,588,292,264,616,264,612,260,620,592,288,588,292,264,616,260,624,588,292,584,292,588,292,588,288,588,300,584,296,580,300,580,292,268,612,260,624,584,292,272,604,264,620,272,608,588,296,584,296,576,300,580,304,256,624,584,292,260,620,260,612,272,608,272,612,260,612,276,608,268,612,268,612,268,616,584,292,268,616,268,608,268,612,272,604,268,612,272,608,272,612,264,612,268,616,264,612,596,280,272,608,268,612,272,608,588,296,592,284,268,612,272,612,268,612,588,288,596,284,592,288,588,296,588,268,588,5000,4984,652,584,288,268,612,592,288,264,612,268,616,264,616,588,296,584,288,272,612,264,624,580,288,592,288,592,288,584,300,588,288,592,292,584,292,588,292,264,608,272,616,588,288,268,608,272,604,276,616,588,292,588,288,592,288,592,284,272,612,592,284,272,604,276,608,268,612,272,608,268,608,272,608,276,604,276,608,268,608,596,284,268,608,272,612,272,604,272,604,276,608,272,608,272,604,272,612,272,608,276,608,592,280,272,604,284,600,272,612,592,288,592,280,276,608,268,604,280,612,588,288,600,284,592,288,592,292,588,256,596,4996,4992,640,592,284,272,608,596,288,268,604,276,608,272,604,600,288,592,280,272,608,272,616,592,284,588,292,588,296,588,292,588,292,588,288,592,284,592,288,268,612,268,612,592,284,272,612,272,604,272,612,588,292,588,292,588,292,592,288,264,616,592,284,272,608,268,612,268,608,272,612,268,604,272,612,268,608,272,608,276,608,592,284,276,604,268,616,272,600,280,604,276,604,268,608,284,596,276,604,276,608,276,608,588,284,276,600,284,596,276,608,600,280,592,284,276,604,280,604,272,608,592,288,596,284,596,288,596,276,596,264,596

Martin-Laclaustra, 2018-06-19:
You will not be able to use rcswitch to receive this remote, because it uses 2 sync bits. Also, for sending commands you will need to send a pre-programmed sequence of timings. Follow:
https://github.com/sui77/rc-switch/pull/170
and the links mentioned there.
Alternatively you could make a custom program just to send your sequence without using the library.
Roughly (I looked approximately into your data) and the signal is 2 sync bits + 64 data bits:
{ 290,
 { 2, 17 }, # 1st sync bit
 { 17, 2 }, # 2nd sync bit (rcswitch is not compatible with this)
 { 1, 2 },
 { 2, 1 },
 false }
Good luck!

kuchod, 2018-06-19:
didnt try to decode, but using perivar's fork of rc-switch, I did managed to send a very similar signal by code (also an AC123 remote). It shouldnt be very difficult to adapt the code to recognize it.

onepas, 2018-06-20:
@Martin-Laclaustra 
Can you explain me how to extract protocol from data?
How can you get this protocol?
{ 290,
{ 2, 17 }, # 1st sync bit
{ 17, 2 }, # 2nd sync bit (rcswitch is not compatible with this)
{ 1, 2 },
{ 2, 1 },
false }

Martin-Laclaustra, 2018-06-20:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1

You will find an answer to your question reading from point 11 on.

onepas, 2018-06-21:
@Martin-Laclaustra 
How you can find 290 value?

paperadio, 2018-06-21:
I did the decoding work almost whole day but doesn't work. kuchod's solution seems a good one, I'll try it again tomorrow.

Martin-Laclaustra, 2018-06-21:
@onepas I just guessed by looking at your numbers. I said "roughly", did not calculate. It seems a reasonable common denominator (integer divisor) of the averages of your 3 different durations. Averages from your data: 277.6, 601.9, 4994. 290 * 1, 290 * 2, 290 * 17. You can try to workout another one.
@paperadio If you read above you could have saved your time:
"**You will not be able to use rcswitch to receive this remote, because it uses 2 sync bits.**"
My branch [transmittimingsarray](https://github.com/Martin-Laclaustra/rc-switch/tree/transmittimingsarray) described in https://github.com/sui77/rc-switch/pull/170 , https://github.com/sui77/rc-switch/issues/146 , and https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310
is capable of sending an array of codes and will probably be able to control your sockets or receivers.

paperadio, 2018-06-22:
This code recorded from AC123-01 remote used https://github.com/sui77/SimpleRcScanner
5144,596,576,212,180,612,564,212,192,600,180,608,180,608,564,216,576,216,180,608,180,612,564,212,576,212,576,216,564,224,180,600,188,604,180,608,576,212,564,216,576,212,180,612,572,216,180,596,192,604,180,608,572,216,180,600,576,212,180,612,572,216,180,608,564,216,180,612,180,608,180,608,180,600,180,608,180,612,180,608,564,216,184,608,180,608,184,608,180,596,192,600,180,608,180,608,180,604,188,600,576,212,184,608,180,596,192,600,180,608,564,224,564,220,188,600,576,212,180,608,180,600,576,216,176,612,180,608,564,212,576,844
I sent it using transmittimingsarray directly, but no act on my projector screen. I guess maybe I haven't 'normalize' the data. Is there any simple method?

kuchod, 2018-06-22:
normalize the data. you can use excel to do it. then send it and capture the sent packet to compare it to the original.

Martin-Laclaustra, 2018-06-22:
@paperadio 
Well, you have to select a sensible interval of numbers (do not copy the whole signal "raw"). Normalization is advisable, but not compulsory.
From your words I guess that the problem is that you copied all the numbers, while you should copy only the numbers from one "844" (3rd number) to the next "844" (excluding it) (6th line).

paperadio, 2018-06-23:
Thank you all guys, It is working! I was totally excited when the screen reacted suddenly. My remote Model is : AC123-01. It is paired to my projector screen. 
These are some mistakes I made before everything goes well:
- I copy the wrong numbers to the array. When using @Martin-Laclaustra 's fork [transmittimingsarray](https://github.com/Martin-Laclaustra/rc-switch/tree/transmittimingsarray) to send the recorded signal, I copied 
`5144,596,...,576,844`
from results (captured by https://github.com/sui77/SimpleRcScanner), but the correct signal number should be:
`844,5144,596,...,576`, take the last number(844) to the first position before the longest timing(5144), thanks @Martin-Laclaustra 
- After one button signal goes well(after exciting for a while),When I was going to capture the other button signal,  I forgot to power off the transmitter board, so the captured results is always wrong. 
- I wrote a wrong transmit pin in code. I used NodeMCU board(esp8266 ic), so I should modify the default pin in rc-switch/examples/SendDemo, I modified it already at beginning, but it changed back when I moved back and forth in several codes.

There are some great issues which include a lot of clues: #170 , #163 , #146 , #170 

Thanks @onepas , you gave a clear issue name for help seeker.
Thanks @Martin-Laclaustra , your fork is quite useful, and your instant, detailed reply.
Thanks @kuchod , your experience of AC123 remote is also useful and make me insist on it.

Martin-Laclaustra, 2018-06-23:
> take the last number(844) to the first position before the longest timing(5144)

That is not exact. The key point is that the list of timings start with the timing of a "low" period (RF off). The other detail to pay attention to is to copy one complete cycle (or various).

paperadio, 2018-06-23:
@Martin-Laclaustra your description is much more clear. '5144' means a low signal on air, is it right? or '844' means a low signal? or in other words, correct sync protocol signal on air should be high level?

Martin-Laclaustra, 2018-06-23:
It is the order of the timings what determines whether they are high or low. The values are just timings in microseconds. Sync can be any combination of those. Practically, reading the first 5 values from your list:
1st  212  low = RF off
2nd  572  high = RF on
3rd  844  low = RF off
4th  5144  high = RF on
5th  596  low = RF off
...
If you paste your values on http://test.sui.li/oszi/ from the link in https://github.com/sui77/SimpleRcScanner you can see a graphical representation.

thomas-henry-gomes, 2018-07-01:
Hi guys,

I have the same problem: i have a "AC123-01" RF Transmitter and i can not replace it by an Arduino (by using RCSwitch library).
The RCSwitch don't recognize it, but when i uncomment the debug variable, i see this:

```
UP BUTTON:
Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 2. Delay: 2506. Delay tolerance: 192. Received Bitlength: 64. First Data Timing Index: 2
Raw timing data: 5012,572,564,192,196,564,572,196,188,572,564,192,572,192,188,572,192,572,192,560,580,188,192,568,196,568,564,192,572,192,188,572,192,576,564,188,576,188,192,572,192,568,196,556,584,180,580,184,196,572,568,188,576,188,192,572,564,200,564,188,204,560,576,188,192,572,196,556,208,556,196,564,200,560,204,548,216,544,208,556,592,176,208,552,212,540,212,552,212,548,212,552,212,540,216,544,216,552,216,544,220,532,216,548,216,548,588,176,216,536,600,164,600,168,588,176,588,168,596,172,592,172,580,184,208,544,220,544,208,552,580,220,
 failed(i=2)
```

```
STOP BUTTON (Middle):
Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 2. Delay: 2516. Delay tolerance: 198. Received Bitlength: 64. First Data Timing Index: 2
Raw timing data: 5032,556,572,184,208,552,584,180,204,560,576,180,584,180,200,560,204,564,200,552,588,176,204,556,208,556,580,176,584,180,200,560,204,564,576,180,584,176,204,560,204,556,208,544,592,176,588,176,204,560,580,176,588,176,204,560,576,188,576,180,212,548,588,176,204,560,208,544,220,544,208,552,212,552,212,540,224,536,216,548,596,172,212,552,212,540,212,548,216,548,216,544,216,536,220,544,216,548,220,544,220,532,604,160,220,544,216,544,220,532,604,164,600,168,216,544,220,532,232,532,604,160,220,544,220,532,228,532,224,540,592,204,
 failed(i=2)
```

```
DOWN BUTTON:
Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 2. Delay: 2514. Delay tolerance: 197. Received Bitlength: 64. First Data Timing Index: 2
Raw timing data: 5028,560,572,184,204,556,584,180,200,564,572,184,580,184,196,564,200,568,200,552,584,180,200,560,204,560,572,184,580,184,196,568,196,568,572,184,580,184,200,560,200,564,200,552,588,176,584,180,204,564,576,176,588,176,204,560,588,176,576,180,212,548,588,180,200,564,204,548,216,548,204,556,208,552,212,540,220,544,208,556,592,176,208,552,212,540,212,552,212,548,212,552,212,540,216,544,216,548,220,544,592,164,216,544,216,548,216,544,220,536,600,164,600,168,216,544,220,536,600,164,600,164,216,548,216,536,224,536,220,544,588,208,
 failed(i=2)
```

I put these data in http://test.sui.li/oszi/ to try to define my protocol (here 12) but it doesn't work : {570, {50, 5}, {5, 1}, {1, 5}, {2, 1}, true} ...

Please, can you help me?
I use this version of RCSwitch library: https://github.com/perivar/rc-switch/

kuchod, 2018-07-01:
I think 570 is wrong, it is more like 200. also 5,1 and 1,5 should be somthing like 3,1 and 1,3

After some tests, my coding was

    {194, {27, 3},  {3, 1}, {1, 3}, {3, 4}, false}

I had to adjust the pulselength to match the recorded signal. 

Try to generate the signal and capture it to see where it fails.

thomas-henry-gomes, 2018-07-02:
@kuchod I have already tested your coding, but i did not succeed to make it work by adjusting it...

@paperadio I also tried to do that, but did not succeed. Can the values provided above be used for this? Do you have an example of use?

Thank you both for your help.

kuchod, 2018-07-02:
did you send and capture the new signal? did you compare with the original? what differences did you notice?

thomasGomes, 2018-07-10:
@kuchod 
If i resend 4x the signal, i receive that:
What should i see?

`Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 4. Delay: 1256. Delay tolerance: 98. Received Bitlength: 64. First Data Timing Index: 1
Raw timing data: 5024,564,568,184,208,556,580,184,196,564,572,184,580,184,196,568,196,568,200,552,584,180,200,560,200,564,572,184,580,184,196,564,200,568,572,180,584,180,200,564,200,560,204,548,588,176,588,180,200,564,572,184,580,184,196,568,568,196,568,184,208,556,580,184,196,568,200,552,212,548,204,560,204,556,208,540,220,544,208,556,592,172,212,552,212,540,212,548,216,548,212,548,216,536,216,544,220,548,220,540,220,532,600,164,216,548,216,544,220,536,600,164,600,168,216,544,220,532,228,536,600,164,220,540,220,532,228,536,216,544,592,208,
 failed(i=1)`

`Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 4. Delay: 1258. Delay tolerance: 99. Received Bitlength: 64. First Data Timing Index: 1
Raw timing data: 5032,556,576,180,212,548,588,176,204,560,576,180,584,180,200,560,204,560,208,544,592,172,208,556,208,552,584,172,592,172,208,552,212,556,584,168,596,172,208,552,212,548,212,540,596,172,592,172,208,556,584,172,588,176,208,556,580,184,580,176,212,548,588,176,204,564,204,548,216,544,208,556,208,552,212,540,224,536,216,548,596,172,212,548,212,540,216,548,212,548,216,548,216,536,216,544,220,544,220,544,220,532,604,164,216,544,216,544,220,532,604,160,604,164,220,544,220,532,228,536,600,164,216,544,220,532,232,532,220,540,592,208,
 failed(i=1)`

`Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 4. Delay: 1258. Delay tolerance: 99. Received Bitlength: 64. First Data Timing Index: 1
Raw timing data: 5032,552,580,176,212,552,584,180,204,556,580,176,588,176,204,560,204,560,204,548,588,176,204,560,204,556,580,176,588,176,204,556,208,560,580,172,592,172,208,556,208,552,212,540,596,172,592,172,208,556,584,172,588,176,204,560,576,188,576,180,212,552,584,176,204,564,204,548,216,544,208,556,208,552,212,540,224,540,212,548,596,172,212,552,208,544,212,548,212,552,212,548,216,536,216,544,220,548,220,540,220,532,604,160,220,544,220,544,216,536,600,164,600,168,216,548,216,536,228,532,604,160,220,544,220,532,228,532,224,540,592,204,
 failed(i=1)`

`Testing if this is protocol 12 using 132 timings. PauseLengthInPulses: 4. Delay: 1258. Delay tolerance: 99. Received Bitlength: 64. First Data Timing Index: 1
Raw timing data: 5032,556,576,176,216,548,588,176,204,560,576,176,588,180,200,560,204,560,208,544,592,172,208,556,208,552,584,172,592,172,208,552,212,556,584,172,592,172,208,552,212,552,208,544,596,168,592,172,212,552,584,172,588,180,204,556,580,184,580,176,216,548,588,176,204,560,208,544,216,548,208,552,208,552,212,540,224,540,212,548,596,172,212,552,212,540,212,548,216,548,216,544,216,536,220,544,216,548,220,540,224,528,604,160,220,544,220,544,220,532,604,160,600,168,216,548,216,536,228,532,604,160,220,544,220,532,232,528,224,540,592,204,
 failed(i=1)`

kuchod, 2018-07-10:
after comparing the original signal posted with these, you have to increase the pulselength to get pulses with the desired length. 

* the long pulses are a little smaller than the original ones: increase the pulselength a little bit until matches.
* the short ones are about 2/3 of the size it should be: looks like the definition of 0 and 1 is incorrect.  what is your configuration?

to make it work, the first thing you will need is to send pulses of the right size, then the right code.

kuchod, 2018-07-10:
I would say you are not using the defined protocol to send the signal. this last example with 850ms pulselength looks like the 570 ones...

defining pulselength=570 and one as { 5, 1 } you should have something like 2850, 570
if you increase it to 850 it would be like 4250, 580

the signal you are capturing is like 572, 192, so it looks like it has a 200ms pulselength and 1 = { 3, 1 }

what you need to do is adjust the definition, translate the codes to 0's and 1's, send them and capture them (you will need 2 boards for this) and compare to the original captured code. if the generated one is "bigger" than the original, play with the pulselength and the number of pulses to create 0 and 1, until they are similar enough to trigger the receiver.

thomasGomes, 2018-07-10:
OK. I didn't understand. I always received the signal from my remote control. So it's normal the result is the same... 😞 
Actually i don't have 2 boards... I think i will test the @paperadio solution.
@paperadio: do you have an example of use of transmittimingsarray?

LeandroAR, 2018-07-12:
Hello there!

I took some samplings from my RF remote AC123-01 and they all look very alike, as below:

5196,540,624,160,228,540,624,160,228,548,232,544,236,532,632,152,636,148,236,540,244,536,616,156,232,548,220,556,600,184,600,172,216,568,212,564,216,560,212,556,208,568,208,572,588,192,208,560,600,184,208,568,596,188,212,556,608,172,612,172,212,568,592,180,220,560,208,572,204,576,200,568,208,572,192,580,196,580,200,568,592,192,200,584,188,588,188,576,204,572,192,588,188,588,192,568,208,568,204,572,204,572,204,560,216,564,600,184,204,572,584,192,588,200,588,196,192,592,184,584,580,192,192,596,184,592,184,596,560,200,584,236,5136,

I take the pulse length is 200, zero = {1,3} and one = {1,1}, right?

Normal   = 1100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001
Reversed = 0011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110

Arduino code:
	#include <RCSwitch.h>
	
	RCSwitch mySwitch = RCSwitch();
	
	void setup() {
	  mySwitch.enableTransmit(4);  
	  mySwitch.setPulseLength(200);
	  
mySwitch.send("0011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110");
	}
	
	void loop() {}

Either using normal or reversed bits, I cannot make it work. What am I missing here?

kuchod, 2018-07-13:
LeandroAR, the pulses are {1, 3} and {3, 1}.

Also, capture the signal you are generating to compare to the original and adjust the pulselength until the two signal matches.

LeandroAR, 2018-07-14:
Thanks Kuchod! But how can I translate the sequence into an array?
Would it be: 5196,540,624,160,228,540,624,160... = {200, {1, 3},  {1, 3}, {3, 1}, {3, 1}, {3, 1}, {3, 1}, {1, 3}, false} ?
How can I use mySwitch.send() function with the decoded result?

paperadio, 2018-07-18:
@thomasGomes I don't know whether you've captured the signal by SimpleRcScanner at first. My above conclusion seems a little confused, so you can try these steps (just described in practical, not accurate in theory):
1. using SimpleRcScanner:
using https://github.com/sui77/SimpleRcScanner to capture signal in air, while you pressing the button on your remote. The results like this: 5144,596,...,576,844
2. find one whole period number list:
the correct signal number like this: 844,5144,596,...,576
3. using @Martin-Laclaustra 's fork transmittimingsarray
install this library manually.
Add a '0' after signal number list in array.
my code as an example:
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();
unsigned int down_signal[] = {844,5144,596,576,212,180,612,564,212,192,600,180,608,180,608,564,216,576,216,180,608,180,612,564,212,576,212,576,216,564,224,180,600,188,604,180,608,576,212,564,216,576,212,180,612,572,216,180,596,192,604,180,608,572,216,180,600,576,212,180,612,572,216,180,608,564,216,180,612,180,608,180,608,180,600,180,608,180,612,180,608,564,216,184,608,180,608,184,608,180,596,192,600,180,608,180,608,180,604,188,600,576,212,184,608,180,596,192,600,180,608,564,224,564,220,188,600,576,212,180,608,180,600,576,216,176,612,180,608,564,212,576,0};

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(9600);
  
  mySwitch.enableTransmit(13);
  Serial.print("setup ok");
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW);   
  delay(500);                       
  digitalWrite(LED_BUILTIN, HIGH);    
  delay(500);                       
  
  mySwitch.send(down_signal);
  delay(5000);  
}

If you still have any problem, just leave a comment, good luck.

cnh0912, 2018-12-12:
hello.

i have a problem controlling AC123-01D

I followed @Martin-Laclaustra / @paperadio's method
1. capture signal using SimpleRcScanner 
2. using @Martin-Laclaustra 's fork transmittimingsarray, send signal

The results are as follows.

[image]

The one above is the signal I sent.
Below is the raw signal received from the AC123-01D remote control

There is a difference between the two signals interval, but I don't know why
(I sent the same signal, and I checked the Excel to see if it was a similar)

And of course, no act on my projector screen

I tried to match the two signals(changing setPulseLength, chagning customprotocol).  but it didn't work

raw signals:
5036,4992,660,576,304,256,632,572,304,256,628,256,628,252,632,576,308,576,304,252,632,252,632,576,308,576,308,572,312,572,308,248,632,252,640,568,308,576,308,576,300,260,628,252,632,576,304,256,628,252,628,256,628,252,636,576,308,572,308,576,308,572,308,576,308,576,308,248,632,252,632,252,632,252,628,252,632,252,632,252,632,576,304,252,628,256,628,256,628,252,632,252,628,256,632,248,628,256,632,252,628,256,632,572,304,256,628,256,628,580,300,256,632,252,632,576,304,252,628,256,628,256,628,252,632,576,304,256,628,256,604,580

send signals:
5036,4992,660,576,304,256,632,572,304,256,628,256,628,252,632,576,308,576,304,252,632,252,632,576,308,576,308,572,312,572,308,248,632,252,640,568,308,576,308,576,300,260,628,252,632,576,304,256,628,252,628,256,628,252,636,576,308,572,308,576,308,572,308,576,308,576,308,248,632,252,632,252,632,252,628,252,632,252,632,252,632,576,304,252,628,256,628,256,628,252,632,252,628,256,632,248,628,256,632,252,628,256,632,572,304,256,628,256,628,580,300,256,632,252,632,576,304,252,628,256,628,256,628,252,632,576,304,256,628,256,604,580,0

Please, can you help me?

gabagee, 2019-04-12:
hi 
how can i add two sync bits in protocol where as there is a provision of only 1 sync bit
thank you for any help

kuchod, 2019-04-13:
you need to use perivar's fork to be able to generate the sync bits needed to replicate this signal:

https://github.com/perivar/rc-switch/

gabagee, 2019-04-13:
my apologies for misunderstanding, i am using the same decoder  that has 2 sync bits
and the protocol i understood is 
{ 280,
{ 2, 18 }, # 1st sync bit
{ 18, 2 }, # 2nd sync bit (rcswitch is not compatible with this)
{ 1, 2 },
{ 2, 1 },
false }

samples i received are
"
602,281,595,287,598,282,598,286,597,283,596,285,596,288,599,281,600,284,603,600,274,286,602,602,281,600,276,283,601,603,278,603,278,606,279,276,604,604,279,602,276,279,580,601,4997,5018,633,598,280,279,608,598,277,279,602,282,602,278,607,600,282,602,276,279,602,282,603,602,280,601,281,601,282,599,282,602,285,597,279,601,283,598,278,282,601,279,602,282,603,601,284,598,282,599,280,277,607,600,280,601,279,280,599,281,607,600,277,280,600,282,602,281,599,284,597,283,601,283,598,282,599,285,598,282,603,603,278,279,599,283,604,276,602,281,601,282,598,284,599,283,597,285,599,283,597,287,597,283,597,287,599,607,271,284,603,603,277,606,272,283,604,604,279,602,278,602,275,282,605,601,282,601,275,280,583,599,4996,5016,636,597,280,277,607,600,277,279,607,276,602,280,608,600,281,600,277,281,601,278,604,603,281,600,280,603,283,599,281,598,283,599,282,600,282,601,278,278,601,282,603,278,604,602,282,600,281,600,278,279,605,602,282,601,277,279,604,276,608,601,277,278,601,283,600,280,601,280,601,281,600,281,601,281,598,285,597,283,601,603,275,283,598,285,597,283,598,285,597,283,596,287,598,282,596,287,597,283,597,284,598,284,597,285,600,606,274,282,608,601,277,602,275,284,605,601,278,601,282,600,275,284,604,602,281,599,277,279,580,603,4994,5015,637,596,280,280,606,598,280,278,607,273,607,275,610,600,283,598,277,279,604,278,605,602,281,599,281,600,283,599,282,600,283,597,285,598,282,598,279,279,603,280,601,278,606,600,282,601,280,601,280,278,606,599,281,601,277,282,605,275,604,601,277,280,603,280,600,284,601,280,571,280,599,283,604,277,599,285,600,284,596,285,598,281,600,281,600,5003,648,589,287,268,617,590,288,267,613,273,610,270,614,593,289,593,280,276,609,273,611,595,287,594,286,594,285,596,286,596,286,596,285,594,287,595,281,278,605,275,607,274,609,596,285,598,284,596,282,275,611,597,285,596,279,278,606,275,607,599,282,277,601,279,604,278,602,279,601,280,601,281,601,282,600,281,601,283,605,600,279,279,602,279,599,281,600,282,599,282,600,283,599,285,599,281,597,283,599,283,597,285,599,282,

which i normalized to 

5000,5000,600,600,280,280,600,600,280,280,600,280,600,280,600,600,280,600,280,280,600,280,600,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,280,600,280,600,280,600,600,280,600,280,600,280,280,600,600,280,600,280,280,600,280,600,600,280,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,600,280,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,280,600,600,280,280,600,600,280,600,280,280,600,600,280,600,280,600,280,280,600,600,280,600,280,280,600,600,

now you said to use [https://github.com/perivar/rc-switch/](url)

but i found the protocols defined in RCCswitch.cpp as 
` {1, 50, 290, {0, 0}, {1, 3}, {3, 1}, {1, 31}, false},   // protocol 1 (EV1527)
    {2, 255, 650, {0, 0}, {1, 2}, {2, 1}, {1, 10}, false},   // protocol 2
    {3, 255, 100, {0, 0}, {4, 11}, {9, 6}, {30, 71}, false}, // protocol 3
    {4, 255, 380, {0, 0}, {1, 3}, {3, 1}, {1, 6}, false},    // protocol 4
    {5, 255, 500, {0, 0}, {1, 2}, {2, 1}, {6, 14}, false},   // protocol 5
    {6, 255, 450, {0, 0}, {1, 2}, {2, 1}, {23, 1}, true},    // protocol 6 (HT6P20B)
    {7, 255, 150, {0, 0}, {1, 6}, {6, 1}, {2, 62}, false},   // protocol 7 (HS2303-PT, i. e. used in AUKEY Remote)
    {8, 255, 250, {1, 10}, {1, 5}, {1, 1}, {1, 40}, false},  // protocol 8 (Nexa)
    {9, 255, 100, {0, 0}, {6, 6}, {6, 12}, {6, 169}, false}, // protocol 9 (Everflourish Single Button)
    {10, 255, 100, {0, 0}, {6, 6}, {6, 12}, {6, 120}, false},// protocol 10 (Everflourish All Buttons)
    {11, 88, 100, {34, 34}, {5, 4}, {5, 13}, {2, 200}, false},// protocol 11 (Cixi Yidong Electronics , sold as AXXEL, Telco, EVOLOGY, CONECTO, mumbi, Manax etc.)
    {12, 26, 333, {0, 1}, {1, 2}, {2, 1}, {45, 0}, true},     // protocol 12 (CAME)
    {13, 68, 100, {0, 0}, {3, 8}, {8, 3}, {3, 100}, false},   // protocol 13 (Shi Qiong) - 32+1 bit protocol. The last bit is a closing "0"`

how can i use my protocol here? 
thank you for your help

kuchod, 2019-04-18:
you must create your own protocol and add it to this list.

if I looked in the right files, my signal looks like this:

{860, 5304, 600, 600, 200, 200, 600, 600, 200, 200, 600, 200, 600, 200, 600, 600, 200, 600, 200, 200, 600, 200, 600, 600, 200, 200, 600, 200, 600, 600, 200, 600, 200, 600, 200, 200, 600, 600, 200, 200, 600, 600, 200, 200, 600, 600, 200, 200, 600, 600, 200, 600, 200, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 600, 200, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 600, 200, 200, 600, 200, 600, 200, 600, 200, 600, 600, 200, 600, 200, 200, 600, 600, 200, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 200, 600, 600, 0};

I worked on this a lot of time ago, it is a little fuzzy now... but I think you can see the 860 pulse between several repetitions captured together. 

so my protocol is this:

    {194, {27, 3},  {3, 1}, {1, 3}, {3, 4}, false},

and I use it like this:

//up
//01011100110110001010101001111111111111101111111111110100111101110
//down
//01011100110110001010101001111111111111101111111110111100101111110

  mySwitch.send("01011100110110001010101001111111111111101111111110111100101111110");

cov89, 2020-04-15:
Thanks for all the advice here it's been really useful in getting my project running using perivar's fork and the protocol settings below:

{15, 132, 50, {99, 13}, {5, 13}, {11, 6}, {11, 101}, false }

I also wanted to add that I've reverse engineered the AC123 protocol as well it appears to work as follows:

Remote ID Byte 1 (possibly Manufacturer ID)|Remote ID Byte 2|Remote ID Byte 3|Remote ID Byte 4|Channel Byte 1|Channel Byte 2|Control Code|Checksum
-- | -- | -- | -- | -- | -- | -- | --
10100011 | 01101110 | 00010100 | 00110101 | 00000001 | 00000000 | 00001011 | 11000011

In my case the Remote ID Byte 1 did not change between several remotes so this is possibly fixed for this manufacturer.

In my case only Channel Byte 1 was used (as I only had 6 channel remotes) and it works as a binary mask LSB is channel 1 MSB is channel 8. You can send several channel simultaneously by activating multiple bits concurrently (i.e. 00111111 would send command to all 6 channels).

The command byte works as follows for my system:

UP|STOP|DOWN
-- | -- | --
00001011|00100011|01000011

Finally the checksum is calculated as the addition of Remote ID Byte 2 + Remote ID Byte 3 + Remote ID Byte 4 + Channel Byte 1 + Channel Byte 2 + Control Code. This may add up to more than 1 byte in which case the MSB are ignored. Note: This excludes the Remote ID Byte 1 which may point to that being a Manufacturer ID. 

Below is my messy test code:

`
#include <RCSwitch.h>

const unsigned long Remote=0xA362281F;

const byte CHANNEL_PADDING=B0000;
const byte UP  =B00001011;
const byte STOP=B00100011;
const byte DOWN=B01000011;

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to NodeMCU Pin D1  
  mySwitch.enableTransmit(5);
  
  // AC123 Protocol define as 15
  mySwitch.setProtocol(15);

  mySwitch.setPulseLength(50);
}

void loop() {
  Serial.println("Send Blind Command");
  sendBlindCommand(Remote,B00001111,DOWN);
  delay(2000);
  sendBlindCommand(Remote,B00001111,STOP);
  delay(2000);
  sendBlindCommand(Remote,B00001111,UP);
  delay(2000);
}

void BlindUp(unsigned long Remote, byte channels){
  sendBlindCommand(Remote,channels,UP)
}

void BlindDown(unsigned long Remote, byte channels){
  sendBlindCommand(Remote,channels,DOWN)
}

void BlindStop(unsigned long Remote, byte channels){
  sendBlindCommand(Remote,channels,STOP)
}

void sendBlindCommand(unsigned long Remote, byte channels,byte command){
  byte RemoteByte1=Remote >> 24;
  byte RemoteByte2=Remote >> 16;
  byte RemoteByte3=Remote >> 8;
  byte RemoteByte4=Remote;
  byte Checksum=RemoteByte2+RemoteByte3+RemoteByte4+channels+CHANNEL_PADDING+command;
  char SendCodeChar[64];
  int bitPos=31;
  for (int i = 0; i <= 31; i++) {
    if bitRead(Remote,bitPos) SendCodeChar[i]='1';
    else SendCodeChar[i]='0';
    bitPos--;
  }
  bitPos=7;
  for (int i = 32; i <=39 ; i++) {
    if bitRead(channels,bitPos) SendCodeChar[i]='1';
    else SendCodeChar[i]='0';
    bitPos--;
  }
  bitPos=7;
  for (int i = 40; i <=47 ; i++) {
    if bitRead(CHANNEL_PADDING,bitPos) SendCodeChar[i]='1';
    else SendCodeChar[i]='0';
    bitPos--;
  }
  bitPos=7;
  for (int i = 48; i <=55 ; i++) {
    if bitRead(command,bitPos) SendCodeChar[i]='1';
    else SendCodeChar[i]='0';
    bitPos--;
  }
  bitPos=7;
  for (int i = 56; i <=63 ; i++) {
    if bitRead(Checksum,bitPos) SendCodeChar[i]='1';
    else SendCodeChar[i]='0';
    bitPos--;
  }
  mySwitch.send(SendCodeChar);
}
`

beng295, 2020-10-22:
Hi @cov89 I am working on something very similar with the same 6 channel remotes and your findings have helped me tremendously thank you!  

I do have one question for you regarding the checksum being larger than 1 byte.  You said that in this case the MSB are ignored... do you mean in the result of the addition? or the MSB in each of the bytes being added?  I have tried most of these combinations without any success.  

Channel 6 DOWN command for example....
10100011 01101110 00010100 00110101 00100000 00000000 00100011 +checksum
adding the last 6 bytes gives 100011010, but ignoring/dropping the MSB to give 00011010 has not worked for me.

Am i doing this properly according to your findings?  Thanks for your help!

cov89, 2020-10-25:
Hi @beng295,

I've put together a Google sheet which does the calculation for you have a look here see if it explains the logic better: [https://docs.google.com/spreadsheets/d/1t9-dtnJ07j9UewccKceUL7Vq9sveMTze6_38FMwlgR0/edit?usp=sharing](url)

Hope that helps.

beng295, 2020-10-25:
Hey @cov89,

That helps me out a bunch, thank you so much.  I had a couple of bytes transposed...  I guess that's what I get for trying to do the math by hand.  I finally have a reliable working setup.

Thanks Again!

nemerdaud, 2020-11-04:
@cov89 ,

**You did a great job** ! Your reverse engineered was perfect. I will give only one more tip to the people with difficults decoding this remote control: it's really difficult to find the patter/sequence that represent the button command itself. My tip is when your are running the SimpleRcScanner, push the button frequently giving no chances to the noises present in air take place int the samples. The result, you can put in a notepad (with text wrapping) and replacing "," by tabs. It's easier to find the pattern repeating. It will always start and end by a high value.

lmpipaon, 2021-09-15:
@cov89

great job !!!!

in C the strings end with a 0.

In the sendBlindCommand function you have to change:

char SendCodeChar [64];

by

char SendCodeChar [65];
SendCodeChar [64] = 0;

without this it sends data until it finds a 0

JuMalIO, 2024-05-08:
Hi, I am trying to read remote with `SimpleRcScanner.ino` with "Nucleo-64" board, and getting strange codes:
29029,101,17335,137,1031,68,1292,52,120,360,909,98,1259,300,8701,476,894,24,51,146,357,61,144,294,2190,209,67,478,11805,208,1418,123,39,29,2521,477,251,17,606,486,100,150,5418,162,1121,111,2686,82,690,546,432,146,7040,754,419,283,282,70,181,215,252,1568,5902,1721,110,3839,17,17,35,542,153,8868,472,2045,666,209,266,221,119,1554,1722,541,887,97,317,62,262,353,336,47,1216,138,1162,356,1973,169,1929,39,694,664,2179,372,64,749,3207,143,467,34,3765,235,1899,22,1638,17,359,368,207,327,2705,150,1018,17,7012,237,4574,17,725,273,2651,453,10381,20,12287,62,6206,108,724,140,16439,59,33114,114,13226,33,22616,530,13231,24,49703,128,16449,101,16493,33,52570,70,33077,115,13627,21,16544,194,2656,210,30100,59,114350,88091,178,3847,306,1171,57,247,28,1938,183,2680,226,1659,291,877,476,231,376,160,963,980,433,17,1117,38,234,125,596,230,3692,26,1285,602,1853,18,1101,17,1353,873,66,64,1708,1028,440,794,28189,56,16546,90,33050,117,33057,47,2797,345,13398,25,2813,98,78357,18,1589,149,2722,25,13677,139,2687,187,13564,69,16500,129,12298,152,50,971,560,3555,343,248,663,118,202,22,77,959,30,109,202,205,1682,70,4193,179,859,247,117,239,236,189,747,149,1348,614,595,258,549,353,474,78,1922,1079,467,339,607,95,1319,45,2742,131,6122,118,3483,358,17,176,13210,118,4277,60,8342,49,2980,17,810,17,2798,101,8722,146,7639,33,13676,43,12612,72,3838,66,16519,215,2654,104,13599,81,31405,110,18147,47,33125,30,3158,415,15884,142,32619,40,13236,149,620,113,16482,104,16466,229,2649,139,16453,33,13675,18,16589,25,2834,17,13680,35,16522,82,49798,75,15602,17,120358,71284,437,11828,535,2784,196,867,154,86,166,1192,146,902,514,19,1043,101,613,387,2531,1183,349,2213,278,466,613,635,1454,168,2196,373,2308,27,1300,57,700,259,1895,363,57,216,108,16,2188,299,282,435,289,289,643,147,70,159,201,116,303,163,2211,196,8057,85,891,50,1804,260,13485,85,2743,182,12736,109,723,69,2765,184,96456,95,49653,61,82760,136,33020,241,33020,29,16530,81,33087,144,2984,59,21336,128,8517,62,2765,147,23217,17,6945,136,33038,164,2677,77,30246,33,5241,93,44382,109,2734,140,33044,115,30164,96,16496,181,15551,40,807,73,2798,49,13662,25,16537,134,2837,53,28511,81,18133,39,2824,83,13634,21,

If I paste it to http://test.sui.li/oszi/ it displays corrupt image. Any ideas why?
I have tried many times and still the same.

JuMalIO, 2024-05-09:
Switched receiver power to 3.3v and started receiving data!

483,71,800,87,790,97,785,98,775,435,445,112,771,114,766,119,768,440,5168,4854,797,415,477,67,814,401,478,76,791,96,792,91,784,427,453,432,441,441,449,425,478,404,484,71,792,420,470,413,488,393,496,49,824,63,813,399,491,62,817,397,495,386,496,50,822,64,820,64,830,52,826,397,480,65,813,401,482,400,492,391,477,80,790,421,476,78,799,79,809,73,801,420,460,86,808,76,816,66,805,81,808,74,807,77,819,62,828,54,808,80,804,80,813,68,813,71,801,85,787,427,454,100,793,418,442,115,773,438,438,118,780,430,448,107,783,102,778,105,778,106,785,424,468,76,804,82,796,89,806,412,5194,4840,774,430,455,99,793,418,462,85,794,92,792,91,791,418,480,401,482,403,469,413,493,390,501,43,834,382,496,386,506,377,511,41,831,56,806,408,472,75,800,422,468,412,484,63,806,82,790,95,797,85,804,407,466,80,801,411,489,391,511,372,514,38,819,398,494,49,838,47,831,53,818,397,481,73,807,78,816,57,824,60,806,81,793,93,788,98,786,95,778,100,777,107,777,107,789,92,786,98,777,435,435,120,787,421,458,89,802,419,450,97,802,407,496,56,816,70,796,82,796,89,799,412,480,72,811,74,782,107,772,436,5168,4856,793,411,466,89,805,408,464,83,802,82,793,92,794,414,473,410,473,411,465,409,485,399,467,89,795,417,461,421,479,395,494,60,806,81,802,409,482,63,806,407,504,376,508,47,812,68,810,74,815,67,811,400,499,45,817,401,492,389,516,365,505,50,801,417,458,90,794,89,797,86,788,423,456,92,797,86,801,83,788,97,788,87,789,95,787,97,788,95,798,84,789,88,797,86,808,73,809,75,800,413,472,81,795,418,473,71,808,406,463,94,785,428,447,109,775,110,779,94,780,105,777,441,449,96,790,96,772,114,770,439,5171,4832,778,428,453,94,797,414,471,84,790,97,775,110,775,433,453,428,466,410,475,407,484,400,465,93,808,399,494,390,496,385,500,44,844,39,842,373,511,42,827,388,502,380,506,48,816,72,794,84,798,86,801,408,484,70,797,418,455,426,471,411,482,64,809,414,463,83,802,83,814,67,816,394,500,53,809,70,812,72,816,66,817,68,804,80,

Still figuring out how to decode it... Dont understand how to get pulse lenght...

JuMalIO, 2024-05-09:
So using this https://test.sui.li/oszi/ graphsI have managed to decode message:
```
Remote ID Byte 1 (possibly Manufacturer ID)    10100011
Remote ID Byte 2                               11101110 
Remote ID Byte 3                               01011000
Remote ID Byte 4                               01011101      
Channel Byte 1                                 00010000
Channel Byte 2                                 00000000    
Control Code                                   10001011  
Checksum                                       00111110
```

So my remote id seems to be `0xA3EE585D`
Then I am trying to send the code using this:

https://stackoverflow.com/questions/61429480/decoding-433mhz-ac123-01-signal-transmitter-not-receiving-controller-info
```
// Transmitter for AC123 - Manufacturer ID 1
#include <RCSwitch.h> // Use this lib variant: https://github.com/perivar/rc-switch/

const unsigned long Remote = 0xA3EE585D; //0xA362281F;

const uint8_t CHANNEL_PADDING = B0000;
const uint8_t UP = B00001011;
const uint8_t STOP = B00100011;
const uint8_t DOWN = B01000011;

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  // Transmitter is connected to Pin 5 change to your setuo
  mySwitch.enableTransmit(5);
  // AC123 Protocol define as 15
  mySwitch.setProtocol(15);
  mySwitch.setPulseLength(50);
}

void loop() {
  Serial.println("Send Command DOWN");
  sendCommand(Remote, B00001111, DOWN);
  delay(2000);
  Serial.println("Send Command STOP");
  sendCommand(Remote, B00001111, STOP);
  delay(2000);
  Serial.println("Send Command UP");
  sendCommand(Remote, B00001111, UP);
  delay(2000);

  Serial.println("Send Command ?");
  sendCommand(Remote, B00001000, B01010101);
  delay(2000);
  
  while(true) {} 
}

void comUp(unsigned long Remote, uint8_t channels) {
  sendCommand(Remote, channels, UP);
}

void comDown(unsigned long Remote, uint8_t channels) {
  sendCommand(Remote, channels, DOWN);
}

void comStop(unsigned long Remote, uint8_t channels) {
  sendCommand(Remote, channels, STOP);
}

// This is the part you'll have to adept to your hardware
void sendCommand(unsigned long Remote, uint8_t channels, uint8_t commandR) {
  uint8_t RemoteByte1 __attribute__((unused)) = Remote >> 24; // possibly Manufacturer ID not used at the moment
  uint8_t RemoteByte2 = Remote >> 16;
  uint8_t RemoteByte3 = Remote >> 8;
  uint8_t RemoteByte4 = Remote;
  uint8_t Checksum = RemoteByte2 + RemoteByte3 + RemoteByte4 + channels + CHANNEL_PADDING + commandR;
  char SendCodeChar[64];
  uint8_t bitPos = 31;
  for (uint8_t i = 0; i <= 31; i++) {
    if bitRead(Remote, bitPos) SendCodeChar[i] = '1';
    else SendCodeChar[i] = '0';
    bitPos--;
  }
  bitPos = 7;
  for (uint8_t i = 32; i <= 39 ; i++) {
    if bitRead(channels, bitPos) SendCodeChar[i] = '1';
    else SendCodeChar[i] = '0';
    bitPos--;
  }
  bitPos = 7;
  for (uint8_t i = 40; i <= 47 ; i++) {
    if bitRead(CHANNEL_PADDING, bitPos) SendCodeChar[i] = '1';
    else SendCodeChar[i] = '0';
    bitPos--;
  }
  bitPos = 7;
  for (uint8_t i = 48; i <= 55 ; i++) {
    if bitRead(commandR, bitPos) SendCodeChar[i] = '1';
    else SendCodeChar[i] = '0';
    bitPos--;
  }
  bitPos = 7;
  for (uint8_t i = 56; i <= 63 ; i++) {
    if bitRead(Checksum, bitPos) SendCodeChar[i] = '1';
    else SendCodeChar[i] = '0';
    bitPos--;
  }
  
  Serial.println("Checksum");
  Serial.println(Checksum);

  mySwitch.send(SendCodeChar);
}
```

But does not seem to work...

JuMalIO, 2024-05-09:
Its kind of working!
Protocol was missing in https://github.com/perivar/rc-switch/
I have added it and it seems to work.
{15, 132, 50, {99, 13}, {5, 13}, {11, 6}, {11, 101}, false }
Though I am controlling 4 blinds and some of them triggers some don't, maybe I need connect better antenna for this.

JuMalIO, 2024-08-15:
Hi, I am having an issue with retransmission. I am controlling 4 roller blinds and want them to close open to the specific saved position. But it seems that transmission sometimes fails, sometimes one roller blind opens to saved position another does not move, another opens fully... and there is no way to get status from the roller blinds I guess... increassing retransmission would work to trigger all blinds but also would open fully (like long press on remote). Is there a way to know roller status? Or safely allow retransmission without acting like long press = open fully?

Current solution I guess to have open fully possition saved at the point I want rc switch to open. But I will lose ability to open full window if I want... basically saved possitions are useless... and I can have just open or closed...

mgwoj, 2024-08-15:
I gave up and bought https://www.zigbee2mqtt.io/devices/ZM-AM02_cover.html#zemismart-zm-am02_cover which works great.

JuMalIO, 2024-08-18:
I did not gave up :D
Here are results:
https://github.com/perivar/rc-switch/pull/7
I have used logic analyzer and sniffed real data from remote:
[image]
`{16, 132, 50, {100, 12}, {5.6, 12}, {12, 5.6}, {12, 100}, false }, // protocol 16 (AC123)`
@cov89 was close :)

I have also attached remotes antenna instead of any custom antennas:
[image]
[image]

Not sure if Antenna or code changes did the trick. Might be both. Now transmitter works like regular remotes, no errors so far. All blinds are mowing together in sync.