sui77/rc-switch #241: Wrong timing on send data for protocol 2

pemca, 2018-07-08:
Hello,

I'm trying to get rc-switch working with an Danish 220V Wireless Outlet (Sartano Model 51085DKX10) and Arduino UNO. My test setup is one UNO with receive program running and one UNO with SendDemo running.

When i measure the received data from the Remote i got measured data and protocol 2.
When i send the exact same data in SendDemo my power outlet dont switch on.
 
If i Decode the Raw data i see timing is not the same, and i suspect that protocol 2 has some fault.

Could you please help or give a hint.

Please find attached test files with measured data.

[Ticket.txt](https://github.com/sui77/rc-switch/files/2173628/Ticket.txt)
[Test.pdf](https://github.com/sui77/rc-switch/files/2173629/Test.pdf)

Thanks
Peter Calum

Bredahl, 2019-03-25:
I am also trying to get rc-switch working with an Danish 220V Wireless Outlet (Sartano Model 51085DKX10)
without any luck.
I can only get it to work when I use my old Silvercrest remote sending 24 bit codes.
Trying with the original Sartano remote sending 32 bit codes I can't get it working.
Jørgen Bredahl.

Bredahl, 2019-03-26:
This is the raw data read with ReceiveDemo_Advanced from the Sartano remote.
Is it the data in the end that causes my problems sending to the Sartano Model 51085DKX10
[image]
Mr Bredahl.