sui77/rc-switch #242: Configuring for Arlec Model RC210 (Australia / New Zealand)

nzbart, 2018-07-09:
Note this was cross posted to the now-closed #213.

I've got the same Arlec sockets, and am able to get a reliable data stream, but can't figure out which branch or code to use to get rc-switch to work.

What I've done so far:

1. Captured the raw data stream:
```
404,744,400,748,400,752,392,232,924,228,916,752,400,748,404,744,404,740,400,232,916,232,924,228,10316,236,920,748,388,760,400,744,400,228,920,752,416,732,412,216,916,756,400,748,392,232,924,744,408,740,404,224,936,216,916,236,924,220,916,236,920,228,932,220,916,752,396,752,408,740,396,752,400,748,400,224,920,232,920,748,412,736,408,744,404,740,412,216,932,220,924,224,10324,228,912,756,392,756,396,748,408,224,916,756,400,744,400,228,928,744,392,760,396,232,920,752,396,752,400,224,932,220,924,224,920,228,920,232,916,232,920,228,916,756,396,756,400,744,408,740,400,748,396,232,920,232,916,748,412,740,408,740,400,748,400,228,924,224,924,228,10316,232,920,748,400,748,400,748,400,224,924,752,400,748,392,232,928,744,400,748,400,228,944,732,396,748,400,228,920,228,924,224,940,212,916,232,924,228,920,224,916,760,396,752,400,748,396,748,404,748,400,224,920,232,924,744,408,740,400,748,412,736,400,228,924,224,924,228,10312,236,928,744,392,756,412,736,400,224,920,756,404,740,400,228,928,7 44,400,7
```
[image]

It appears that the bit length is 1150 microseconds, 0 bit is 225/925, and 1 bit is 745/405 microseconds.

2. Made a stab at coming up with the protocol configuration, which looks odd compared to the existing configurations:
```
{ 58, { 0,  180 }, {  4,  16 }, {  13,  7 }, true }
```
3. Added this configuration and tried rc-switch.
4. Tried the protocollessreceiver branch mentioned above.

Note that this is the first time I have used the library, so could be committing some rookie mistakes. Any help would be greatly appreciated!

Martin-Laclaustra, 2018-07-09:
Check the wiki: https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2
Your remote uses 32 data bits + 2 sync bits, but fortunately one of the sync bits is formatted like a data bit so you can send 33 data bits + 1 sync bit. You will need to modify the library.
Your protocol is approximately:
{ 385, { 1,  17 }, {  1,  2 }, {  2,  1 }, false }
Good luck!

nzbart, 2018-07-10:
@Martin-Laclaustra That worked, thanks!

I was able to create a "blink" app using:

```
mySwitch.send("011101101101100000001111100111100"); 
delay(1000); 
mySwitch.send("011101101101100000001110100111110"); 
delay(1000);
```

The second-to-last bit is the on/off bit. The previous 31 bits appear to be the switch's ID, but I wasn't able to arbitrarily program the switch with a new code. Perhaps there's a checksum encoded in the number?

What's the best way for me to make it easier for others to use these switches? Should I submit a pull request with the configuration for these devices?

Martin-Laclaustra, 2018-07-10:
> That worked, thanks!

I am glad to hear.

> I wasn't able to arbitrarily program the switch with a new code. Perhaps there's a checksum encoded in the number?

I guess part of the bits are "brand invariable". I have not tested this with any brand, but if you have several buttons on a set of sockets or sockets from different sets (but same brand and model), you could probably record and deduce which are the constant bits, and you would probably be able to reprogram your sockets within the variable range.

> What's the best way for me to make it easier for others to use these switches? Should I submit a pull request with the configuration for these devices?

Unfortunately there is no coordinated effort to create a database of sockets. You could drive the initiative, if you have the time. You do not need to submit a pull request. You can just create a new wiki page in this repository an call it "description of socket protocols from different brand and models". Bellow I list the items that I think that are relevant.
If you do not want to go so far, I would suggest at least to document here the best way possible your socket with:
- brand
- model
- (maybe link to manufacturer and seller)
- photographs of the socket and remote (sometimes you can find them in the web, but do not just link them, upload the images here because the web is volatile).
- (if you are skilled) photographs of the inner parts... and describe the sender and receiver chipsets. (same thought about the web).
- If the sockets have it: a label with a number/identifier of socket set within the model.
- codes of every button.
These two last items would help to figure out what each bit means (when other owners share also their data).

nzbart, 2018-07-12:
@Martin-Laclaustra I've made an attempt at adding a new wiki page:

* [Description of socket protocols from different brands and models](https://github.com/sui77/rc-switch/wiki/Description-of-socket-protocols-from-different-brands-and-models).

Unfortunately, I don't have permission to upload images.

Martin-Laclaustra, 2018-07-12:
Great job! Thank you! I will edit some minor details in the wiki page when I get the time.
I think that you can upload images here. May be then you can use the link to them in the wiki.