sui77/rc-switch #245: HT680 remote

SloCompTech, 2018-07-15:
Hi,
I have HT680 remote and I can't seem to get it working, has anybody got any idea how I could get it working ?

So far I found following:
- [PDF docs](https://files.elv.com/service/manuals_hw/18346_Datenblatt_HT600.pdf)
- [Some project on github](https://github.com/igorok107/HT680)
- [Tutorial on identifying remote type](https://www.solidremote.com/blog/identify-rf-remote-control/)
- [Code in rtl433](https://github.com/merbanan/rtl_433/blob/9e6d99cd6a50a77181ff11722669ba5eae14d1d4/src/devices/ht680.c)
- Data recoreded with RTL-SDR
    ```
Detected OOK package	@ 2018-07-15 18:55:21
2018-07-15 18:55:21 :	HT680 Remote control
	Tristate code:	 XXXXXXXXXXXXXXXXXXXX
	Address:	 0xXXXXXX
	Button 1:	 
	Button 2:	 
	Button 3:	 
	Button 4:	 PRESSED
Analyzing pulses...
Total count:   41,  width: 11981		(47.9 ms)
Pulse width distribution:
 [ 0] count:   29,  width:    85 [81;89]	( 340 us)
 [ 1] count:   12,  width:   184 [181;194]	( 736 us)
Gap width distribution:
 [ 0] count:   12,  width:   113 [111;116]	( 452 us)
 [ 1] count:   28,  width:   212 [204;217]	( 848 us)
Pulse period distribution:
 [ 0] count:   11,  width:   198 [196;201]	( 792 us)
 [ 1] count:   10,  width:   397 [394;400]	(1588 us)
 [ 2] count:   19,  width:   296 [294;300]	(1184 us)
Level estimates [high, low]:  16204,   5222
Frequency offsets [F1, F2]:   19938,      0	(+76.1 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 85, long_limit: 0, reset_limit: 218, sync_width: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {52} xx xx xx xx xx xx xx 
   ```

Martin-Laclaustra, 2018-07-17:
read the wiki and provide https://github.com/sui77/SimpleRcScanner output (the list of captured timings).

SloCompTech, 2018-07-17:
Thanks, I'll report back when I gather all the things