sui77/rc-switch #247: RCSwitch working when pins are directly connected, but not when used with RF modules.

Tvde1, 2018-07-19:
Hi, I am using an attiny85 as remote and a nodemcu as a receiver and flashed the example sketches on both and I have corrected the pin numbers.

Only the problem is, this works:
[image]
Now the serial monitor shows that the receiver receives data.
And this doesn't work:
[image]
If I'm correct, it should behave the same. I don't expect RF transmission to be 100% reliable of course but nothing shows up.

I am using [these](https://www.aliexpress.com/item/433Mhz-RF-wireless-receiver-module-transmitter-module-kit-for-Arduino-Raspberry-Pi-AVR-ARM/32416047991.html?spm=2114.search0104.3.2.17c072dbsx0uDx&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10344_10068_10342_10343_10340_10341_10696_10084_10083_10618_10304_10307_10820_10821_10301_10843_10059_100031_10103_10624_10623_10622_10621_10620,searchweb201603_2,ppcSwitch_5&algo_expid=6b6aa2ee-2fa6-44f7-8cf4-31c894ea99d2-0&algo_pvid=6b6aa2ee-2fa6-44f7-8cf4-31c894ea99d2&transAbTest=ae803_2&priceBeautifyAB=0) transmitters. (433 Hz)

I also have [these](https://www.aliexpress.com/item/433-mhz-RF-Wireless-Transmitter-and-Receiver-superheterodyne-UHF-ASK-remote-control-Module-low-power-For/32832359980.html?spm=2114.search0104.3.207.c6541fadzJ1CLK&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10344_10068_10342_10343_10340_10341_10696_10084_10083_10618_10304_10307_10820_10821_10301_10843_10059_100031_10103_10624_10623_10622_10621_10620,searchweb201603_2,ppcSwitch_5&algo_expid=f46d4d8f-9f23-4f22-9776-c8ac25a2549c-29&algo_pvid=f46d4d8f-9f23-4f22-9776-c8ac25a2549c&transAbTest=ae803_2&priceBeautifyAB=0) but I wouldn't know how to connect the transmitter since it has an extra pin.

Tvde1, 2018-08-06:
Is there no one that can help me?

abellisco, 2019-04-18:
Hi, not sure if this comment will help you at this point.
I am testing the same receiver you show in your first line, and not knowing your "physical" setup it is difficult to have a proper idea of a solution.
But perhaps this helps: I am testing it with a standard 433Mhz remote (the ones usually used for garage doors openers), with the same antenna you show.
Surprisingly, the receiver works (receives properly a 24bit code) only when it is quite close (1 meter away).  This remote is working with a device that is several meters away from my desk, so...
In your case, I would first test the receiver with a standard 433Mhz remote, to reduce the "unknowns"
If it works, then try the ATtinny, and then if you need a better distance, change the antenna and try again.
The radio environment is very noisy, when you see on your Serial Monitor that RCSwitch has got one RC code decoded, it actually received MANY, and discarded many of them.  I guess the number of repetitions can provide you some help here, not sure...
Rgds

crolopez, 2020-06-14:
I have exactly the same problem.