sui77/rc-switch #250: how to use this with orange Pi

earth08, 2018-08-01:
Dear,
I have tried this with orangePi+2e but without any success,
after giving command sudo ./RFSniffer, nothing shows,

is there any workout for orangePi+2e
please guide

Martin-Laclaustra, 2018-08-02:
Have you looked at: https://github.com/ninjablocks/433Utils ?

earth08, 2018-08-03:
Yes i have already tried, but i have installed wiringop as some site/users of opi suggested to installed it.
Can you suggest me some way.
If you require some data i can provide, but please guide me thanks

Martin-Laclaustra, 2018-08-03:
Sorry. No experience with orangePi.
I would suggest setting it up step by step: first make sure that you can light up an LED and read a physical switch attached to GPIOs. (you may need to run your compiled programs with sudo to access GPIO. If so, the same applies to rcswitch.).
When you have that, then you could read a stream of values from the receptor attached to GPIO: I did the port of SimpleRCScanner to Rpi. You can use it:
https://github.com/sui77/SimpleRcScanner/tree/master/htdocs/Raspberrypi
If that works, you can climb up the ladder by interpreting the reads, following information that you can find in the wiki.
Good luck!

serafisnz, 2018-08-31:
I've written a C version of rf_send that works fine on my Orange Pi PC under Armbian. It uses wiringOP to control a GPIO pin connected to one of those cheap 433MHz transmitters, to operate various RF switches around the house.

While I'm only an amateur at this stuff, I'd be more than happy to make the source code available if anyone wants it.

Usage is: _rf_send [-h] CODE [-g GPIOPIN] [-r REPEATS] [-t PROTOCOL] [-p PULSELENGTH] [-b BITS]_

The only mandatory argument is _CODE_ (the decimal code to transmit ). It uses Protocol 1 as a default, and all other arguments have appropriate defaults for the protocol specified. 

I use an Arduino with  rcswitch's  ReceiveDemoAdvanced script (or ProtocolAnalyzeDemoscript from https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver/examples/) to sniff the protocols and codes sent by any RF remote switch I want to use.

As I only have protocol-1 RF switches, I can't test the actual operation of the other protocols, but I can sniff them fine with the Arduino so I'd expect them to work ok.

Cheers!

WH76, 2018-11-23:
I'm interested!

serafisnz, 2018-12-03:
> I'm interested!

Here's the source code for rf_send.c. Unzip it to your download folder (e.g. /home/<user>/download). It has dependencies on WiringOP, which you can install as follows:
> git clone https://github.com/zhaolei/WiringOP.git -b h3
> cd WiringOP/
> ./build

Then compile rf_send.c with:
>cd /home/<user>/downoad
> gcc rf_send.c -lwiringPi -o /usr/local/bin/rf_send

Because WiringPi needs root, you'll need sudo to run rf_send, e.g:
> sudo rf_send 12345

For help type:
> rf_send -h 

Full  description of operation can be found in the source code comments, 

Cheers.

[rf_send.zip](https://github.com/sui77/rc-switch/files/2637849/rf_send.zip)

closed 2018-12-13 by earth08

codermapuche, 2021-06-11:
@serafisnz do you have the code for receiver module? I want to send a code with your rf_send and receive it with receive module in another opi

serafisnz, 2021-06-11:
No sorry, all of my RF receivers use Arduinos so I never needed a Pi receiver.

Regards, Jon.