sui77/rc-switch #251: Issue with NodeMCU and MX-RM-5V - cannot get it working

stich86, 2018-08-08:
hi guys,

i'm trying to sniff a 315mhz static code remote with NodeMCU v2 and MX-RM-5V module (version at 315) without success.

It seems that the NodeMCU don't found the board. I've used different GPIO (D0, D1, D2, D3, D7). Also putting the module on D0 (should be GPIO 16), pressing the button on my remote I see the onboard led on the NodeMCU blinking, this is confirmed by a muli-meter, where attaching a cable to DATA and one to GND (pushing a button on my remote) increate voltage from 0.08V to 1.8V.

NodeMCU and the MX module are powered at 5V using an external PSU. I've also tried to simulate code sent using the trasmitter coming with the module but without success.. so i think there is something missing or very wrong.

As sketch, i'm using ReceviveDemo_Simple.ino

Hope someone can help me.

Thanks!

Snatchfold, 2018-08-29:
Hi
I've the same problem

NodeMCU ESP8266 DEVKIT V4 doit.am

Arduino IDE
Board: "NodeMCU 1.0 (ESP-12E Module)"
Flash Size: "4M (1M SPIFFS)"
Debug Port: "Disabled"
Debug Level: "None"
IwIP Varian: "v2 Lower Memory"
VTables: "Flash"
CPU Frequency: "80 MHz"
Upload Speed: "115200"
Erease memory: "All Flash Content"
Port: COM4

rc-switch library with ReceviveDemo_Simple.ino
 
and modified source from:

`/*
  Simple example for receiving
  
  https://github.com/sui77/rc-switch/
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
}

void loop() {
  if (mySwitch.available()) {
    
    int value = mySwitch.getReceivedValue();
    
    if (value == 0) {
      Serial.print("Unknown encoding");
    } else {
      Serial.print("Received ");
      Serial.print( mySwitch.getReceivedValue() );
      Serial.print(" / ");
      Serial.print( mySwitch.getReceivedBitlength() );
      Serial.print("bit ");
      Serial.print("Protocol: ");
      Serial.println( mySwitch.getReceivedProtocol() );
    }

    mySwitch.resetAvailable();
  }
}`

to

`/*
  Simple example for receiving
  
  https://github.com/sui77/rc-switch/
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(115200);
  mySwitch.enableReceive(D3);
}

void loop() {
  if (mySwitch.available()) {
    
    int value = mySwitch.getReceivedValue();
    
    if (value == 0) {
      Serial.print("Unknown encoding");
    } else {
      Serial.print("Received ");
      Serial.print( mySwitch.getReceivedValue() );
      Serial.print(" / ");
      Serial.print( mySwitch.getReceivedBitlength() );
      Serial.print("bit ");
      Serial.print("Protocol: ");
      Serial.println( mySwitch.getReceivedProtocol() );
    }

    mySwitch.resetAvailable();
  }
}`

Can someone help me please?

Schneemann2, 2018-12-03:
Same problem here with both ReceiveDemo Examples.
Sometimes it works , after a reset or plug it out an in again it doesnt work anymore.

jordaofranca, 2025-05-07:
I found a solution. The pin needs to have the interrupt function to be able to receive the signal

```c++
/*
  Simple example for receiving
  
  http://code.google.com/p/rc-switch/
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();
// Defina o pino da interrupção
const int pinInterrupt = D2;

void setup() {
  Serial.begin(115200);
  // Configura o pino como entrada
  pinMode(pinInterrupt, INPUT_PULLUP); 
  // Configura o pino como interrupção (RISING = interrupção no sinal de alta para baixa)
  digitalPinToInterrupt(pinInterrupt);
  mySwitch.enableReceive(D2);  // Receiver on inerrupt 0 => that is pin #2
  delay(1000);
  Serial.println("receiving");
}

void loop() {

  if (mySwitch.available()) {
    
    Serial.println( mySwitch.getReceivedValue() );
    Serial.print("Received ");
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( mySwitch.getReceivedProtocol() );

    mySwitch.resetAvailable();
  }
}
```