sui77/rc-switch #252: Problems adding a new protocol

aussieW, 2018-08-16:
Hi 

I am trying to add a new protocol for the Heller iRemote (also sold under the OWL brand).
[image]

 I have captured and analyzed the signal and added the following line to RCSwitch.cpp
{ 447, { 0, 8 }, {1, 2}, {2, 1}, false },               // iHeller

The timings generated by rc-switch seem spot on when I compare to the original remote. The screenshot below show the two signals in audacity.

[image]

The screen shot shows the following signal being sent
mySwitch.send("1100000100111000000010110000100100");

The first two bits should be ones but as highlighted in the screenshot, they are coming out as zeros.  

Any suggestions on how to address this issue.

Thanks
aussieW

Martin-Laclaustra, 2018-08-21:
You are sending 34 bits.
The way the library sends "text" binary is by converting it to a 32 bits int (variable called "code"), by bitshifting:
https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L474-L476
Your first 2 bits just "fall off" the integer and are lost. Thus, they are send as 0s.
You should change the library lines to cope with 64 bits codes.
See:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2

aussieW, 2018-08-24:
Thanks. For the info. I found that the instructions for handling 64bit codes on https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2 did not work for me. I ended up appyling the changes in detailed here https://github.com/sui77/rc-switch/pull/221 and now transmit is working.

closed 2018-08-24 by aussieW

aussieW, 2018-08-24:
Just for completeness of the record, I got this remote to work on the unmodified 32 bit version rc-switch by using the following protocol, which combines the last bit and the sync bit together.
{ 447, { 1, 10 }, { 1, 2 }, { 2, 1}, false }

I also got it to work under a modified 64 bit version of rc-switch (after applying the patches from issue https://github.com/sui77/rc-switch/pull/221) using the following protocol
{ 447, { 0, 8 }, { 1, 2 }, { 2, 1}, false }