sui77/rc-switch #261: RC Switch on ESP32

jpadie, 2018-10-14:
Hi 

I am getting kernel panics when using the lib with an ESP32.  

the backtrace decodes as follows:
```

PC: 0x40080fba: RCSwitch::receiveProtocol(int, unsigned int) at /var/folders/9g/v7p0hv_x2h12tx0fg8wzvrv40000gn/T/arduino_build_84018/sketch/radio.h line 749
EXCVADDR: 0x00000000

Decoding stack results
0x40080fba: RCSwitch::receiveProtocol(int, unsigned int) at /var/folders/9g/v7p0hv_x2h12tx0fg8wzvrv40000gn/T/arduino_build_84018/sketch/radio.h line 749
0x400810c7: RCSwitch::handleInterrupt() at /var/folders/9g/v7p0hv_x2h12tx0fg8wzvrv40000gn/T/arduino_build_84018/sketch/radio.h line 825
0x40081215: __onPinInterrupt at /Users/justinadie/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/esp32-hal-gpio.c line 219

```

ESP32 is defined 

A serial trace shows this

```
Radio code received and waiting to be picked up: 	3818400
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
16:38:04.457 -> Core 1 register dump:
16:38:04.457 -> PC      : 0x40080fba  PS      : 0x00060034  A0      : 0x800810ca  A1      : 0x3ffc0b90  
16:38:04.457 -> A2      : 0x00000001  A3      : 0x00000032  A4      : 0x3f402230  A5      : 0x3f402230  
16:38:04.491 -> A6      : 0x00000008  A7      : 0x00000001  A8      : 0x80081482  A9      : 0x3ffc0b80  
16:38:04.491 -> A10     : 0x089a5579  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc66fc  
16:38:04.491 -> A14     : 0x0000000c  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
16:38:04.491 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
16:38:04.491 -> Core 1 was running in ISR context:
16:38:04.525 -> EPC1    : 0x400621c6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080fba
16:38:04.525 -> 
16:38:04.525 -> Backtrace: 0x40080fba:0x3ffc0b90 0x400810c7:0x3ffc0bc0 0x40081215:0x3ffc0bf0 0x40081bc9:0x3ffc0c10 0x400621c3:0x00000000

```

so the panic is occurring _after_ the radio has correctly received and logged the received code.

there is a thin wrapper class controlling the app's interaction with the radio.  this is as follows:

```
class Radio{
  private:
     RCSwitch rc;
      
  public:
    void begin(uint8_t radioPin);
    void loop();
    void reset();
    unsigned long getCode();
    int available();
    unsigned long code;
    
};

void Radio::reset(){
  code = NULL;
}

unsigned long Radio::getCode(){
  return code;
}

void Radio::begin(uint8_t radioPin){
  Serial.println("beginning radio");
  rc = RCSwitch();
  rc.enableReceive( radioPin );
}

int Radio::available(){
  if(rc.available()){
    Serial.print("Radio code received and waiting to be picked up: \t");
    code = rc.getReceivedValue();
    Serial.println(code);
    rc.resetAvailable();
    return 1;
  }
  return 0;
}
```

any thoughts on a fix would be gratefully received.

thanks
Justin

abellisco, 2019-04-18:
the problem seems to be in the Interrupt routine of the gpio handler 
Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/esp32-hal-gpio.c line 219
if you are using the Arduino IDE, instead of 
rc.enableReceive( radioPin );
try
rc.enableReceive(digitalPinToInterrupt(radioPin))
in the Arduino boards you map the interrupt to the PIN directly, in the ESP32 you have to 
use digitalPinToInterrupt 

hope this helps

twinking92, 2019-05-09:
Hi,
i am also getting core panics on my esp32.
`Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000026  PS      : 0x00060031  A0      : 0x40081664  A1      : 0x3ffbe7a0
A2      : 0x00000004  A3      : 0x3ffbfec4  A4      : 0x00000000  A5      : 0x00000010
A6      : 0x3ffbfcec  A7      : 0x3ffb0060  A8      : 0x80080e7c  A9      : 0x80000020
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x800d0bd0  A13     : 0x3ffb1f70
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000024  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe
Core 1 was running in ISR context:
EPC1    : 0x00000026  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400826d9

Backtrace: 0x00000026:0x3ffbe7a0 0x40081661:0x3ffbe7c0 0x400d1bb3:0x3ffb1fb0 0x40087911:0x3ffb1fd0`
I tried the above solution using the `digitalPinToInterrupt` macro. This is my Code:
`#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  Serial.println("booting");
  mySwitch.enableReceive(digitalPinToInterrupt(16));
}

void loop() {
  if (mySwitch.available()) {
    Serial.print("Received ");
    Serial.print( mySwitch.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( mySwitch.getReceivedProtocol() );
    mySwitch.resetAvailable();
  }
}`

abellisco, 2019-05-11:
hi twinking92.
I compiled your code but I had no issues downloading it.  It did not reboot.
I did not run any test, I just downloaded it.
It seems to work.
If you want to go more deeply, here follows the ExceptionDecoder output of the error you report

PC: 0x00000026
EXCVADDR: 0x00000024

Decoding stack results
0x400d1bb3: initApbChangeCallback at /home/abel/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/esp32-hal-cpu.c line 46
0x40087911: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
regards

twinking92, 2019-05-15:
Hi, thanks for you reply. I just found out, that i do not have any problems when i use the Arduino IDE instead of using PlatformIO.

abellisco, 2019-05-15:
glad to hear that.
rgds

1technophile, 2020-01-05:
As the problem is solved with Arduino IDE I close the issue

closed 2020-01-05 by 1technophile