sui77/rc-switch #268: Data sent does not trigger relay but looks the same from receiver

DavidVentura, 2018-12-02:
I have connected a sender and receiver pair to my arduino leonardo; with the receive demo I decoded the pattern used to send data per channel.

But sending data from the arduino does not trigger the receiver; yet the arduino sensor sees the same values coming from the remote control and from my code

Sent from the arduino
```
Received 4276500 / 24Bit  // Protocol: 1- Delay: 335
Received 4276501 / 24Bit  // Protocol: 1- Delay: 336
```
Sent from the control
```
PRESS NOW:Received 4276500 / 24Bit  // Protocol: 1- Delay: 335
``

I have tried various adjustements to the `pulseInterval` and the only change is re received Delay, the receiver still does not trigger. I have tried putting my emitter 1cm away from the receiver and still nothing.

How can I debug this?

Martin-Laclaustra, 2018-12-04:
A couple of suggestions:
- Try increasing the number of repeats
- Compare both signals by inspecting them more in detail with https://github.com/sui77/SimpleRcScanner

DavidVentura, 2018-12-04:
I basically played with the code to get output that matched what i saw from the control but i still am not sure on how to get the first line as it changes slightly every time i use it:

code:
```
long shifted = (base_value | controls[1] | state_off) >> 1;
  mySwitch.send(0, 65);
  mySwitch.send(0, 65);
  mySwitch.send(0, 65);
  mySwitch.send(0, 57);
  mySwitch.send(0, 57);
  mySwitch.send(0, 57);
  mySwitch.send(base_value | controls[1] | state_off, 24);
  mySwitch.send(base_value | controls[1] | state_off, 24);
  mySwitch.send(base_value | controls[1] | state_off, 24);
  mySwitch.send(base_value | controls[1] | state_off, 24);
  mySwitch.send(base_value | controls[1] | state_off, 24);
  mySwitch.send(shifted, 23);
```

Output of the code

```
*** signal_start = 7008802, signal_end = 7239924, signal_len = 231122, pulses_found = 521
Iteration 1. t: 171    min: 85 (484)    max: 258 (37)    delta 10634
Iteration 2. t: 171    min: 85 (484)    max: 258 (37)    delta 0
Pulse coding: Short pulse length 85 - Long pulse length 258

Short distance: 98, long distance: 269, packet distance: 2650

p_limit: 171
bitbuffer:: Number of rows: 12 
[00] {66} 00 00 00 00 00 00 00 00 00 
[01] {66} 00 00 00 00 00 00 00 00 00 
[02] {66} 00 00 00 00 00 00 00 00 00 
[03] {58} 00 00 00 00 00 00 08 00 
[04] {58} 00 00 00 00 00 00 00 00 
[05] {58} 00 00 00 00 00 00 00 00 
[06] {25} 41 44 14 00                : 01000001 01000100 00010100 0
[07] {25} 41 44 14 00                : 01000001 01000100 00010100 0
[08] {25} 41 44 14 00                : 01000001 01000100 00010100 0
[09] {25} 41 44 14 00                : 01000001 01000100 00010100 0
[10] {25} 41 44 14 00                : 01000001 01000100 00010100 0
[11] {24} 41 44 14                   : 01000001 01000100 00010100 
```

DavidVentura, 2018-12-04:
Apparently my thing is using the 'waveman' protocol, as rtl_433 can decode it trivially.. I'll see if i can get it to work tomorrow

hycday, 2019-02-03:
@DavidVentura : so did you solve your issue ? 
I have a similar behavior : sending signal that is correctly received from my PCB, but not from my relay/outlet (even when putting it into "registration" mode)
i also noticed that the signals has 5 variants that loop, but not sure what to do with it then...