sui77/rc-switch #269: Cotech switches with Kangtai protocol

thallaa, 2018-12-10:
Hi,

I've been struggling with these Cotech switches I recently bought from Clas Ohlson (North-European chain selling everything for home and garden etc). With some tips from various forums (https://forum.pimatic.org/topic/1291/new-switch-cotech-from-clas-ohlson mainly), I've learned these devices use a protocol referenced as the Kangtai protocol.

This page has source code for decoding the protocol with RTL USB stick: https://github.com/merbanan/rtl_433/blob/a8f438a5fb52f24f854368b581a08aa413bb7654/src/devices/kangtai.c

The page also contains description of the protocol, which also seems to contain a form of "encryption". With these lookup tables that part should be doable with ESP8266, but I would first need the data in.

Here's an excerpt of the protocol:

>  * Protocol:
>  * Start bit = 375us high, 2250us low
>  * followed by 24 bits of data
>  * One = 1125us high, 375us low
>  * Zero = 375us high, 1125us low
>  *
>  * Data layout:
>  * 16 bit address of transmitter
>  * 2 bit rolling code (00, 01, 10, 11)
>  * 1 bit device command (1 = On, 2 = Off)
>  * 5 bit unit address (00000 = group, 11111 = factory use only)
>  * Data is 'decrypted' with a lookup table

Here are the captured timings for A ON. Capture 1:
`473,282,1221,293,1213,1051,469,1042,464,291,1221,1047,463,1047,482,1029,476,1035,476,279,1235,1033,470,1041,474,281,1231,282,1222,291,1222,292,1222,295,2341,300,1205,1054,482,273,1224,291,1219,294,1208,1054,466,1045,482,1029,472,284,1223,290,1223,1042,473,1038,469,287,1221,1045,468,1043,477,1034,480,1031,491,263,1235,1034,503,1009,478,278,1231,282,1233,281,1219,294,1220,297,2343,297,1205,1056,468,288,1236,277,1218,296,1214,1048,469,1042,474,1037,471,284,1224,290,1223,1042,477,1034,481,274,1232,1036,472,1039,471,1039,475,1037,475,279,1233,1035,484,1028,474,282,1228,285,1221,293,1230,282,1217,302,2336,304,1213,1047,473,282,1222,293,1214,298,1217,1046,470,1040,470,1041,474,281,1222,291,1224,1042,471,1039,480,276,1229,1038,477,1034,480,1031,477,1035,476,280,1228,1038,481,1029,478,278,1229,284,1229,285,1221,292,1223,2934,7173,436,1073,943,593,418,1089,426,1091,426,1089,928,589,926,591,924,597,416,1087,428,1093,925,596,920,592,419,1096,923,596,919,589,927,594,920,598,415,1103,916,598,918,596,415,1100,416,1101,415,1095,421,1107,2927,7168,441,1075,942,578,433,1088,428,1079,436,1091,927,596,918,590,926,586,425,1083,432,1092,928,596,919,589,424,1087,932,598,916,591,925,593,922,599,414,1109,909,597,919,601,411,1095,421,1098,417,1094,428,1096,2933,7171,439,1080,936,584,428,1083,432,1089,427,1080,939,581,934,588,927,583,429,1093,422,1095,925,587,928,601,411,1093,926,600,915,591,924,599,917,593,420,1098,920,594,922,597,415,1100,416,1106,411,1098,418,1108,2927,7172,438,1077,939,584,428,1087,429,1083,432,1108,910,591,924,586,929,587,426,1093,422,1091,928,594,921,595,416,1101,920,590,925,595,921,597,918,593,419,1098,921,602,914,595,418,1096,419,1098,417,1097,420,1208,292,2345,296,1212,1049,465,292,1221,291,1215,301,1209,300,1215,1047,469,1042,465,1045,471,284,1230,1038,470,1041,481,274,1227,287,1218,1048,472,1039,471,284,1224,290,1221,1044,469,1041,474,282,1226,288,1216,297,1216,296,1228,291,2343,297,1224,1036,462,295,1220,293,1217,296,1224,287,1221,1042,466,1045,472,1038,468,287,1221,1048,469,1041,474,281,1222,293,1220,1044,472,1039,476,280,1229,285,1221,1044,471,1040,472,283,1226,287,1221,293,1216,296,1224,294,2338,304,1207,1053,461,294,1224,290,1225,288,1209,304,1208,1053,466`
Capture 2:
`577,434,1078,437,1084,431,1080,436,1078,941,581,933,583,931,577,435,1087,932,580,935,593,418,1092,423,1088,932,599,916,587,424,1091,424,1090,930,605,909,592,419,1094,423,1089,426,1097,419,1202,298,2334,306,1205,1056,468,288,1209,304,1208,306,1216,1045,456,300,1212,301,1211,302,1208,1055,457,1053,464,1045,468,1042,464,1048,476,1034,468,1043,471,285,1220,1046,479,276,1218,296,1225,288,1217,296,1208,304,1203,309,1211,307,2331,307,1200,1059,468,288,1212,303,1207,304,1203,1059,461,294,1218,295,1216,298,1210,1052,466,1044,466,1044,466,1045,467,1044,470,1040,470,1041,467,289,1222,1044,485,270,1223,291,1219,292,1221,294,1215,297,1214,297,1214,304,2325,315,1202,1056,467,289,1213,301,1220,292,1205,1057,472,283,1222,292,1223,289,1212,1051,466,1045,465,1046,462,1048,478,1033,467,1044,467,1044,473,282,1226,1041,473,283,1224,289,1220,294,1212,300,1210,303,1207,305,1211,305,2336,305,1199,1059,464,294,1215,298,1214,298,1209,1053,462,294,1218,294,1220,293,1218,1046,471,1039,459,1052,471,1039,471,1040,471,1039,472,1040,469,285,1222,1046,471,284,1223,291,1221,291,1220,294,1214,298,1211,301,1215,2940,7166,443,1091,925,579,432,1078,437,1079,436,1082,937,577,434,1085,430,1082,433,1089,930,582,933,588,927,594,921,589,925,585,931,588,926,593,419,1090,929,596,417,1093,422,1097,419,1098,417,1092,424,1083,433,1094,2940,7172,437,1070,946,573,438,1079,435,1079,437,1082,936,577,435,1090,425,1086,430,1083,935,580,934,586,929,587,928,589,927,583,931,592,923,595,416,1089,931,589,422,1098,418,1093,423,1091,425,1093,423,1089,427,1098,2935,7164,445,1072,944,579,433,1088,427,1085,431,1073,945,573,437,1080,436,1086,429,1081,938,578,937,594,920,587,928,582,932,592,923,596,920,591,422,1101,917,592,419,1099,417,1092,424,1093,423,1089,427,1098,418,1094,2940,7160,450,1075,939,574,438,1080,435,1082,433,1079,939,573,439,1086,429,1083,433,1081,938,580,934,583,932,591,924,588,927,590,925,592,923,598,414,1097,922,598,414,1092,424,1090,425,1089,427,1090,426,1086,430,1199,300,2335,306,1200,1061,452,305,1212,300,1211,302,1203,309,1212,300,1205,307,1200,311,1194,1065,464,293,1228,284,1226,1038,461,295,1213,1053,464,1046,471,284,1223,1044,466,1044,466,1045,471,284,1221,291,1219,294,1218,296,1216,303,2331,308,1207`
Capture 3:
`467,1044,468,287,1222,292,1218,1046,464,1047,467,289,1225,288,1217,1048,467,1043,470,286,1234,279,1217,297,1213,300,1221,298,2332,307,1201,1058,459,297,1216,296,1213,301,1213,299,1213,1049,464,1046,466,1045,461,296,1228,1037,462,1048,471,285,1220,293,1214,1051,467,1044,469,286,1231,282,1221,1045,469,1042,463,292,1226,287,1217,297,1211,300,1218,2939,7167,443,1072,943,577,434,1082,433,1080,436,1077,438,1082,937,582,932,583,932,584,429,1091,928,583,931,593,419,1110,406,1093,925,587,928,588,424,1091,425,1088,931,589,925,595,417,1100,416,1096,420,1092,422,1096,2940,7162,448,1084,931,580,432,1077,438,1078,437,1076,439,1088,931,581,934,592,922,584,428,1085,933,586,929,587,425,1088,428,1093,925,590,925,599,411,1096,422,1092,927,600,914,594,418,1093,424,1089,425,1089,428,1096,2939,7167,444,1063,951,586,426,1078,437,1092,423,1076,439,1073,945,579,936,583,931,587,426,1086,932,582,933,593,419,1093,422,1098,921,591,923,585,428,1089,427,1091,928,586,928,592,421,1094,421,1091,425,1087,429,1096,2939,7163,446,1069,946,586,425,1098,418,1094,421,1081,435,1082,936,588,927,585,929,591,421,1081,937,591,924,586,426,1100,416,1095,924,589,926,586,426,1090,425,1092,927,588,927,588,424,1093,425,1089,425,1089,426,1200,301,2342,298,1208,1054,458,297,1216,297,1212,301,1206,1057,462,295,1219,293,1209,303,1216,1048,457,1053,462,1048,470,1041,461,1049,465,1046,468,1043,469,286,1225,1043,466,290,1222,291,1216,297,1214,299,1216,297,1212,299,1214,304,2325,315,1198,1060,460,297,1211,303,1207,304,1208,1054,470,286,1216,298,1205,307,1208,1054,466,1045,462,1049,471,1039,466,1045,468,1043,472,1038,480,276,1232,1034,475,281,1226,288,1218,294,1215,298,1218,294,1207,306,1218,300,2329,309,1203,1057,471,285,1218,296,1212,300,1202,1060,465,291,1215,298,1226,286,1217,1047,460,1049,463,1048,469,1041,471,1040,472,1039,480,1031,471,284,1227,1040,473,282,1231,284,1216,297,1212,301,1212,300,1219,293,1212,305,2329,311,1200,1058,472,285,1213,302,1209,302,1221,1042,458,297,1214,300,1208,304,1206,1057,462,1048,465,1046,463,1048,474,1037,464,1046,471,1040,469,286,1226,1041,469,288,1232,280,1228,287,1214,298,1213,299,1215,298,1211,2943,7168,443,1065,950,577,435,1076,439,1083,432,1084,935,580,432,1077,438,1084,432,1083,936,582,933`

How can I get this going?

thallaa, 2018-12-11:
I did more digging. The timings don't quite seem to match the quoted protocol but seems very similar. Comparing several pulses I first see a 2900us high, followed by 7200us low and then 24 bits with about the mentioned 375us and 1125us widths.

Here are 4 ON pulses:
[image]

And 4 OFF pulses:
[image]

Looking at pulses (ONE is 1125us high + 375us low and ZERO is 375us high + 1125us low) I get these bits for ON:
010001110011011110110000
010000111011001100110000
010001000111111101000000
010001110011011110110000

and these for OFF:
010001110011011110110000
010000111011001100110000
010001000111111101000000
010001110011011110110000

Bits seem to differ quite a lot so the so called encryption is probably there but I still don't have understanding how to get these pulses out from rc-switch to start processing them.

thallaa, 2018-12-11:
Oh I did not mention all details of the device. The back of the remote says
Cotech 36-6010/18-1255
Type: 50074
It has four channels, A-D.

The switch says:
Cotech 36-6010
Type: 51058X10

Frequency is 433.92MHz

thallaa, 2018-12-11:
I re-read the wiki page on adding a new remote and I think I got a bit further.

Looking at the raw numbers alone, there seemed to be quite many numbers around 600, but when both high and low lengths summed, they very closely match 1500. So probably this is just inaccuracy on my HW setup (Wemos D1 Mini _clone_ and a very cheap 433 receiver).

So when looking raw numbers in Excel, I got these groups popping up: 288, 493, 1086, 2576 and 7171.

But if assuming the 375 would be the greatest common divisor, I figured the more accurate numbers would be 375, 1125, 3000 and 7125.

So, if I've understood this right, the protocol would be:
{375, {8, 19}, {3, 1}, {1, 3}, false}

Does this look sensible? I'm yet to test this.

And btw. I hope you don't mind if I document my steps here.

thallaa, 2018-12-12:
I tried modifying the library with this but the codes keep getting detected as protocol 3 and randomly protocol 5. Also, sending back the same codes does not work.

This variation of pulse lengths is bothering me, also the fact that I have been using another set of this exact same hardware successfully for about a year now - it works flawlessly with a set of Nexa switches using protocol 1.

johnou, 2018-12-17:
@thallaa sorry to add noise but I am in the same boat with a different remote, what software did you use to calculate the timings? I managed to capture some samples with Audacity but I am not sure how to progress with it.. 
[cardin.zip](https://github.com/sui77/rc-switch/files/2687638/cardin.zip)

You may want to try this version of RCSwitch https://github.com/1technophile/OpenMQTTGateway/blob/master/lib/rc-switch/RCSwitch.cpp

thallaa, 2018-12-18:
I've followed the instructions of adding a new remote here on the wiki:
[https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1](url)
As the wiki page instructs, I used SimpleRcScanner for obtaining the timings.

I see the version you linked contains a lot more protocols, I will definitely give it a go. Without reading any further, I'm just wondering why it's not merged to this "official" version of rc-switch?

Atihinen, 2020-01-03:
Hi,

so was this resolved or not? :D

I've exactly the same device Go/Tech 36-6010 and can't make the sending work. Reading the "secrets" using this instruction works perfectly fine: https://www.instructables.com/id/Control-CoTech-Remote-Switch-With-Arduino-433Mhz/

1technophile, 2020-01-03:
> You may want to try this version of RCSwitch https://github.com/1technophile/OpenMQTTGateway/blob/master/lib/rc-switch/RCSwitch.cpp

THe fork is now located there:
https://github.com/1technophile/rc-switch

thallaa, 2020-01-03:
I never continued with this (I have the other Nexa switch set working) but I see I should now that there is quite a bit more info available.

Atihinen, 2020-01-04:
> > You may want to try this version of RCSwitch https://github.com/1technophile/OpenMQTTGateway/blob/master/lib/rc-switch/RCSwitch.cpp
> 
> THe fork is now located there:
> https://github.com/1technophile/rc-switch

At least couldn't make it work with this code:

`#include "RCSwitch.h"

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  // Transmitter is connected to Arduino Pin #4  
  mySwitch.enableTransmit(6);
  // Optional set pulse length. This can be change to optimize connection.
  mySwitch.setPulseLength(400);
  // Optional set protocol. 4 works best with the Co/tech
  mySwitch.setProtocol(4);
  // Optional set number of transmission repetitions.
  mySwitch.setRepeatTransmit(1);
  
}

void loop() {
        
        Serial.println("A - on button");  // testing show that not all the codes are neeeded. But atleast 2.
        mySwitch.send("100001....");
        mySwitch.send("110000.... ");
        mySwitch.send("100010....");
        mySwitch.send("100000....");
        delay(5000);

        Serial.println("A - off button");
        mySwitch.send("11001....");
        mySwitch.send("11001....");
        mySwitch.send("11001....");
        mySwitch.send("11000....");
        delay(5000);
}`

I copied the RCSwitch.cpp and RCSwitch.h to root folder of the ino file from https://github.com/1technophile/rc-switch from master branch.

1technophile, 2020-01-04:
> At least couldn't make it work
I would need more details, do you mean that the sending doesn't trigger what you are expecting?

Atihinen, 2020-01-04:
> > At least couldn't make it work
> 
> I would need more details, do you mean that the sending doesn't trigger what you are expecting?

@1technophile hi, yeah. It doesn't do anything to remote switch that I've within 0.5m away.

I've paired the switch with the controller and sending A ON or OFF do work. But using the code to send ON or OFF doesn't do anything.

I've also tried to give separate power from cellphone powerbank to transmitter and to Arduino Nano as well.

1technophile, 2020-01-04:
Hi,
Are you able to receive the signal from the remote control with the library?

Atihinen, 2020-01-04:
Yes, that's how I got the counter codes that I'm trying to send.

1technophile, 2020-01-04:
I suppose you get it with forked library isn't it?

Atihinen, 2020-01-04:
Ah that is true, I'll check if the codes changes. https://cdn.instructables.com/ORIG/F2B/RARN/IMGJD9CD/F2BRARNIMGJD9CD.ino this is the script that I used to get the codes: https://www.instructables.com/id/Control-CoTech-Remote-Switch-With-Arduino-433Mhz/

Atihinen, 2020-01-04:
Yeah, I did get different codes when changing to use the forked library but the end result is still the same. Sending doesn't seem to do anything even if I move the transmitter closer to socket.

Atihinen, 2020-01-05:
Just tried with the send & receive example from here: https://github.com/sui77/rc-switch/tree/master/examples (with forked lib) and can't get anything to receive from transmitter. So it might be also HW issue. I'll try to get new transmitter somewhere to test it out.

Atihinen, 2020-01-10:
now I got the hw working with the examples. I can see that transmitter is trying to send something. But the CoTech is not responding to the codes.

1technophile, 2020-01-13:
Maybe try with different PulseLength values

rywi, 2020-02-24:
I think it's is encrypted: 
https://github.com/merbanan/rtl_433/blob/a8f438a5fb52f24f854368b581a08aa413bb7654/src/devices/kangtai.c
sort of rolling codes. I'm trying to get my SolverCrest remote to work with 433mhz nodemcu sender reciever. 
No success so far

Atihinen, 2020-02-27:
@thallaa do you have code samples somewhere and any idea if the same code would work with this kit? https://www.clasohlson.com/fi/Kaukokytkinsarja-Nexa-MYC-3,-3-kpl/p/Pr366902000

Atihinen, 2020-03-10:
Yeah got my solution to work with Nexa MYC-3 remote switches, once I've refactored the code and updated documentation I can share the results here as well.

Atihinen, 2020-04-04:
https://github.com/Atihinen/Arduino-Nexa if somebody is interested, hopefully this ticket is closed at some point. I still have the co-tech gear if somebody wants me to tryout some example code to verify if it's working or not.

vegovs, 2020-04-10:
@Atihinen Hello, did you get the Cotech gear to work? I am in the same boat, only its from another produced Luxorparts: 

A On:

Decimal: 3798560147 (32Bit) Binary: 11100010011010010110110110010011 Tri-State: not applicable PulseLength: 740 microseconds Protocol: 2
Raw data: 7400,1368,684,1388,680,1392,680,352,1712,356,1712,352,1720,1380,700,336,1724,348,1740,1364,680,1388,684,352,1728,1372,700,336,1716,352,1716,1384,676,360,1716,1384,680,1392,676,360,1716,1384,672,1400,676,360,1728,1376,672,1400,664,372,1700,364,1708,1392,676,360,1716,348,1712,1388,688,1392,676,

Decimal: 3450469075 (32Bit) Binary: 11001101101010011111101011010011 Tri-State: not applicable PulseLength: 738 microseconds Protocol: 2
Raw data: 7384,1388,684,1388,680,356,1712,356,1728,1372,676,1396,676,360,1708,1392,676,420,96,872,680,356,1724,1380,676,360,1716,1384,692,344,1708,360,1704,1396,684,1392,676,1396,680,1392,704,1368,712,324,1724,1380,708,328,1720,1380,696,1380,696,340,1732,1368,700,336,1748,336,1720,1380,688,1384,740,

A Off:

Decimal: 3910506643 (32Bit) Binary: 11101001000101011001100010010011 Tri-State: not applicable PulseLength: 743 microseconds Protocol: 2
Raw data: 7424,1348,688,1384,700,1372,688,348,1716,1384,684,356,1720,344,1724,1376,688,336,1724,340,1732,336,1736,1364,684,352,1724,1380,688,348,1728,1372,680,1392,684,352,1712,344,1716,1384,688,1384,684,352,1716,352,1724,340,1724,1376,696,344,1716,348,1724,1376,700,336,1716,348,1736,1368,688,1384,696,

Decimal: 3102632147 (32Bit) Binary: 10111000111011100110100011010011 Tri-State: not applicable PulseLength: 741 microseconds Protocol: 2
Raw data: 7416,1356,696,340,1744,1360,692,1380,688,1384,688,348,1724,344,1728,340,1724,1364,712,1364,692,1380,684,352,1716,1384,692,1384,680,1392,676,360,1716,352,1720,1380,688,1384,692,344,1708,1396,688,348,1712,352,1724,344,1716,1384,692,1384,680,356,1724,1380,680,356,1724,344,1724,1380,680,1392,680,

B On: 

Decimal: 118815957 (32Bit) Binary: 00000111000101001111110011010101 Tri-State: 00F10FF011101FFF PulseLength: 740 microseconds Protocol: 2
Raw data: 7412,328,16,1740,308,1752,312,1760,308,1768,296,1784,1320,716,1356,708,1364,336,1756,308,1736,332,1740,1364,708,328,1740,1360,700,336,1728,340,1744,1356,712,1364,696,1376,700,1376,680,1392,692,1380,672,368,1716,352,1720,1380,684,1388,672,368,1716,1388,680,356,1744,1356,688,352,1724,1376,692,

Decimal: 4037445525 (32Bit) Binary: 11110000101001101000011110010101 Tri-State: not applicable PulseLength: 745 microseconds Protocol: 2
Raw data: 
7440,1336,716,1356,708,1364,688,1384,704,336,1712,352,1744,324,1732,332,1752,1360,700,336,1748,1356,712,324,1744,324,1756,1344,696,1380,708,328,1724,1376,700,340,1724,332,1728,340,1740,328,1736,1364,700,1376,704,1368,696,1376,700,340,1720,344,1736,1368,696,340,1740,1364,692,344,1736,1368,696,

B Off:

Decimal: 4025677525 (32Bit) Binary: 11101111111100101111011011010101 Tri-State: not applicable PulseLength: 743 microseconds Protocol: 2
Raw data: 7420,1356,712,1364,700,1372,696,344,1736,1364,692,1384,688,1384,684,1388,696,1388,676,1400,684,1392,680,1392,680,356,1720,348,1712,1392,676,360,1716,1388,680,1396,672,1400,688,1388,684,352,1708,1396,688,1384,676,360,1716,1388,676,1400,680,356,1720,1384,684,356,1712,1392,680,356,1732,1372,696,

Decimal: 3882577813 (32Bit) Binary: 11100111011010110110111110010101 Tri-State: not applicable PulseLength: 744 microseconds Protocol: 2
Raw data: 7424,1352,708,1364,700,1376,700,336,1728,340,1740,1364,708,1364,688,1384,684,356,1744,1360,680,1392,680,360,1716,1388,692,344,1720,1384,696,1376,680,360,1712,1392,692,1380,680,356,1716,1388,680,1396,672,1400,688,1388,680,1392,684,356,1712,356,1704,1396,680,360,1708,1392,688,352,1728,1376,692,

C On:

Decimal: 2571151835 (32Bit) Binary: 10011001010000001010100111011011 Tri-State: not applicable PulseLength: 740 microseconds Protocol: 2
Raw data: 7408,1364,700,336,1744,320,1740,1356,700,1372,692,344,1748,316,1728,1372,692,356,1728,1372,692,344,1720,344,1728,336,1744,324,1752,312,1760,304,1752,1348,708,328,1748,1356,716,320,1756,1344,696,340,1736,328,1756,1344,708,1364,700,1372,680,356,1724,1372,696,684,1388,688,348,1708,1388,700,1372,

Decimal: 1140521371 (32Bit) Binary: 01000011111110101111100110011011 Tri-State: not applicable PulseLength: 741 microseconds Protocol: 2
Raw data: 7424,316,1748,1352,700,336,1740,324,1744,320,1752,312,1756,1344,716,1352,708,1360,696,1372,696,1380,680,1384,700,1368,680,360,1716,1380,680,356,1720,1380,688,1384,676,1396,668,1396,680,1388,672,364,1712,352,1716,1384,684,1388,680,356,1712,352,1720,1380,680,1392,692,348,1716,1380,684,1388,684,

C Off:

Decimal: 4254933211 (32Bit) Binary: 11111101100111010010000011011011 Tri-State: not applicable PulseLength: 741 microseconds Protocol: 2
Raw data: 7404,1368,684,1388,696,1376,684,1388,684,1384,676,1396,684,352,1704,1400,692,1396,684,356,1708,356,1708,1392,676,1396,684,1388,684,352,1708,1392,680,356,1732,332,1708,1392,688,348,1712,356,1712,352,1716,352,1720,344,1736,1364,684,1388,676,360,1720,1380,680,1396,676,360,1704,1396,680,1392,680,

Decimal: 583958939 (32Bit) Binary: 00100010110011101000000110011011 Tri-State: not applicable PulseLength: 741 microseconds Protocol: 2
Raw data: 7408,332,1732,332,1752,1348,696,340,1732,332,1740,328,1728,1368,688,348,1728,1372,696,1372,688,352,1716,348,1732,1364,700,1372,684,1384,680,356,1720,1384,676,360,1708,344,1724,344,1716,348,1728,336,1728,340,1724,1376,696,1372,688,348,1728,336,1724,1376,712,1360,688,348,1720,1380,680,1388,684,

D On: 

[... cut ...]
Decimal: 1489881152 (32Bit) Binary: 01011000110011011100100001000000 Tri-State: not applicable PulseLength: 743 microseconds Protocol: 2
Raw data: 7432,316,1764,1348,712,328,1736,1372,704,1376,680,360,1744,328,1728,340,1760,1360,692,1388,704,336,1740,328,1752,1360,712,1372,684,356,1728,1380,688,1396,688,1392,680,340,1716,356,1724,1388,684,352,1720,356,1716,352,1728,344,1732,1380,712,328,1724,344,1772,300,1752,320,1764,304,1756,316,1780,

I've tried creating my own protocol but never seem to the timing perfect. Wall sockets wont respond at all with very similar signals.

Atihinen, 2020-04-14:
@vegovs unfortunately didn't make it work, but yeah I still have the gear ready if somebody has a solution that needs to be tested out

Victoare, 2021-12-30:
Chiming in as an another one getting to this conclusion. 
In here there is a vendor "Home by Somogyi electronics" selling these rather cheap but well made rc sockets. They also sell a smart bridge solution which uses the kangtai smart app, so this is how i find this thread while tried to make it work with rc-switch.
Unfortunately i have to move on and find an another solution, because i wanted to decode more type of signals (weather stations, pir sensors and whatnot) and disect they properties, but have to acknowledge, that this lib is a fine one, just it does not work with this kangtai protocol.

F48i, 2025-02-27:
Hey,
I came across this issue thread when researching how to automate my Cleverio Outdoor Remote Switches from Kjell & Company (https://www.kjell.com/se/varumarken/cleverio/smarta-hem/startpaket/cleverio-mini-fjarrstrombrytare-for-utomhusbruk-3000-w-4-pack-p51584).

Since this thread gave me some valuable input I want to leave behind some of my findings, in case someone is still interested or stumbles across this thread in the future.

I had quite some issues sniffing the data communication, so I opened up the remote and scoped the output data instead. 

Every button push can be broken down into the following 3 sections.

| 8x Same repeating code of start bit and 24 Data Bit | 
| 8x Same repeating code with different bit encoding |
| 8x repeating code with same bit encoding like the first section |

Section 1 Example:
[image]
Section 2 Example:
[image]
Section 3 Example:
[image]

Each button has 4 different patterns that repeat over and over again. I started to look at the first code a bit more detailed and figured out some kind of pattern that seems to be the same for all buttons.

(Ignoring the Start Bit)
Button A On:
00010010110001000100 0000
00010110010100000111 0000
00010001000001101011 0000
00010101111111100011 0000

Button A Off:
00011010100010001001 0000
00011101001110011010 0000
00010111111000100010 0000
00011001010001010001 0000

Button B On:
00010010110001000100 0100
00010110010100000111 0100
00010001000001101011 0100
00010101111111100011 0100

Button B Off:
00011010100010001001 0100
00011101001110011010 0100
00010111111000100010 0100
00011001010001010001 0100

Button C On:
00010010110001000100 1100
00010110010100000111 1100
00010001000001101011 1100
00010101111111100011 1100

Button C Off:
00011010100010001001 1100
00011101001110011010 1100
00010111111000100010 1100
00011001010001010001 1100

[Button D seems to invert all the logic from above]

Button D On:
00011010100010001001 0010
00011101001110011010 0010
00010111111000100010 0010
00011001010001010001 0010

Button D Off:
00010010110001000100 0010
00010110010100000111 0010
00010001000001101011 0010
00010101111111100011 0010

It looks like that the last 4 bits are a device identification, while Button A-C have the same 4 codes for On and same 4 codes for Off. Button D inverts the on/off logic but uses the same codes.
Those codes looked somewhat familiar with what was posted above from @thallaa. 

So I wrote a small program:
```
RCSwitch::Protocol cleverio = { 370, {  1, 6 }, {  1,  3 }, {  3,  1 }, false };
// Optional set protocol (default is 1, will work for most outlets)
mySwitch.setProtocol(cleverio);
// Optional set number of transmission repetitions.
mySwitch.setRepeatTransmit(4);

Serial.println("A - on button");
mySwitch.send("000100101100010001000000");
Serial.println("A - off button");
mySwitch.send("000110101000100010010000");
```

After fine tuning the protocol a bit and transmitting any of the above codes, all of my outlets started working :)
When adding @thallaa codes into my program, one of my outlet also started switching. The reason why I even started to work on this in the first place. Not sure what the initial 20 bits are, maybe some kind of remote identifier.

I never really investigated section 2 and 3 from the data sequence, since my goal was reached. I assume they are somewhat used for pairing and unpairing the outlets. But I just use the remote for that. 

I was able to add the codes and protocol into my ESPHOME project and connect it to my homeassistant for remote controlling a bunch of outdoor lights around my house.
Thanks to everyone that posted here in the past.

Fabian