sui77/rc-switch #271: ESP8266 build-in LED

SampusCZE, 2019-01-13:
Hi,
is there any way to disable build-in LED? 

When i upload transmiting sketch with rc-switch lib. built-in led is still turned on and flashing while transmitting. Is a bit annoying.

Tested with Wemos D1 mini, NodeMcu so far.

RO-ny9, 2019-04-02:
Hi, did it work with Wemos D1 mini? 
I am trying, but without any success... please can you send me code that you used with wemos D1? Thanks!

1technophile, 2020-01-05:
> When i upload transmiting sketch with rc-switch lib. built-in led is still turned on and flashing while transmitting. Is a bit annoying.
> 
> Tested with Wemos D1 mini, NodeMcu so far.

Hi,

You could use a different pin for transmitting (not used by the BUILD_IN led)

closed 2020-01-05 by 1technophile