sui77/rc-switch #272: Making it work with DOIT ESP32 DEVKIT V1

hycday, 2019-01-26:
Hi,

I got this DOIT ESP32 DEVKIT V1 board (https://cdn.hackaday.io/files/869123718443008/ESP32-Devkit-Pinout_19.pdf), and a pair of receiver/transmitter at hand.
I am trying to capture/decode 433mhz signals.
I am using the very simple example below, and for the wiring, I have VCC of receiver to VIN of ESP32, GND to GND and DATA of receiver to D27 of ESP32.

`#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(115200);
  Serial.print("Ready to receive.");  
  mySwitch.enableReceive(27); 
}

void loop() {
  if (mySwitch.available()) {
    
    Serial.print("Received ");
    Serial.print( mySwitch.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( mySwitch.getReceivedProtocol() );

    mySwitch.resetAvailable();
  }
}`

After upload, I open the Serial monitor and...... nothing happens.
Not sure what I did wrong to be honest. I know the ESP32 works and I trying different receivers, with no result.
I think it may be because of the enableReceive value, but I am struggling to understand what to put here... I tried 0 1 2 3 etc with no result.

Any help is welcome !

roblatour, 2019-02-06:
did you make any progress on this? I am having the same problem.

hycday, 2019-02-06:
I actually did right now by doing a very silly thing : i soldered antenna to the transmitter and to the receiver and i now have some signals....

roblatour, 2019-02-06:
ok thanks I'll try that; but the problem I am getting is that the MySwitch.available() is returning a value false

hycday, 2019-02-10:
yes i had the same, until I wired the antenna on the receiver, and then it worked.

I used `mySwitch.enableReceive(27);` when wiring the DATA to pin 27 (the 9th one, on the left column, counting from bottom, on this pinout layout : https://cdn.hackaday.io/files/869123718443008/ESP32-Devkit-Pinout_19.pdf)

closed 2019-02-12 by hycday