sui77/rc-switch #284: Random lad on receive

anil8anil, 2019-03-28:
Hi, i have an iot project that i use ESP8266-01 and  QIACHIP RF 433Mhz Superheterodyne Receiver. I control an on/off relay on internet and by 433mhz rf. I use blynk as wireless platform. My my main loop on arduino  editor is like this:
`void loop() {
  if(Blynk.connected())  Blynk.run();   
  else BlynkConExist=false; //checks blynk connection every 30seconds and sets this variable
  timer.run(); //use it for 30 second blynk checks                                         
  rfConnection();  //my rf function
}`
And my rfConnection function is like this:
`void rfConnection(){
  if (rfSwitch.available()) {                            
    Serial.println("----------------------------------------------------------------");
    Serial.println("Gelen RF Kodu Ayrıntıları: ");
    Serial.print("Received ");                        
    int incomingMsg = rfSwitch.getReceivedValue();
    Serial.print( incomingMsg );
    Serial.print(" / ");
    Serial.print( rfSwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( rfSwitch.getReceivedProtocol() );
    if(incomingMsg==myRFCode){
      Serial.println("TANIMLI RF KODU ALINDI, role durumu degistiriliyor.");
      digitalWrite(relayPin,!digitalRead(relayPin));      //toggle relay state
      if(BlynkConExist)  Blynk.virtualWrite(V1,digitalRead(relayPin));
      delay(500);                                  
    }
    rfSwitch.resetAvailable();
    Serial.println("----------------------------------------------------------------");
  }
}`
My problem is random fais on received rf packages. When i press wall switch transmitter's (ev1527 chip, 12v battery) button, sometimes i can receive the rf code, sometimes i can't.

What i have tried:
1. **Arduino board, SendDemo example sketch and cheap 433mhz transmitetr on transmitter end; esp8266 with my own project codes on receiver end:** If i transmit my rf code by arduino board, i can receive rf code successfully everytime.
2. **Arduino board, ReceiveDemo_Simple example sketch, qiachip receiver module on receiver end; a wall switch as transmitter (ev1527 chip, 12v battery):** When i upload ReceiveDemo_Simple example on arduino board and use the qiachip receiver module, every time when i press the transmitter's (wall switch with ev1527 chip) button, signal is successfully read.

After 1st try, i thought there is a problem about my wall switch transmitter. After 2nd try, i saw wall switch transmitter is ok when i receive rf code with arduino board and example receive sketch.

What is the problem you think? blynk connection? timer? transmitter signal strength? my project codes? decoupling capacitor on receiver? (i didnt use it even on 2nd try)

Thanks

anil8anil, 2019-04-07:
After some research, i have found out that, there is a problem about interrupts on esp8266 when using rcswitch library. After reading Issue (https://github.com/sui77/rc-switch/issues/46) i saw, some solutions were released already. But i still have the problem. 

Then i saw another post [https://forum.arduino.cc/index.php?topic=583166.0](url) In the post it is written that receiving does not work well when wifi is enabled. If i erase esp01's memory (all the wifi credentials) and dont connect to a network, receiving works well. But if i connect a network (even if i disconnect afterwards) receiving is unstable (stable only when i long press transmitter's button) 

Can someone help me about this issue? Thanks in advance...

nathalizator, 2021-03-21:
> After some research, i have found out that, there is a problem about interrupts on esp8266 when using rcswitch library. After reading Issue (#46) i saw, some solutions were released already. But i [...]

Did you find a solution ? i have the same issue with a ESP 32 CAM