sui77/rc-switch #285: MX-RM-5V not connecting?

kiritoavs, 2019-04-18:


kiritoavs, 2019-04-18:
Hi, im pretty new to this as im only really doing this as a module for university so im skill and knowledge is very limited so please bear with me. 
    i am doing an arduino project that can measure temperature, heart rate and rotation (using accelerometer) and it is working well.  Now I need it to pass through a Bluetooth sensor to remove some of the wires as i want my data to be shown on a LCD screen that is not connected via wires.
    I have a MX-RM-5V transmitter and receiver (which was given by the lecture) and was trying just some short range things such as it saying "hello world" or it trying to light up a LED light however none of these seem to work as i do not get anything on any of the serial monitors. Also, I have verified that both arduinos do indeed work.
For example, this is some code I got off the internet.
    Transmitter
#include <RH_ASK.h>
#include <SPI.h> // Not actually used but needed to compile

RH_ASK driver;

void setup()
{
    Serial.begin(9600);    // Debugging only
    if (!driver.init())
         Serial.println("init failed");
}

void loop()
{
    const char *msg = "Hello World!";
    driver.send((uint8_t *)msg, strlen(msg));
    driver.waitPacketSent();
    delay(1000);
}

Reciever
#include <RH_ASK.h>
#include <SPI.h> // Not actualy used but needed to compile

RH_ASK driver;

void setup()
{
    Serial.begin(9600);  // Debugging only
    if (!driver.init())
         Serial.println("init failed");
}

void loop()
{
    uint8_t buf[12];
    uint8_t buflen = sizeof(buf);
    if (driver.recv(buf, &buflen)) // Non-blocking
    {
      int i;
      // Message with a good checksum received, dump it.
      Serial.print("Message: ");
      Serial.println((char*)buf);         
    }
}

Any help will be appreciated. Thanks in advance!

1technophile, 2020-01-05:
As you are using #include <RH_ASK.h> and not rcswitch I advise you to ask into the corresponding repository.
Rcswitch is not able to send text messages for info.

closed 2020-01-05 by 1technophile