sui77/rc-switch #286: Please help with Raw DATA

madajjan, 2019-04-26:
Raw DATA is:

16048,1524,1452,524,960,516,964,520,964,1020,480,1012,484,1008,488,504,980,1000,496,500,980,504,980,504,980,500,984,500,980,996,500,992,508,992,500,988,512,488,992,492,992,492,988,992,508,484,1000,484,1000,484,992,988,520,476,1000,484,1000,480,1000,984,512,984,516,476,1004,976,520,976,516,480,1004,480,1004,972,532,968,520,972,524,972,524,472,1012,472,1008,476,1008,476,1004,976,524,472,1008,476,1008,972,524,968,528,472,1012,964,532,964,528,968,520,1460

and repeat.

I need for this protocol. Can someone help? Thanks

kehibvbll, 2019-06-02:
@madajjan How did you get raw data from an unknown protocol?

Smart111222, 2019-06-05:
756,800,236,280,756,280,760,796,240,796,244,792,240,796,248,784,248,792,244,792,248,788,248,788,252,780,260,256,772,276,760,272,764,276,760,276,760,280,756,276,764,792,248,272,3084,284,756,288,748,288,748,800,244,276,760,276,756,796,244,796,240,796,244,792,244,792,244,788,252,788,248,784,256,784,256,780,252,268,764,276,764,268,764,280,756,276,760,280,756,280,756,792,252,272,3084,292,748,284,752,284,752,800,240,276,760,280,756,800,236,796,240,792,248,788,252,788,244,788,248,788,252,788,252,784,248,784,256,264,768,268,764,276,764,272,764,276,756,276,764,280,756,792,248,272,3088,800,248,272,760,280,752,796,244,268,768,276,760,796,244,792,244,792,248,788,248,788,248,788,252,784,252,784,252,780,256,784,252,268,772,264,768,268,768,272,764,272,764,276,760,276,756,792,248,276,3088,792,248,280,756,276,760,796,244,276,760,276,756,796,244,796,244,788,244,792,252,788,248,784,252,784,252,784,252,784,252,784,252,268,772,268,764,272,764,272,764,272,768,272,760,272,768,788,252,268,3088,804,236,280,760,276,760,792,248,272,760,276,764,792,244,792,248,792,244,788,248,788,248,788,252,784,252,784,252,784,256,780,256,264,772,260,776,264,772,264,764,276,768,272,760,276,764,784,248,276,3088,800,244,272,760,280,756,800,244,272,764,276,756,792,248,796,240,792,244,788,248,788,248,788,252,784,252,788,248,784,256,784,256,256,776,264,772,268,764,272,768,272,760,276,760,276,760,792,252,268,3092,800,240,276,764,272,764,792,244,276,760,280,760,788,244,792,248,792,240,796,244,792,252,776,256,780,256,780,260,780,252,784,256,260,772,272,768,268,768,268,768,268,768,272,760,276,764,780,256,272,3092,796,248,272,760,276,760,792,248,272,760,276,764,792,244,792,244,792,248,788,248,788,252,784,248,788,248,784,256,784,256,780,252,264,776,264,768,268,768,272,764,272,760,280,760,272,764,792,248,268,3092,792,252,272,760,280,756,792,252,272,764,272,760,796,244,792,244,788,248,792,252,780,252,784,252,784,252,780,260,780,256,780,256,260,768,272,768,268,768,268,768,268,768,272,764,272,764,788,248,272,3088,284,756,280,756,284

help to deal with the protocol

Smart111222, 2019-06-06:
760,280,756,280,760,272,768,276,756,792,248,272,3088,796,244,792,252,272,756,280,760,276,756,280,756,800,244,788,248,788,248,792,244,792,244,788,252,784,252,788,248,784,256,780,256,264,768,272,764,272,764,268,772,268,764,276,764,272,764,788,248,272,3092,800,244,792,240,276,760,276,760,276,760,280,756,796,248,788,248,788,252,784,252,784,248,788,260,780,252,784,248,784,256,784,256,264,768,268,768,272,760,276,764,268,764,280,760,272,764,788,256,264,3092,800,244,792,244,276,764,272,760,280,756,284,756,796,240,792,248,788,248,788,248,788,252,788,248,780,256,784,256,784,252,780,256,264,772,272,764,268,768,268,768,272,764,276,760,276,760,792,248,272,3088,796,252,788,248,268,764,276,764,276,760,272,760,800,240,792,248,788,248,788,248,792,252,780,256,780,256,780,256,780,256,780,256,268,764,268,768,272,772,264,768,268,764,276,760,276,764,792,252,264,3092,796,248,792,244,272,764,276,760,276,760,280,756,796,244,792,248,792,244,788,252,784,252,784,252,784,248,788,256,780,260,776,252,268,772,264,768,272,768,268,764,272,764,276,760,280,760,788,256,264,3096,796,244,788,252,268,764,276,760,276,760,276,760,792,248,792,244,788,248,788,248,784,260,784,248,784,256,780,256,784,252,784,256,260,772,268,768,268,768,272,764,272,768,268,764,276,764,788,252,264,3092,800,244,792,248,268,768,276,756,280,756,280,760,792,248,792,244,788,252,788,248,788,248,788,252,780,256,784,256,780,256,784,252,264,772,268,772,264,772,264,768,272,764,272,764,276,760,788,252,268,3096,796,244,792,248,268,768,272,764,272,760,280,756,792,248,792,248,788,244,792,248,784,256,772,260,788,248,788,248,788,256,776,252,264,772,268,772,268,772,264,768,268,764,276,768,268,764,788,252,268,3096,280,752,800,244,276,756,280,760,276,764,272,764,792,240,796,240,792,244,792,252,784,248,792,252,784,252,784,252,784,256,780,252,264,772,272,764,272,764,272,764,276,756,280,760,276,760,792,248,272,3092,284,752,800,244,272,760,280,756,280,756,276,760,796,248,784,248,792,244,792,248,784,252,792,248,788,248,784,260,780,248,784,256,268,768,268,764

Smart111222, 2019-06-06:
764,796,244,784,252,272,764,272,764,788,248,784,256,780,252,788,248,784,260,780,256,776,260,776,260,772,268,772,256,264,772,268,768,264,768,268,776,260,768,272,764,272,768,788,248,268,3088,800,248,268,764,788,252,788,248,268,772,268,760,788,256,780,256,780,264,772,256,784,252,780,260,780,252,784,252,780,260,776,260,264,772,260,768,272,772,264,772,264,768,268,768,264,772,788,252,264,3092,796,248,272,764,788,252,780,252,272,768,268,764,784,256,784,256,780,252,784,260,776,256,780,252,780,256,776,264,780,260,772,264,256,780,256,772,268,764,272,768,268,764,276,764,272,764,784,252,268,3096,792,248,268,768,792,244,784,256,264,772,268,764,788,248,788,256,780,252,784,248,788,256,776,256,780,260,780,256,776,260,768,272,256,776,264,768,264,776,264,768,272,764,268,768,268,768,784,256,264,3096,792,248,268,772,784,248,784,252,268,768,272,760,788,256,780,252,784,252,784,256,788,248,780,260,776,256,776,260,780,264,768,264,256,772,264,776,260,772,264,772,268,772,264,768,268,772,780,256,260,3096,284,756,276,760,792,248,788,248,272,760,276,764,788,248,792,244,788,252,780,256,784,252,780,256,780,256,784,252,780,256,780,260,260,772,268,768,268,768,264,768,276,764,268,764,272,764,788,252,264,3096,284,752,280,760,788,256,784,248,272,764,272,760,796,244,788,248,784,256,784,252,784,252,780,256,780,256,780,252,784,260,776,260,260,772,264,776,264,764,268,768,272,768,272,760,276,760,788,256,264,3092,280,760,280,752,796,244,792,248,272,756,280,760,792,244,792,244,788,248,788,256,780,256,784,252,780,256,780,260,780,256,776,256,260,772,268,772,268,764,272,764,272,764,268,768,272,764,788,252,264,3092,284,756,284,752,788,256,788,248,272,760,276,760,792,252,788,244,788,248,784,252,776,260,784,256,784,248,788,252,780,256,780,256,264,772,268,764,268,768,268,768,272,764,272,760,280,764,788,248,272,3088,284,752,284,760,784,248,792,252,268,760,272,764,792,244,792,248,784,252,784,252,788,248,784,252,784,252,788,252,780,260,776,256,264,772,268,764,272,764,272,768,264,768,272,768,268,768,784,248,268,3092,288,752,284

Smart111222, 2019-06-06:
252,268,3088,796,248,272,764,792,240,280,760,276,756,284,756,792,244,796,244,792,244,784,252,788,248,788,248,788,252,784,248,792,248,784,256,260,772,268,768,272,764,272,760,276,764,268,764,276,764,792,244,272,3092,796,244,276,764,788,248,272,756,280,760,276,760,796,244,792,240,800,244,784,248,792,248,784,252,784,256,784,252,784,252,784,252,268,768,268,764,272,768,272,764,272,764,272,768,268,760,792,252,268,3088,800,244,276,760,792,244,276,760,280,756,276,760,796,248,788,248,788,248,788,248,788,248,784,256,784,252,784,248,788,252,784,252,260,772,272,764,272,768,268,764,276,764,268,768,276,760,788,248,272,3088,800,244,272,764,792,248,272,760,280,756,276,760,792,244,792,248,788,252,788,248,784,252,788,252,780,252,788,252,784,252,784,256,264,768,268,768,268,764,276,764,272,760,276,764,272,760,796,248,264,3096,800,244,272,760,796,244,272,764,272,760,280,756,796,248,788,248,788,244,788,252,788,248,788,248,792,244,784,256,784,256,780,252,268,772,264,768,272,764,276,760,272,768,272,760,276,760,788,248,268,3096,800,240,276,764,788,248,272,764,276,756,280,756,796,244,788,252,788,248,788,248,788,252,784,252,784,252,784,256,776,264,772,256,264,772,264,772,268,772,264,764,276,760,276,768,268,760,792,252,268,3092,796,252,268,764,788,252,268,760,280,760,276,760,792,248,788,248,788,252,788,244,792,248,784,256,776,256,784,256,780,260,776,256,264,772,260,768,276,764,272,760,280,760,272,764,276,756,796,248,268,3092,796,248,272,760,796,248,264,772,268,764,276,756,792,248,788,252,788,244,788,248,792,244,788,252,784,256,780,252,784,256,776,264,260,768,268,768,268,768,272,760,276,764,268,764,280,760,788,248,268,3092,288,748,284,760,792,244,272,764,276,756,284,760,796,240,792,244,792,248,784,252,788,248,784,252,792,248,788,244,788,256,776,260,264,768,268,768,268,772,272,760,272,760,280,760,276,760,788,248,272,3092,284,752,284,748,804,244,276,756,280,756,280,756,796,248,792,240,796,244,788,244,792,248,788,248,784,256,788,248,784,256,776,260,260,768,272,764,272,768,272,760,276,760,276,760,276,760,796

Smart111222, 2019-06-06:
{ 250, { 1, 12  }, {  1,  3 }, {  3,  1 }, false }  
This code does not work. Please help.

kehibvbll, 2019-06-08:
@Smart111222 How did you get raw data?!
When a protocol is unknown, Rc-switch does not show any thing!

kuchod, 2019-06-08:
> { 250, { 1, 12 }, { 1, 3 }, { 3, 1 }, false }
> This code does not work. Please help.

use it, capture its output, compare it with the original one and modify the code to make it more alike.

kehibvbll, 2019-06-12:
@kuchod How i can capture output?
Rc-switch must known an protocol to capture it.