sui77/rc-switch #295: Please, help to find MD-2018R vibration module protocol

TDave007, 2019-07-05:
Hi, my name is Dave. 
Im trying to find the protocol of MD-2018R vibration module. It works over 433Mhz.
I have tried it with Martin-Laclaustra's ino and library's examples but they dosnt work for me, they dont show anything. I have captured the RAW with simplercscanner:
4400,24,10444,44,6516,40,4324,40,3368,24,148,24,2504,20,1636,52,132,20,3304,36,12,16,6144,12,984,12,4860,64,2036,56,764,56,1856,48,9792,40,5672,68,2028,12,340,68,16,40,1196,64,452,68,128,100,1408,36,748,44,2900,24,1144,400,400,796,404,792,408,788,804,392,408,788,804,392,408,788,408,788,808,388,412,784,412,784,412,784,412,784,412,784,412,784,808,388,808,388,412,784,812,384,412,788,808,388,412,784,808,388,412,784,412,784,808,388,412,784,412,784,412,788,808,384,412,784,412,788,808,384,812,384,412,780,420,780,4004,384,412,784,412,784,416,780,812,384,416,780,812,384,416,780,416,780,812,384,416,780,416,784,416,776,420,776,420,776,420,780,812,380,816,380,420,776,816,384,416,776,816,384,416,780,816,380,416,780,416,780,816,380,416,780,420,776,420,776,816,380,420,776,420,780,812,380,820,376,420,784,412,776,4008,380,420,776,420,776,420,776,816,380,420,776,816,380,420,776,420,776,820,376,420,776,420,776,420,776,420,776,424,772,424,772,820,376,820,376,424,772,820,376,424,772,820,376,424,772,820,380,420,772,424,772,824,376,420,772,424,776,420,776,820,376,424,772,424,772,820,376,820,376,424,776,420,772,4012,376,420,776,420,776,420,776,820,376,420,776,820,376,424,772,424,772,820,376,424,772,424,772,424,772,424,772,424,772,424,772,824,372,824,372,424,776,820,376,420,776,820,376,420,776,820,376,420,776,424,772,820,376,424,772,424,772,424,772,820,376,424,772,424,772,820,376,824,372,424,776,420,772,4012,376,424,772,428,768,424,772,820,376,420,776,824,372,424,772,424,772,824,376,424,772,424,772,424,768,428,772,424,772,424,772,820,376,820,376,424,772,820,376,424,772,820,376,424,772,824,372,424,772,424,772,824,372,424,772,424,776,420,776,820,372,428,768,428,772,820,376,820,372,428,772,424,772,4012,372,424,772,428,768,428,768,824,376,428,764,824,376,424,772,420,776,820,376,424,772,424,772,424,772,424,772,424,772,424,772,820,376,824,372,424,772,824,372,428,768,824,372,428,768,824,376,424,772,424,768,824,376,424,772,424,772,424,772,824,372,424,772,424,772,824,372,824,372,424,772,424,772,

I have tried the make the protocol by myself, but rc-switch hasnt recognized it...

Could sameone help to find it?

Thanks a lot

Dave

LaCocoRoco, 2019-07-11:
On a first glance this probably is the signal: 4004,384,412,784,412,784,416,780,812,384,416,780,812,384,416,780,416,780,812,384,416,780,416,784,416,776,420,776,420,776,420,780,812,380,816,380,420,776,816,384,416,776,816,384,416,780,816,380,416,780,416,780,816,380,416,780,420,776,420,776,816,380,420,776,420,780,812,380,820,376,420,784,412,776

You could try to change "nSeparationLimit = 4300;" to "nSeparationLimit = 3500;"
All numbers together my not exceed 67.

Martin-Laclaustra, 2019-07-11:
try
{ 400, { 2, 10 }, { 1, 2 }, { 2, 1 }, false }
but you will need more than 32 bits. You will find information about the needed modifications in the wiki.