sui77/rc-switch #303: only 32 bit

mrgfx, 2019-11-04:
Hi. i would like to send longer signal than 32 bit. Is it possible with this library(I´ve tested tutorial Add_New_Remote_Part_2, but i couldnt recognize new protocol). Is it possible to send binary longer than 32bit with arduino? I´ve received binary code with SDR.

mrgfx, 2019-11-04:
Or could you please help me with: 880,408,880,404,456,832,452,832,456,832,884,404,452,836,452,832,884,404,880,408,452,832,456,832,456,832,456,832,456,832,880,404,460,828,25728,2548,452,836,880,404,880,408,880,404,884,404,880,408,448,836,880,408,452,832,456,832,880,404,884,404,884,404,880,404,880,408,452,836,452,832,880,408,880,404,888,400,880,408,880,404,452,836,452,832,456,832,884,404,452,836,452,832,456,832,884,404,880,404,884,404,884,404,880,404,884,400,884,404,880,408,452,832,456,832,452,836,884,400,460,828,456,832,884,404,880,404,456,832,452,836,452,836,452,832,456,832,884,400,460,832,25720,2552,452,836,880,404,880,408,880,404,880,408,876,408,452,836,880,404,456,836,448,836,880,404,880,408,884,400,884,400,884,404,452,836,452,836,880,404,880,408,880,404,880,408,880,404,456,832,452,836,452,832,884,404,452,836,452,832,456,832,884,404,880,408,880,404,884,404,880,404,884,404,880,404,884,404,456,832,456,828,456,832,884,404,456,832,456,832,880,404,884,404,456,828,460,832,452,836,452,832,456,832,884,404,452,832,25728,2552,452,836,880,404,884,404,880,404,880,408,880,408,452,832,880,408,452,836,452,836,880,404,884,400,884,404,884,404,880,408,448,836,456,832,884,400,884,404,880,408,880,408,880,404,452,836,452,832,456,836,880,404,456,832,452,832,456,836,880,404,884,404,880,404,884,404,880,408,880,404,884,404,880,404,456,832,452,836,452,832,884,404,456,832,456,832,884,404,884,400,456,832,456,832,456,832,456,828,460,828,884,404,456,832,25720,2548,456,832,880,404,880,408,880,404,880,408,880,404,456,832,880,408,448,836,452,836,880,404,880,408,880,408,880,404,880,404,456,832,456,828,884,404,884,404,880,404,884,400,884,404,456,832,452,832,456,832,884,404,456,832,452,832,456,832,884,400,888,400,884,404,880,404,884,404,880,404,884,404,884,400,456,832,456,832,452,832,884,404,452,832,456,832,884,404,884,400,456,832,456,832,456,828,456,832,456,832,884,404,456,828,25724,2548,452,836,880,404,880,408,880,404,884,404,880,404,452,836,880,404,456,832,456,832,880,404,884,404,880,404,884,404,880,408,452,832,452,836,884,404,884,400,880,404,

https://test.sui.li/oszi/

1technophile, 2019-12-03:
There is several forks implementing 64bits support, did you tested with them?
https://github.com/atsage/rc-switch
https://github.com/lulol/rc-switch

mrgfx, 2019-12-03:
Thank you. I will test it tomorrow.

a1studmuffin, 2020-01-11:
This also fixed my issue - my particular RF outlets (generics marked "RCS-19AS") use a 33-bit code. With a 64-bit version of rc-switch and an additional protocol (`{ 275, { 38,  1 }, {  1,  3 }, {  3,  1 }, false },`), all was well and I could sniff and transmit codes. It'd be great if this functionality was added into the main branch of rc-switch as it seems more RF switches are popping up using >32-bit codes, and <=32-bit codes would be backwards compatible with a 64-bit implementation.

1technophile, 2020-01-11:
Yes I plan to do it