sui77/rc-switch #308: Can receive but can't transmit

frbar, 2019-12-06:
Wiring between arduino and receiver/transmitter is fine and works with many remotes with different protocols.

I'm struggling with a last one - I receive fine:

```
19:08:48.387 -> Received 3305378 / 24bit Protocol: 0
19:08:48.423 -> data bits of pulse train duration: 21712
19:08:48.459 -> proposed protocol: { 302, { 70, 3 }, { 1, 2 }, { 2, 1 }, true }
19:08:48.531 -> ====
19:08:48.531 -> first level down
19:08:48.567 -> 21100 812 
19:08:48.567 -> 320 612 288 612 584 316 588 316 292 608 292 608 588 312 300 624 
19:08:48.639 -> 300 600 596 304 596 308 300 600 596 308 596 304 596 308 596 328 
19:08:48.711 -> 600 304 300 600 596 308 300 600 304 596 308 592 600 304 304 600 
19:08:48.780 -> 
19:08:48.780 -> ====
```

But nothing happens when I transmit:

```
RCSwitch mySwitch = RCSwitch();
mySwitch.enableTransmit(3);
mySwitch.setProtocol(0);
mySwitch.setPulseLength(302);
mySwitch.setRepeatTransmit(7);
delay(500);
mySwitch.send(3305379, 24);
delay(2000);
```

Any idea?

frbar, 2019-12-06:
With RC Scanner:

[image]

frbar, 2019-12-08:
With protocol 6, it's fine.

cgmckeever, 2020-12-13:
I dont think 0 is a valid send protocol, gets set to 1
https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L134

```
void RCSwitch::setProtocol(int nProtocol) {
  if (nProtocol < 1 || nProtocol > numProto) {
    nProtocol = 1;  // TODO: trigger an error, e.g. "bad protocol" ???
  }
```