sui77/rc-switch #310: How can I determine s specific pin

ttkdroid, 2019-12-11:
In all the examples I find it says

`  mySwitch.enableReceive(0);  // Receiver input on interrupt 0 (D2)`

But what if I would like to use another pin for the receiver? 

I have a Wemos D1 mini and I can't use pin 2. How could I set up a different pin?

1technophile, 2019-12-18:
Just replace 0 by a compatible pin, see the sheet below:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1864843263

ttkdroid, 2019-12-18:
I did try using “D1” for instance, which maps to GPIO 5 in Wemos D1 Mini, but when I run the “receiver advanced” sketch I get zero output on serial monitor. I’m testing with my garage remote, and in very close proximity with the RF receiver. Both receiver, remote and Wemos are working fine. 

Could anyone make the example sketches work on Wemos D1 mini (esp8266)?

1technophile, 2019-12-19:
If you connect your receiver to D1 you have to put 5 : 
`mySwitch.enableReceive(5);`